m-labs/artiq #746: Missing pythonparser in artiq-2.3 conda installation

dleibrandt, 2017-06-09:
Installed artiq-2.3 on linux using conda:
```
rabi@688101H116ARTIQ:~/dev/artiq_clocks$ conda create -n artiq-2.3 artiq-kc705-nist_clock
```

Attempted to flash the fpga and got the following error:
```
(artiq-2.3) rabi@688101H116ARTIQ:~/dev/artiq_clocks$ artiq_flash -t kc705 -m nist_clock
Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/artiq-2.3/bin/artiq_flash", line 7, in <module>
    from pkg_resources import load_entry_point
  File "/home/rabi/anaconda3/envs/artiq-2.3/lib/python3.5/site-packages/pkg_resources/__init__.py", line 3019, in <module>
    @_call_aside
  File "/home/rabi/anaconda3/envs/artiq-2.3/lib/python3.5/site-packages/pkg_resources/__init__.py", line 3003, in _call_aside
    f(*args, **kwargs)
  File "/home/rabi/anaconda3/envs/artiq-2.3/lib/python3.5/site-packages/pkg_resources/__init__.py", line 3032, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/home/rabi/anaconda3/envs/artiq-2.3/lib/python3.5/site-packages/pkg_resources/__init__.py", line 655, in _build_master
    ws.require(__requires__)
  File "/home/rabi/anaconda3/envs/artiq-2.3/lib/python3.5/site-packages/pkg_resources/__init__.py", line 963, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/home/rabi/anaconda3/envs/artiq-2.3/lib/python3.5/site-packages/pkg_resources/__init__.py", line 849, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'pythonparser' distribution was not found and is required by artiq
```

The following packages are installed:
```
(artiq-2.3) rabi@688101H116ARTIQ:~/dev/artiq_clocks$ conda list
# packages in environment at /home/rabi/anaconda3/envs/artiq-2.3:
#
aiohttp                   2.0.7                    py35_0    conda-forge/label/main
artiq                     2.3                       <pip>
artiq                     2.3             py_0+gitc78aa86e7272    m-labs/label/main
artiq-kc705-nist_clock    2.3             py_0+gitc78aa86e7272    m-labs/label/main
async-timeout             1.2.1                    py35_0    conda-forge/label/main
asyncserial               0.1             py_13+git340e430    m-labs/label/main
binutils-or1k-linux       2.27                          5    m-labs/label/main
ca-certificates           2017.4.17                     0    conda-forge/label/main
certifi                   2017.4.17                py35_0    conda-forge/label/main
cffi                      1.10.0                   py35_0    conda-forge/label/main
chardet                   3.0.2                    py35_1    conda-forge/label/main
curl                      7.52.1                        0    conda-forge/label/main
dateutil                  2.4.2                      py_2    m-labs/label/main
dbus                      1.10.10                       3    conda-forge/label/main
expat                     2.1.0                         2    conda-forge/label/main
fontconfig                2.12.1                        4    conda-forge/label/main
freetype                  2.7                           1    conda-forge/label/main
gettext                   0.19.7                        1    conda-forge/label/main
glib                      2.51.4                        0    conda-forge/label/main
gst-plugins-base          1.8.0                         0    conda-forge/label/main
gstreamer                 1.8.0                         2    conda-forge/label/main
h5py                      2.7.0               np113py35_1    conda-forge/label/main
hdf5                      1.8.18                        0    conda-forge/label/main
icu                       58.1                          1    conda-forge/label/main
jpeg                      9b                            0    conda-forge/label/main
levenshtein               0.12.0                   py35_1    m-labs/label/main
libffi                    3.2.1                         3    conda-forge/label/main
libgfortran               3.0.0                         1  
libgit2                   0.25.0                        0    conda-forge/label/main
libiconv                  1.14                          4    conda-forge/label/main
libpng                    1.6.28                        0    conda-forge/label/main
libusb                    1.0.20                        0    m-labs/label/main
libxcb                    1.12                          1    conda-forge/label/main
libxml2                   2.9.4                         4    conda-forge/label/main
lit                       0.4.1                      py_9    m-labs/label/main
llvmlite-artiq            0.10.0.dev              py35_24    m-labs/label/main
mkl                       2017.0.1                      0  
multidict                 2.1.4                    py35_0    conda-forge/label/main
ncurses                   5.9                          10    conda-forge/label/main
numpy                     1.13.0                   py35_0  
openocd                   0.10.0                        1    m-labs/label/main
openssl                   1.0.2k                        0    conda-forge/label/main
outputcheck               0.4.2                      py_7    m-labs/label/main
pcre                      8.39                          0    conda-forge/label/main
pip                       9.0.1                    py35_0    conda-forge/label/main
prettytable               0.7.2                    py35_1    conda-forge/label/main
pycparser                 2.17                     py35_0    conda-forge/label/main
pygit2                    0.25.0                   py35_0    conda-forge/label/main
pyqt                      5.6.0                    py35_4    conda-forge/label/main
pyqtgraph                 0.10.0                   py35_2    conda-forge/label/main
pyserial                  2.7                      py35_0  
python                    3.5.2                         5    conda-forge/label/main
pythonparser              1.0                        py_8    m-labs/label/main
qt                        5.6.2                         3    conda-forge/label/main
quamash                   0.5.5                      py_4    m-labs/label/main
readline                  6.2                           0    conda-forge/label/main
regex                     2017.06.07               py35_0    conda-forge/label/main
scipy                     0.19.0              np113py35_0  
setuptools                33.1.1                   py35_0    conda-forge/label/main
sip                       4.18                     py35_1    conda-forge/label/main
six                       1.10.0                   py35_1    conda-forge/label/main
sqlite                    3.13.0                        1    conda-forge/label/main
tk                        8.5.19                        1    conda-forge/label/main
wheel                     0.29.0                   py35_0    conda-forge/label/main
xorg-libxau               1.0.8                         3    conda-forge/label/main
xorg-libxdmcp             1.1.2                         3    conda-forge/label/main
xz                        5.2.2                         0    conda-forge/label/main
yarl                      0.10.0                   py35_0    conda-forge/label/main
zlib                      1.2.11                        0    conda-forge/label/main
```

Also, I tried installing using the instructions in #729, but this didn't help.

Perhaps this is related to #711?

sbourdeauducq, 2017-06-09:
What conda version? Try updating it and reinstalling.

dleibrandt, 2017-06-09:
That fixed the problem.  Thanks.

closed 2017-06-09 by dleibrandt