m-labs/artiq #748: SWAG: phase offset between channels

jbqubit, 2017-06-12:
Example below is a mockup of a spin-echo where the TTL is used to switch on/off an RF switch. There is an unexpected phase offset between sawg0 and sawg1. 

```
$ cat test_latency1.py

from artiq.experiment import *

sawg_latency_mu = 988

class SAWGTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")
        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def sawg0_lc_p0_set(self, p):
        delay_mu(-1*sawg_latency_mu)
        self.sawg0.phase0.set(p)
        delay_mu(sawg_latency_mu)

    @kernel
    def sawg1_lc_p0_set(self, p):
        delay_mu(-1*sawg_latency_mu)
        self.sawg1.phase0.set(p)
        delay_mu(sawg_latency_mu)

    @kernel
    def sawg0_lc_a1_set(self, a):
        delay_mu(-1*sawg_latency_mu)
        self.sawg0.amplitude1.set(a)
        delay_mu(sawg_latency_mu)

    @kernel
    def sawg1_lc_a1_set(self, a):
        delay_mu(-1*sawg_latency_mu)
        self.sawg1.amplitude1.set(a)
        delay_mu(sawg_latency_mu)

    @kernel
    def run(self):
        self.core.reset()
        self.ttl_sma.output()
        delay(5*ms)

        # prepare
        tpi = 1*us
        tarm = 2*us
        a0 = 0.5
        f = 200*MHz
        self.sawg0.config.set_clr(1, 1, 1)
        self.sawg0.frequency0.set(f)
        self.sawg0.frequency1.set(0*MHz)
        self.sawg0.frequency2.set(0*MHz)
        self.sawg0.amplitude1.set(0.)
        self.sawg0.amplitude2.set(0.)
        self.sawg1.config.set_clr(1, 1, 1)
        self.sawg1.frequency0.set(f)
        self.sawg1.frequency1.set(0*MHz)
        self.sawg1.frequency2.set(0*MHz)
        self.sawg1.amplitude1.set(0.)
        self.sawg1.amplitude2.set(0.)
        delay(100 * us)

        while True:
            self.ttl_sma.pulse(3*us)
            delay(1.0*us)

            with parallel:
                self.ttl_sma.pulse(tpi/2)
                self.sawg0_lc_p0_set(0.)
                self.sawg1_lc_p0_set(0.)
                self.sawg0_lc_a1_set(a0)
                self.sawg1_lc_a1_set(a0)
            self.sawg1_lc_a1_set(0.)

            self.sawg1.config.set_clr(0, 0, 0)

            delay(tarm)

            self.sawg1_lc_p0_set(0.5)
            self.sawg1_lc_a1_set(a0)
            self.ttl_sma.pulse(tpi)
            self.sawg1_lc_a1_set(0.0)

            delay(tarm)

            self.sawg1_lc_p0_set(0.0)
            self.sawg1_lc_a1_set(a0)
            self.ttl_sma.pulse(tpi/2)
            self.sawg1_lc_a1_set(0.0)

            delay(1000*us)
```
swag0 is yellow on scope
swag1 is green on scope
TTL is pink on scope -- trigger on pulse duration > 2.8 us

Scope field of view zoomed to rising edge of first pi/2 pulse. Scope shows unexpected phase offset between sawg0 and sawg1. 

[image]

Secondarily, this exemplifies annoying latency compensation that's needed to align sawg with TTL. To be fully general there's quite a zoo of methods that need to be trivially wrapped. Is there a better way of doing this??

jordens, 2017-06-12:
You are clearing the accu on one channel but not on the other.

jbqubit, 2017-06-12:
> You are clearing the accu on one channel but not on the other.
I expect that the following is setting the phase register of both channels at the same time. What code should I be using?  
```
            with parallel:
                self.ttl_sma.pulse(tpi/2)
                self.sawg0_lc_p0_set(0.)
                self.sawg1_lc_p0_set(0.)
```

> You can latency-compensate the TTL instead.
I'm treating TTL as the timing reference plane. 

Doesn't this ad hoc solution wrinkle your sensitive clean-design nose?

sbourdeauducq, 2017-06-13:
@jbqubit Regarding TTL/SAWG latency matching, you are welcome to offer a constructive comment and/or funding in #40. Sure, it is a feature we think is desirable, otherwise #40 would not be open.

jordens, 2017-06-13:
Please read the current documentation of SAWG (again). Are you satisfied that you have understood the way the DDS/DUC and the phase accumulator clear work? Especially in the light of #744 and #745...

jordens, 2017-06-13:
```python
...

        self.sawg0.config.set_clr(1, 0, 0)
        self.sawg0.phase0.set(0.)
        self.sawg0.frequency0.set(f)
        self.sawg0.frequency1.set(0.)

        self.sawg1.config.set_clr(1, 0, 0)
        self.sawg1.phase0.set(0.)
        self.sawg1.frequency0.set(f)
        self.sawg1.frequency1.set(0.)
        delay(100 * us)
        self.sawg0.config.set_clr(0, 0, 0)
        self.sawg1.config.set_clr(0, 0, 0)

        while True:
            self.raman(a0, tpi/2)
            delay(tarm)
            self.sawg1.phase0.set(.5)
            self.raman(a0, tpi)
            self.sawg1.phase0.set(0.)
            delay(tarm)
            self.raman(a0, tpi/2)
            delay(1000*us)

    def raman(self, a, t):
            self.sawg0.amplitude1.set(a)
            self.sawg1.amplitude1.set(a)
            self.ttl_sma.pulse(t)
            self.sawg0.amplitude1.set(0.)
            self.sawg1.amplitude1.set(0.)
```

But there are several other schemes that would do these kind of pulses.

I suspect you have questions about functionality. Just ask.

jbqubit, 2017-06-27:
OK. I see that in your sawg API sawgN.xxx.set() does not advance the timeline. Therefore ```with parallel``` is not needed to achieve phase alignment. Please add a comment in the documentation to state this clearly. This is different than with the DDSs which used ```with parallel```.

I've tested again and find that that phase alignment does not work as expected for sawgN on the same device. Here are two examples. 

```
$ artiq_corelog
[           7us]  INFO(runtime): ARTIQ runtime starting...
[        3855us]  INFO(runtime): software version 3.0.dev+1173.g07f5e991
[       10204us]  INFO(runtime): gateware version 3.0.dev+1173.g07f5e991
```

# Example 1
```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    purpose: test absolute alignment of frequency0
    test: sawg0 and sawg1 should be phase aligned
    expectation: no phase offset between sawg0 and sawg1
    observation: 1.7 ns delay between rising edges -- FAIL
    """
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")

        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        delay(300 * us)
        self.sawg0.reset()
        self.sawg1.reset()
        self.ttl_sma.output()
        delay(1 * ms)

        f0 = 50*MHz
        a0 = 0.5

        self.sawg0.amplitude1.set(a0)
        self.sawg0.frequency0.set(f0)
        self.sawg0.phase0.set(0.)
        self.sawg1.amplitude1.set(a0)
        self.sawg1.frequency0.set(f0)
        self.sawg1.phase0.set(0.)

        delay(1*ms)
```
[image]

# Example 2
```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    purpose: test absolute alignment of sawg0.frequency0 and sawg1.frequency1
    test: make sawg0 and sawg1 phase aligned
    expectation: no phase offset between sawg0 and sawg1
    observation: 9 ns delay between rising edges -- FAIL
    """
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")

        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        delay(300 * us)
        self.sawg0.reset()
        self.sawg1.reset()
        self.ttl_sma.output()
        delay(1 * ms)

        f0 = 50*MHz
        a0 = 0.5

        self.sawg0.amplitude1.set(a0)
        self.sawg0.frequency0.set(f0)
        self.sawg0.phase0.set(0.)
        self.sawg1.amplitude1.set(a0)
        self.sawg1.frequency1.set(f0)
        self.sawg1.phase1.set(0.)

        delay(1*ms)
```

jordens, 2017-06-28:
You can still use `with parallel`. Nobody is stopping you. And we generally try avoid describing all the things the API does not. Please file an issue if you are think the DDS docs are incomplete.
Is that 1.7 ns repeatable? Are the electrical paths matched? The second one might involve a bug. I'll have a look.

jbqubit, 2017-06-28:
> Are the electrical paths matched?
I was using a pair of MiniCircuits power splitters which are nominally identical but which are responsible for the 1.7 ns differential delay. So, Example 1 now works as expected with power splitters removed. 

With power splitters removed, Example 2 still has a delay (below).  

[image]

jordens, 2017-06-28:
I take it that you had mismatched electrical lengths then.

Please answer my other question as well.

That delay in your second case is an intrinsic property of the design (simple cordic + HBF has one (sub-)sample more latency than the parallelized cordic). I'll document it. After all you are comparing the DUC phase with the low-f oscillators. That's not comparing equivalent things and delay matching here is decidedly not required for the envisioned use cases.

I can see a way to actually make them the same latency but it will increase resource usage and increase the overall latency. And again, I don't see the use case.

jordens, 2017-06-28:
I found an easy approach that does not add resources and only marginally increases latency. Should be coming right up.

jbqubit, 2017-06-28:
>> Is that 1.7 ns repeatable? 
> I take it that you had mismatched electrical lengths then.
Yes.

> Are the electrical paths matched?
They are now matched. 

>I found an easy approach that does not add resources and only marginally increases latency. Should be coming right up.

Please let me know when the conda artiq-kc705-phaser binary is available and I will test.

commit f2632e0f (2017-06-28): sawg: adapt latency to fir changes

closes #748
closed 2017-06-28 by jordens