m-labs/artiq #749: sawg: asymmetric sinusoid duty cycle

jbqubit, 2017-06-12:
Example below is a mockup of a spin-echo where the TTL is used to switch on/off an RF switch. There is an unexpected change in duty cycle from 50% high to 75% high for sawg1.

```
$ cat test_latency2.py

from artiq.experiment import *

sawg_latency_mu = 988

class SAWGTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")
        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def sawg0_lc_p0_set(self, p):
        delay_mu(-1*sawg_latency_mu)
        self.sawg0.phase0.set(p)
        delay_mu(sawg_latency_mu)

    @kernel
    def sawg1_lc_p0_set(self, p):
        delay_mu(-1*sawg_latency_mu)
        self.sawg1.phase0.set(p)
        delay_mu(sawg_latency_mu)

    @kernel
    def sawg0_lc_a1_set(self, a):
        delay_mu(-1*sawg_latency_mu)
        self.sawg0.amplitude1.set(a)
        delay_mu(sawg_latency_mu)

    @kernel
    def sawg1_lc_a1_set(self, a):
        delay_mu(-1*sawg_latency_mu)
        self.sawg1.amplitude1.set(a)
        delay_mu(sawg_latency_mu)

    @kernel
    def run(self):
        self.core.reset()
        self.ttl_sma.output()
        delay(5*ms)

        # prepare
        tpi = 1*us
        tarm = 2*us
        a0 = 0.5
        f = 200*MHz
        self.sawg0.config.set_clr(1, 1, 1)
        self.sawg0.frequency0.set(f)
        self.sawg0.frequency1.set(0*MHz)
        self.sawg0.frequency2.set(0*MHz)
        self.sawg0.amplitude1.set(0.)
        self.sawg0.amplitude2.set(0.)
        self.sawg1.config.set_clr(1, 1, 1)
        self.sawg1.frequency0.set(f)
        self.sawg1.frequency1.set(0*MHz)
        self.sawg1.frequency2.set(0*MHz)
        self.sawg1.amplitude1.set(0.)
        self.sawg1.amplitude2.set(0.)
        delay(100 * us)

        while True:
            self.ttl_sma.pulse(3*us)
            delay(1.0*us)

            with parallel:
                self.ttl_sma.pulse(tpi/2)
                self.sawg0_lc_p0_set(0.)
                self.sawg1_lc_p0_set(0.)
                self.sawg0_lc_a1_set(a0)
                self.sawg1_lc_a1_set(a0)
            self.sawg1_lc_a1_set(0.)

            self.sawg1.config.set_clr(0, 0, 0)

            delay(tarm)

            self.sawg1_lc_p0_set(0.5)
            self.sawg1_lc_a1_set(a0)
            self.ttl_sma.pulse(tpi)
            self.sawg1_lc_a1_set(0.0)

            delay(tarm)

            self.sawg1_lc_p0_set(0.0)
            self.sawg1_lc_a1_set(a0)
            self.ttl_sma.pulse(tpi/2)
            self.sawg1_lc_a1_set(0.0)

            delay(1000*us)
```
swag0 is yellow on scope
swag1 is green on scope
TTL is pink on scope -- trigger on pulse duration > 2.8 us

Scope field of view zoomed to rising edge of pi pulse. Scope shows There is an unexpected change in duty cycle from 50% high to 75% high for sawg1.

[image]

jordens, 2017-06-12:
Duty cycle? You are running the DUC at 200 MHz frequency with a 300 MHz data rate, i.e. 100 MHz output as expected. The initial cycle is the AA filter trying its best to help you band-limit your signal.

jbqubit, 2017-06-12:
By DUC I guess you mean digital up converter. 
By AA I guess you mean anti-aliasing filter. 

So this gives a time-average DC offset. If driving an AOM this won't be a problem. Looking at it on a spectrum analyzer. 
- 200 MHz is -15 dBm
- 300 MHz peak is -72 dBm
- 400 MHz peak is -21 dBm 

I will double check the filter attenuation at 400 MHz.
- https://github.com/m-labs/sinara/issues/108 
- github/m-labs/sinara/ARTIQ_ALTIUM/PCB_mezzanine_analog_allaki/Docs/Filters/DAC_Filter_A/

jbqubit, 2017-06-12:
The DAC_Filter_A filter in the Sinara repository is a single-pole Butterworth. Insertion loss is 
- -3 dB at 300 MHz
- -8.5 dB at 400 MHz
- TODO: update to higher-order 

Otherwise Issue as raised is a non-issue. Closing.

closed 2017-06-12 by jbqubit

jordens, 2017-06-13:
One of the Nyquist images is at 0 Hz.
Expect a sinc response.