m-labs/artiq #750: sawg: phaser documentation

jbqubit, 2017-06-13:
Some updates to swag documentation would be helpful. 

# Features
> The code can be reconfigured. Possible example configurations are: support 2 channels at 1 GHz datarate, support 4 channels at 300 MHz data rate, no interpolation, and using mix mode to stress the second and third Nyquist zones (150-300 MHz and 300-450 MHz).

Please note location in source to choose among these configurations prior to build. 

# README_PHASER.rst#Setup

>A 300 MHz clock of roughly 10 dBm (0.2 to 3.4 V peak-to-peak into 50 Ohm) must be connected to the AD9154-FMC-EBZ J1. The input is 50 Ohm terminated. The RTIO clock, DAC deviceclock, FPGA deviceclock, and SYSREF are derived from this signal.

Please state what RTIO clock, DAC device clock, FPGA device clock and SYSREF division ratios are for this demonstration. 

# README_PHASER.rst#Setup
> Compile the ARTIQ Phaser bitstream, bios, and runtime (c.f. ARTIQ manual):

Recommend updating to tell user they also have option to use pre-built conda-distributed binaries. And that they're located here.

~/anaconda3/envs/my_env/lib/python3.5/site-packages/artiq/binaries/kc705-phaser 

# SWAG and Coarse RTIO
Coarse RTIO is mentioned briefly in core_drivers. Please link back to the core drivers reference in SWAG doc. 

https://m-labs.hk/artiq/manual-release-2/core_drivers_reference.html#artiq.coredevice.exceptions.RTIOCollision

# SWAG diagram
A high-level diagram illustrating DSP blocks comprising two-channel SWAG would be helpful.

# SWAG doc error
In core_drivers_reference.html#artiq.coredevice.sawg.SAWG this passage is wrong.
> They can represent frequencies within their first Nyquist zone from -f_RTIO/2 to f_RTIO/2.

S

closed 2017-06-14 by jordens

jbqubit, 2017-06-14:
Regarding https://github.com/m-labs/artiq/commit/fecc42fd0c49c3b3735e67aacd35fd18fb7e5709 . Thank you for the additional docs. 

> The code can be reconfigured. Possible example configurations are: support 2 channels at 1 GHz datarate, support 4 channels at 300 MHz data rate, no interpolation, and using mix mode to stress the second and third Nyquist zones (150-300 MHz and 300-450 MHz).

>> Please contact M-Labs if you need help with this.

I'd like to test this. Please tell me which file I should take a look at to get started. 

> From time to time and on request there may be pre-built binaries in the
``artiq-kc705-phaser`` package on the M-Labs conda package label.

Telling the build bot to build a package on IRC...  
bb-m-labs: force build --props=package=artiq-kc705-phaser artiq-board 

> * The RTIO coarse clock (the rate of the RTIO timestamp counter) is 150 MHz. The RTIO ``ref_period`` is 1/150 MHz = 5ns/6. The RTIO ``ref_multiplier`` is ``1``. C.f. ``device_db.py`` for both variables. The JED204B DAC data rate and DAC device clock are both 300 MHz. The JESD204B line rate is 6 GHz.

Thank you. This is helpful. 

> A high-level diagram illustrating DSP blocks comprising two-channel SWAG would be helpful.

Do you disagree? 

> In core_drivers_reference.html#artiq.coredevice.sawg.SAWG this passage is wrong.

Thank you for fixing this.

jordens, 2017-06-14:
https://github.com/m-labs/artiq/blob/master/artiq/gateware/dsp/sawg.py https://github.com/m-labs/artiq/blob/master/artiq/gateware/targets/phaser.py https://github.com/m-labs/artiq/blob/master/artiq/firmware/libboard/ad9154.rs, https://github.com/m-labs/artiq/blob/master/artiq/firmware/libboard/ad9516.rs and a couple more files. But those should get you started. Then the datasheets for the chips.

I don't disagree. There are many things that would be helpful.

commit fecc42fd (2017-06-14): sawg/phaser: expand documentation (closes #750)