m-labs/artiq #751: sawg: can't produce simple sinusoid

jbqubit, 2017-06-15:
Latest gateware can't produce simple sinusoids. 

```
 conda list artiq*
# packages in environment at /home/britton/anaconda3/envs/artiq3test4:
#
artiq                     3.0.dev         py_1126+git10fb6c62    m-labs/label/dev
artiq                     3.0.dev0+1075.gfdb4e3f8c           <pip>
artiq-kc705-phaser        3.0.dev         py_1126+git10fb6c62    m-labs/label/dev
llvmlite-artiq            0.12.0.dev              py35_30    m-labs/label/main
```

```
# cat test_simple_sin.py

from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    Test output of simple sinusoids.
    """
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")
        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        self.ttl_sma.output()
        delay(1*ms)

        while True:
            self.ttl_sma.pulse(3 * us)
            self.test()
            delay(1*ms)

    @kernel
    def test(self):
        # test: t1
        f0 = 5*MHz
        t = 1*us
        a0 = 0.1

        self.sawg0.frequency1.set(f0)
        self.sawg0.amplitude1.set(a0)
        self.sawg1.frequency1.set(f0)
        self.sawg1.amplitude1.set(a0)
```

[image]

jbqubit, 2017-06-15:
OK. I see what's happening. The sawg configuration persists from one experiment to the next. Zeroing the other oscillator frequencies permits generation of the expected output. 

```
        self.sawg0.frequency0.set(0.0*MHz)
        self.sawg0.frequency1.set(f0)
        self.sawg0.amplitude1.set(a0)
        self.sawg0.frequency2.set(0.0 * MHz)
        self.sawg1.frequency0.set(0.0 * MHz)
        self.sawg1.frequency1.set(f0)
        self.sawg1.amplitude1.set(a0)
        self.sawg1.amplitude2.set(a0)
```

I'm changing the title of this Issue to reflect this observation. It would be helpful to have a sawg.reset() function to zero out all the registers. A single RTIO command for this would be faster too.

sbourdeauducq, 2017-06-16:
Shouldn't ``core.reset()`` reset the SAWG state?

jordens, 2017-06-16:
Yes. I assume he meant fine grained per-channel reset control.

jbqubit, 2017-06-16:
I don't see that ```core.reset()``` resets SAWG state. Example follows.

```
$ cat test_simple_sin.py

from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    Test output of simple sinusoids.
    """
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")
        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def sawg_reset(self):
        self.sawg0.frequency0.set(0.0)
        self.sawg0.frequency0.set(0.0)
        self.sawg0.phase0.set(0.0)
        self.sawg0.amplitude1.set(0.0)
        self.sawg0.frequency1.set(0.0)
        self.sawg0.phase1.set(0.0)
        self.sawg0.amplitude2.set(0.0)
        self.sawg0.frequency2.set(0.0)
        self.sawg0.phase2.set(0.0)
        self.sawg0.config.set_clr(1, 1, 1)

        self.sawg1.frequency0.set(0.0)
        self.sawg1.frequency0.set(0.0)
        self.sawg1.phase0.set(0.0)
        self.sawg1.amplitude1.set(0.0)
        self.sawg1.frequency1.set(0.0)
        self.sawg1.phase1.set(0.0)
        self.sawg1.amplitude2.set(0.0)
        self.sawg1.frequency2.set(0.0)
        self.sawg1.phase2.set(0.0)
        self.sawg1.config.set_clr(1, 1, 1)

    @kernel
    def run(self):
        self.core.reset()
        self.ttl_sma.output()
        delay(5 * ms)
        self.sawg_reset()

        while True:
            self.ttl_sma.pulse(3 * us)
            self.test()
            delay(1*ms)

    @kernel
    def test(self):
        # test: t1
        f0 = 10*MHz
        t = 1*us
        a0 = 0.5

        self.sawg0.frequency1.set(f0)
        self.sawg0.amplitude1.set(a0)
        self.sawg1.frequency1.set(f0)
        self.sawg1.amplitude1.set(a0)
```
While program is running I see sinusoid on scope for sawg1. As expected. 

Once program is terminated I continue to see sinusoid on scope for sawg1. So sawg1 PHY is continuing to operate. As expected 

Now I run the following program. 
```
$ cat test_reset.py

from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    Test output of simple sinusoids.
    """
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        self.core.reset()
```
Once artiq_run terminates I still see sinusoid on scope for sawg1.

jordens, 2017-06-16:
Currently, we reset a couple things on `core.reset()`:
* TTL input sensitivity
* RTIO2WB
* DDS interface RTIO side

but not others:
* TTL clock gen
* TTL output state and oe
* SPI core
* DDS interface DDS side

This is consistent with the documentation.

It would also mean (at least to me) that the "output side" of the SAWG should not reset on `core.reset()`, only the communication side and data state machinery (but there is none). We would need to expose `reset_phy` separately.

jbqubit, 2017-06-16:
This doesn't change the fact that the implementation lacks an adequate reset mechanism for swag. Establishing a well known starting state is crucial. Some reasons. 
- experiments can be scheduled in any order
- debugging of sawg is difficult

jordens, 2017-06-16:
Don't worry Joe. That comment was not a response to you.

closed 2017-06-16 by jordens

jbqubit, 2017-06-16:
Please don't close the Issue until I've tested it.

jordens, 2017-06-16:
Please don't try to dictate our issue workflow. We have explained the procedure to you multiple times.

jbqubit, 2017-06-19:
observation: sawg.reset() breaks swag

```
$ cat test_simple_sin1.py

from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    purpose: test sinusoidal output
    expectation:
      10 MHz from CORDIC frequency1 on sawg0 and sawg1
      playback continues after program termination
      noise
    """
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")
        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def sawg_reset(self):
        self.sawg0.frequency0.set(0.0)
        self.sawg0.frequency0.set(0.0)
        self.sawg0.phase0.set(0.0)
        self.sawg0.amplitude1.set(0.0)
        self.sawg0.frequency1.set(0.0)
        self.sawg0.phase1.set(0.0)
        self.sawg0.amplitude2.set(0.0)
        self.sawg0.frequency2.set(0.0)
        self.sawg0.phase2.set(0.0)
        self.sawg0.config.set_clr(1, 1, 1)

        self.sawg1.frequency0.set(0.0)
        self.sawg1.frequency0.set(0.0)
        self.sawg1.phase0.set(0.0)
        self.sawg1.amplitude1.set(0.0)
        self.sawg1.frequency1.set(0.0)
        self.sawg1.phase1.set(0.0)
        self.sawg1.amplitude2.set(0.0)
        self.sawg1.frequency2.set(0.0)
        self.sawg1.phase2.set(0.0)
        self.sawg1.config.set_clr(1, 1, 1)

    @kernel
    def run(self):
        self.core.reset()
        self.ttl_sma.output()
        self.core.break_realtime()
        delay(1 * ms)
        self.sawg_reset()

        f0 = 10*MHz
        a0 = 0.5

        self.sawg0.frequency1.set(f0)
        self.sawg0.amplitude1.set(a0)
        self.sawg1.frequency1.set(f0)
        self.sawg1.amplitude1.set(a0)
```

```
$ cat test_reset.py

from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    purpose: test sawg.reset()
    expectation:
      waveform stops after program termination

    """
    def build(self):
        self.setattr_device("core")
        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.break_realtime()
        delay(1*ms)
        self.sawg0.reset()
        self.sawg1.reset()
```

Steps to reproduce: 
1. reset KC705
2. artiq_run test_simple_sin1.py -- observe signal on scope (but it's ugly #752)
3. artiq_run test_reset.py -- observe signal on scope disappears
4. artiq_run test_simple_sin1.py -- no signal on scope

commit b5772f47 (2017-06-16): sawg: add channel reset (closes #751)