m-labs/artiq #752: sawg: harmonics > -10 dBc for frequency1.set(10*MHz)

jbqubit, 2017-06-16:
I can't seem to get sawg to produce clean sinusoids at low frequencies. This is new in the last couple days. When run the program below I get

10 MHz -4 dBm
20 MHz -33 dBm
30 MHz -13 dBm
40 MHz -47 dBm
50 MHz -18 dBm

I expect the spectrum to be free of harmonics out to f_rtio_coarse/2.

```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    Test output of simple sinusoids.
    """
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")
        self.setattr_device("sawg0")
        self.setattr_device("sawg1")
        #self.sawg_reset = getattr(self, 'sawg_reset')

    @kernel
    def sawg_reset(self):
        self.sawg0.frequency0.set(0.0)
        self.sawg0.frequency0.set(0.0)
        self.sawg0.phase0.set(0.0)
        self.sawg0.amplitude1.set(0.0)
        self.sawg0.frequency1.set(0.0)
        self.sawg0.phase1.set(0.0)
        self.sawg0.amplitude2.set(0.0)
        self.sawg0.frequency2.set(0.0)
        self.sawg0.phase2.set(0.0)
        self.sawg0.config.set_clr(1, 1, 1)

        self.sawg1.frequency0.set(0.0)
        self.sawg1.frequency0.set(0.0)
        self.sawg1.phase0.set(0.0)
        self.sawg1.amplitude1.set(0.0)
        self.sawg1.frequency1.set(0.0)
        self.sawg1.phase1.set(0.0)
        self.sawg1.amplitude2.set(0.0)
        self.sawg1.frequency2.set(0.0)
        self.sawg1.phase2.set(0.0)
        self.sawg1.config.set_clr(1, 1, 1)

    @kernel
    def run(self):
        self.core.reset()
        self.ttl_sma.output()
        delay(5 * ms)
        self.sawg_reset()

        while True:
            self.ttl_sma.pulse(3 * us)
            self.test()
            delay(1*ms)

    @kernel
    def test(self):
        # test: t1
        f0 = 10*MHz
        t = 1*us
        a0 = 0.1

        self.sawg0.frequency1.set(f0)
        self.sawg0.amplitude1.set(a0)
        self.sawg1.frequency1.set(f0)
        self.sawg1.amplitude1.set(a0)
```

```
$ conda list artiq
# packages in environment at /home/britton/anaconda3/envs/artiq3test4:
#
artiq                     3.0.dev         py_1126+git10fb6c62    m-labs/label/dev
artiq                     3.0.dev0+1075.gfdb4e3f8c           <pip>
artiq-kc705-phaser        3.0.dev         py_1126+git10fb6c62    m-labs/label/dev
llvmlite-artiq            0.12.0.dev              py35_30    m-labs/label/main
```

jordens, 2017-06-16:
* The spectrum will never be completely "free of harmonics out to f_rtio_coarse/2".
* The data out of SAWG in gateware simulation for this experiment is squeaky clean, with harmonics below -80 dBc and the Nyquist image at 290 MHz at -60 dBc.
* There is a 8 MHz lower limit on the balun. I don't know how it behaves towards that.

* You are certain this was not the case with `3.0.dev-py_1092+gitebb46602`?
* Please do that same measurement with that older version.
* A -9 dBc third harmonic would be easily visible on a scope. Is it?
* Please test this with the DUC upconverting it to somewhere else, e.g. `self.sawg0.frequency0.set(25*MHz)`.

jordens, 2017-06-16:
[image]

jbqubit, 2017-06-16:
Roll back to previous version. 
```
$ conda install artiq-kc705-phaser=3.0.dev=py_1092+gitebb46602
Fetching package metadata ...............
Solving package specifications: .

Package plan for installation in environment /home/britton/anaconda3/envs/artiq3test4:

The following NEW packages will be INSTALLED:

    artiq:              3.0.dev-py_1092+gitebb46602 m-labs/label/dev 
    artiq-kc705-phaser: 3.0.dev-py_1092+gitebb46602 m-labs/label/dev 
    pythonparser:       1.0-py_8                    m-labs/label/main
```
Flash & reset KC705. 
```
(artiq3test4) britton@britton1:~/anaconda3/envs/artiq3test4/lib/python3.5/site-packages/artiq/binaries/kc705-phaser$ openocd –s /home/britton/anaconda3/envs/artiq3test4/share/openocd/scripts -f board/kc705.cfg -c "init; jtagspi_init 0 bscan_spi_xc7k325t.bit; jtagspi_program top.bit 0x000000; jtagspi_program bios.bin 0xaf0000; jtagspi_program runtime.fbi 0xb00000; jtagspi_program /home/britton/gitlab/jbqubit/artiqlab/phaser_config.bin 0xb80000; xc7_program xc7.tap; exit"
```
This build doesn't respond to ping. I think it's the one that's foobar'd. Try earlier build.   
```
(artiq3test4) britton@britton1:~/gitlab/jbqubit/artiqlab$ conda search artiq-kc705-phaserFetching package metadata ...............
artiq-kc705-phaser           3.0.dev         py_33+git55b2fc4  m-labs/label/dev 
                             3.0.dev         py_35+git09f46da  m-labs/label/dev 
                             3.0.dev         py_38+git1feb7cf  m-labs/label/dev 
                             3.0.dev         py_88+git0472ab6  m-labs/label/dev 
                             3.0.dev         py_160+git062aca2  m-labs/label/dev 
                             3.0.dev         py_288+git2e48250  m-labs/label/dev 
                             3.0.dev         py_303+git0ee47e7  m-labs/label/dev 
                             3.0.dev         py_426+gitf6071a5  m-labs/label/dev 
                             3.0.dev         py_634+gitcfb6611  m-labs/label/dev 
                             3.0.dev         py_635+gitf5f6622  m-labs/label/dev 
                             3.0.dev         py_1075+gitfdb4e3f8cfad  m-labs/label/dev 
                          *  3.0.dev         py_1092+gitebb46602  m-labs/label/dev 
                             3.0.dev         py_1126+git10fb6c62  m-labs/label/dev 
(artiq3test4) britton@britton1:~/gitlab/jbqubit/artiqlab$ conda install artiq-kc705-phaser=3.0.dev=py_1075+gitfdb4e3f8cfad
Fetching package metadata ...............
Solving package specifications: .

Package plan for installation in environment /home/britton/anaconda3/envs/artiq3test4:

The following packages will be DOWNGRADED:

    artiq:              3.0.dev-py_1092+gitebb46602 m-labs/label/dev --> 3.0.dev-py_1075+gitfdb4e3f8cfad m-labs/label/dev
    artiq-kc705-phaser: 3.0.dev-py_1092+gitebb46602 m-labs/label/dev --> 3.0.dev-py_1075+gitfdb4e3f8cfad m-labs/label/dev
```
OK can ping this build.

10 MHz tone is beautiful. 
- -65 dBc at 140 MHz
- -30 dBc at 290 MHz

jordens, 2017-06-16:
An oscilloscope shot would help a lot. Then try again with b5772f47 (and properly initializing the channel).

jbqubit, 2017-06-16:
```
(artiq3test4) britton@britton1:~/gitlab/jbqubit/artiqlab$ conda install artiq-kc705-phaser=3.0.dev=py_1126+git10fb6c62
Fetching package metadata ...............
Solving package specifications: .

Package plan for installation in environment /home/britton/anaconda3/envs/artiq3test4:

The following packages will be UPDATED:

    artiq:              3.0.dev-py_1075+gitfdb4e3f8cfad m-labs/label/dev  --> 3.0.dev-py_1126+git10fb6c62 m-labs/label/dev 
    artiq-kc705-phaser: 3.0.dev-py_1075+gitfdb4e3f8cfad m-labs/label/dev  --> 3.0.dev-py_1126+git10fb6c62 m-labs/label/dev 
    pythonparser:       1.0-py_8                        m-labs/label/main --> 1.1-py_8                    m-labs/label/main
```

```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    Test output of simple sinusoids.
    """
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")
        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def sawg_reset(self):
        self.sawg0.frequency0.set(0.0)
        self.sawg0.frequency0.set(0.0)
        self.sawg0.phase0.set(0.0)
        self.sawg0.amplitude1.set(0.0)
        self.sawg0.frequency1.set(0.0)
        self.sawg0.phase1.set(0.0)
        self.sawg0.amplitude2.set(0.0)
        self.sawg0.frequency2.set(0.0)
        self.sawg0.phase2.set(0.0)
        self.sawg0.config.set_clr(1, 1, 1)

        self.sawg1.frequency0.set(0.0)
        self.sawg1.frequency0.set(0.0)
        self.sawg1.phase0.set(0.0)
        self.sawg1.amplitude1.set(0.0)
        self.sawg1.frequency1.set(0.0)
        self.sawg1.phase1.set(0.0)
        self.sawg1.amplitude2.set(0.0)
        self.sawg1.frequency2.set(0.0)
        self.sawg1.phase2.set(0.0)
        self.sawg1.config.set_clr(1, 1, 1)

    @kernel
    def run(self):
        self.core.reset()
        self.ttl_sma.output()
        delay(5 * ms)
        self.sawg_reset()

        f0 = 10*MHz
        a0 = 0.5

        self.sawg0.frequency1.set(f0)
        self.sawg0.amplitude1.set(a0)
        self.sawg1.frequency1.set(f0)
        self.sawg1.amplitude1.set(a0)

```

[image]
[image]

jbqubit, 2017-06-19:
@jordens What's the status on this?

jordens, 2017-06-19:
Joe, the status is waiting. Quote from above:
* Try again with b5772f4 (and properly initializing the channel).
* Please test this with the DUC upconverting it to somewhere else, e.g. `self.sawg0.frequency0.set(25*MHz)`.

jbqubit, 2017-06-19:
I see. I tested and posted screen shots for  
```3.0.dev-py_1126+git10fb6c62``` which predated  b5772f4. 

Since the build for  b5772f4 failed I can't do verbatim as you ask. I'll use the most recent build. 

```
$ conda search artiq-kc705-phaser
Fetching package metadata ...............
artiq-kc705-phaser           3.0.dev         py_33+git55b2fc4  m-labs/label/dev 
                             3.0.dev         py_35+git09f46da  m-labs/label/dev 
                             3.0.dev         py_38+git1feb7cf  m-labs/label/dev 
                             3.0.dev         py_88+git0472ab6  m-labs/label/dev 
                             3.0.dev         py_160+git062aca2  m-labs/label/dev 
                             3.0.dev         py_288+git2e48250  m-labs/label/dev 
                             3.0.dev         py_303+git0ee47e7  m-labs/label/dev 
                             3.0.dev         py_426+gitf6071a5  m-labs/label/dev 
                             3.0.dev         py_634+gitcfb6611  m-labs/label/dev 
                             3.0.dev         py_635+gitf5f6622  m-labs/label/dev 
                             3.0.dev         py_1075+gitfdb4e3f8cfad  m-labs/label/dev 
                             3.0.dev         py_1092+gitebb46602  m-labs/label/dev 
                             3.0.dev         py_1126+git10fb6c62  m-labs/label/dev 
                             3.0.dev         py_1132+git424b2bfb  m-labs/label/dev 
                          *  3.0.dev         py_1138+git09d198c7  m-labs/label/dev 
(artiq3test4) britton@britton1:~/gitlab/jbqubit/artiqlab$ conda list artiq
# packages in environment at /home/britton/anaconda3/envs/artiq3test4:
#
artiq                     3.0.dev         py_1138+git09d198c7    m-labs/label/dev
artiq                     3.0.dev0+1075.gfdb4e3f8c           <pip>
artiq-kc705-phaser        3.0.dev         py_1138+git09d198c7    m-labs/label/dev
llvmlite-artiq            0.12.0.dev              py35_30    m-labs/label/main

```
I observe the same harmonics.

jordens, 2017-06-19:
Thanks. Testing the current version fine.

* Is this using `sawgX.reset()` to initialize the channels?
* Can I assume you don't have an `idle_kernel` or `startup_kernel` set?

jbqubit, 2017-06-19:
This is not using sawgx.reset(). See #751.

```
(artiq3test4) britton@britton1:~/gitlab/jbqubit/artiqlab$ artiq_corelog
[           7us]  INFO(runtime): ARTIQ runtime starting...
[        3851us]  INFO(runtime): software version 3.0.dev+1138.g09d198c7
[       10200us]  INFO(runtime): gateware version 3.0.dev+1138.g09d198c7
[       16533us]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     1016006us]  INFO(runtime): continuing boot
[     1055877us]  INFO(runtime): using MAC address 00-0a-35-03-1e-53
[     1060704us]  INFO(runtime): using IP address 192.168.1.71
[     1066211us] ERROR(runtime::rtio_mgt): unrecognized startup_clock configuration entry
[     1073959us]  INFO(runtime::rtio_mgt): startup RTIO clock: Internal
[     1080511us]  INFO(runtime): UART log level set to INFO by default
[     1087040us]  INFO(runtime::mgmt): management interface active
[     1100731us]  INFO(runtime::session): accepting network sessions
[     1114565us]  INFO(runtime::session): running startup kernel
[     1119026us]  INFO(runtime::session): no startup kernel found
[     1124715us]  INFO(runtime::session): no connection, starting idle kernel
[     1131585us]  INFO(runtime::session): no idle kernel found
[     3945145us]  INFO(runtime::session): new connection from 192.168.1.68:53394
[     3955940us]  WARN(runtime): network error: dropped by socket
[     3960932us]  WARN(runtime): network error: dropped by socket
[     4205562us]  WARN(runtime): network error: dropped by socket
[     4210185us]  WARN(runtime): network error: dropped by socket
[     4733703us]  INFO(runtime::kern_hwreq): resetting RTIO
[     4739349us]  INFO(runtime::session): no connection, starting idle kernel
[     4745088us]  INFO(runtime::session): no idle kernel found
[     8035064us]  INFO(runtime::mgmt): new connection from 192.168.1.68:34223
```

jbqubit, 2017-06-19:
I tested again this afternoon. The output still includes enormous harmonics. Can you reproduce on your hardware? 

```
(artiq3test4) britton@britton1:~/gitlab/jbqubit/artiqlab$ artiq_corelog
[           7us]  INFO(runtime): ARTIQ runtime starting...
[        3851us]  INFO(runtime): software version 3.0.dev+1138.g09d198c7
[       10200us]  INFO(runtime): gateware version 3.0.dev+1138.g09d198c7
[       16533us]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     1016006us]  INFO(runtime): continuing boot
[     1058376us]  INFO(runtime): using MAC address 00-0a-35-03-1e-53
[     1063202us]  INFO(runtime): using IP address 192.168.1.71
[     1068710us] ERROR(runtime::rtio_mgt): unrecognized startup_clock configuration entry
[     1076458us]  INFO(runtime::rtio_mgt): startup RTIO clock: Internal
[     1083010us]  INFO(runtime): UART log level set to INFO by default
[     1089539us]  INFO(runtime::mgmt): management interface active
[     1103230us]  INFO(runtime::session): accepting network sessions
[     1117064us]  INFO(runtime::session): running startup kernel
[     1121525us]  INFO(runtime::session): no startup kernel found
[     1127214us]  INFO(runtime::session): no connection, starting idle kernel
[     1134083us]  INFO(runtime::session): no idle kernel found
[     6993160us]  INFO(runtime::session): new connection from 192.168.1.68:54277
[     7004225us]  WARN(runtime): network error: dropped by socket
[     7009068us]  WARN(runtime): network error: dropped by socket
[     7255307us]  WARN(runtime): network error: dropped by socket
[     7259930us]  WARN(runtime): network error: dropped by socket
[     7783667us]  INFO(runtime::kern_hwreq): resetting RTIO
[     7789583us]  INFO(runtime::session): no connection, starting idle kernel
[     7795321us]  INFO(runtime::session): no idle kernel found
[   109483627us]  INFO(runtime::mgmt): new connection from 192.168.1.68:35108
```

jbqubit, 2017-06-20:
@jordens Do you have what you need from me for this? Label is still state:waiting.

jordens, 2017-06-22:
07f5e991

closed 2017-06-22 by jordens

jbqubit, 2017-06-22:
@jordens Please post what you see on your spectrum analyzer.

jordens, 2017-06-22:
[image]
(the blue trace is just testing the limiter)

jbqubit, 2017-06-23:
OK. Confirmed. 

[image]
[image]

```
$ artiq_corelog 
[           7us]  INFO(runtime): ARTIQ runtime starting...
[        3855us]  INFO(runtime): software version 3.0.dev+1173.g07f5e991
[       10203us]  INFO(runtime): gateware version 3.0.dev+1173.g07f5e991
```