m-labs/artiq #753: version mismatch error due to string length mismatch

jbqubit, 2017-06-16:
```
(artiq3test4) britton@britton1:~/gitlab/jbqubit/artiqlab$ artiq_run repository/testing/test_simple_sin.py 
WARNING:artiq.coredevice.comm_kernel:Mismatch between gateware (3.0.dev+1075.gfdb4e3f8) and software (3.0.dev+1075.gfdb4e3f8c) versions
```
3.0.dev+1075.gfdb4e3f8
3.0.dev+1075.gfdb4e3f8c

jordens, 2017-06-16:
Does this occur with current master? There is no point in reporting issues that have already been fixed.

jbqubit, 2017-06-16:
(artiq3test4) britton@britton1:~/gitlab/jbqubit/artiqlab$ artiq_run repository/testing/test_simple_sin.py 
WARNING:artiq.coredevice.comm_kernel:Mismatch between gateware (3.0.dev+1126.g10fb6c62) and software (3.0.dev+1126.g10fb6c6) versions

sbourdeauducq, 2017-06-19:
fixed

closed 2017-06-19 by sbourdeauducq