m-labs/artiq #754: sawg: parallelism

jbqubit, 2017-06-16:
in doc for: core_drivers_reference.html#artiq.coredevice.sawg.SAWG

>The sum of their outputs is then interpolated by a factor of parallelism (2, 4, 8 depending on the bitstream)

class artiq.coredevice.sawg.SAWG(dmgr, channel_base, parallelism, core_device='core')

If parallelism is set in bitstream it shouldn't also be user-configurable in device_db.py.

jordens, 2017-06-16:
Many things (like RTIO channel numbers) are determined by the bitstream and still have to be specified  in the device database.
#355

closed 2017-06-16 by jordens

jbqubit, 2017-06-16:
Why perpetuate this annoying (and I gather vestigial) aspect of ARTIQ in SAWG? If the value of parallelism is already specified in the part of ARTIQ that defines .bit, it's poor design to have the same value repeated in device_db.py. 

You'd reject a code submission with a similar lazy implementation, no? It also confuses which elements of SAWG are soft-configurable and which require a recompile of .bit.

sbourdeauducq, 2017-06-16:
Because having automatic generation of device database (#355) would solve it for all devices (as far as it can be done, for example things like exact DDS frequencies may not be known) and not only for SAWG.

jbqubit, 2017-06-16:
OK.