m-labs/artiq #755: swag: up conversion depends on amplitude

jbqubit, 2017-06-16:
Goal: Test amplitude-dependence of up-conversion.

Expectation: Up-conversion is all digital. Spectral content should not vary with amplitude. That is,
digital mixer should have no amplitude-dependent non-linearity. 

```
(artiq3test4) britton@britton1:~/gitlab/jbqubit/artiqlab$ conda list artiq
# packages in environment at /home/britton/anaconda3/envs/artiq3test4:
#
artiq                     3.0.dev         py_1075+gitfdb4e3f8cfad    m-labs/label/dev
artiq                     3.0.dev0+1075.gfdb4e3f8c           <pip>
artiq-kc705-phaser        3.0.dev         py_1075+gitfdb4e3f8cfad    m-labs/label/dev
llvmlite-artiq            0.12.0.dev              py35_30    m-labs/label/main
```

```
$ cat test_upconvert3.py

from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
Goal: Test amplitude-dependence of up-conversion.

Expectation: Up-conversion is all digital. Spectral content should not vary with amplitude. That is,
digital mixer should have no amplitude-dependent non-linearity.
    """
    def build(self):
        self.setattr_device("core")
        self.setattr_device("sawg1")

    @kernel
    def sawg_reset(self):
        self.sawg1.frequency0.set(0.0)
        self.sawg1.frequency0.set(0.0)
        self.sawg1.phase0.set(0.0)
        self.sawg1.amplitude1.set(0.0)
        self.sawg1.frequency1.set(0.0)
        self.sawg1.phase1.set(0.0)
        self.sawg1.amplitude2.set(0.0)
        self.sawg1.frequency2.set(0.0)
        self.sawg1.phase2.set(0.0)
        self.sawg1.config.set_clr(1, 1, 1)

    @kernel
    def run(self):
        self.core.reset()
        delay(1*ms)
        self.sawg_reset()

        amin = 1
        amax = 100

        self.sawg1.frequency0.set(150*MHz)
        self.sawg1.frequency1.set(25*MHz)
        self.sawg1.frequency2.set(40*MHz)

        while(True):
            for a in range(amin, amax, 5):
                self.sawg1.amplitude1.set(a/100.0)
                self.sawg1.amplitude2.set(a/100.0)
                delay(500 * ms)
```

Observation: for a>0.5 spurious output exceeds 0 dBc.

jordens, 2017-06-16:
What do you mean by "spurious output"?
Your initial expectation is incorrect. Spectral content will vary with amplitude. There is truncation and there is limiting.
Obviously, once you add two oscillators with amplitudes of 0.5 each you get more than the DUC (and the DAC) can handle. The limiter will clip and produce harmonics. That can't be surprising.
Then finally, you are exacerbating the effect by reproducing #743

jbqubit, 2017-06-16:
Good point. With the range in this example the combined amplitude exceeds 1 and I shouldn't have an expectation of this working. My bad. 

See below for an example that better demonstrates the amplitude-dependence I see.

__Case 1__: a <0.4 I observe that for all combinations of f1 and f2 spurrs are < -50 dBc. Great! 

__Case 2__: a = 0.49 I observe that for some combinations of f1 and f2 spurrs are > -20 dBc. Is there a reason why the gateware can't include enough headroom to prevent this from happening? 

Hopefully Case 2 is remedied by fixing #743. Please reopen #743. 
```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    Test up-conversion feature of sawg.
    Expectation
    """
    def build(self):
        self.setattr_device("core")
        self.setattr_device("sawg1")

    @kernel
    def sawg_reset(self):
        self.sawg1.frequency0.set(0.0)
        self.sawg1.frequency0.set(0.0)
        self.sawg1.phase0.set(0.0)
        self.sawg1.amplitude1.set(0.0)
        self.sawg1.frequency1.set(0.0)
        self.sawg1.phase1.set(0.0)
        self.sawg1.amplitude2.set(0.0)
        self.sawg1.frequency2.set(0.0)
        self.sawg1.phase2.set(0.0)
        self.sawg1.config.set_clr(1, 1, 1)

    @kernel
    def run(self):
        self.core.reset()
        delay(1*ms)
        self.sawg_reset()

        f0 = 150
        fmin = 25
        fmax = 75
        a = 0.40

        self.sawg1.frequency0.set(f0*MHz)
        self.sawg1.amplitude1.set(a)
        self.sawg1.amplitude2.set(a)

        while(True):
            for f1 in range(fmin, fmax, 5):
                self.sawg1.frequency1.set(f1 * MHz)
                for f2 in range(fmin, fmax, 1):
                    self.sawg1.frequency2.set(f2 * MHz)
                    print('f0', f0, 'f1', f1, 'f2', f2)
                    delay(300 * ms)
```

jordens, 2017-06-16:
The gateware has headroom to prevent this from happening.

Please don't try to dictate our issue workflow. We have explained the procedure to you multiple times.

closed 2017-06-16 by jordens

jbqubit, 2017-06-28:
There are no unusual spurs that appear when running the following test code. OK. Confirmed fixed. 

```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """

    purpose: test amplitude dependence of two-tone up-conversion
    expectation:
        Since mixing is all digital spectral content should not vary with amplitude. That is,
        digital mixer should have no amplitude-dependent non-linearity.
    setup: spectrum analyzer with range set to 100 kHz to 500 MHz
    """
    def build(self):
        print(self.__doc__)
        self.setattr_device("core")
        self.setattr_device("sawg0")

    @kernel
    def run(self):
        self.core.reset()
        delay(300*us)
        self.sawg0.reset()

        amin = 1
        amax = 49

        self.sawg0.frequency0.set(150*MHz)
        self.sawg0.frequency1.set(25*MHz)
        self.sawg0.frequency2.set(40*MHz)

        for a in range(amin, amax, 5):
            self.sawg0.amplitude1.set(a/100.0)
            self.sawg0.amplitude2.set(a/100.0)
            delay(1000 * ms)
```