m-labs/artiq #756: sawg.reset() broken

jordens, 2017-06-19:
from #751 

test_simple_sin1.py

```python
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    purpose: test sinusoidal output
    expectation:
      10 MHz from CORDIC frequency1 on sawg0 and sawg1
      playback continues after program termination
      noise
    """
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")
        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def sawg_reset(self):
        self.sawg0.frequency0.set(0.0)
        self.sawg0.frequency0.set(0.0)
        self.sawg0.phase0.set(0.0)
        self.sawg0.amplitude1.set(0.0)
        self.sawg0.frequency1.set(0.0)
        self.sawg0.phase1.set(0.0)
        self.sawg0.amplitude2.set(0.0)
        self.sawg0.frequency2.set(0.0)
        self.sawg0.phase2.set(0.0)
        self.sawg0.config.set_clr(1, 1, 1)

        self.sawg1.frequency0.set(0.0)
        self.sawg1.frequency0.set(0.0)
        self.sawg1.phase0.set(0.0)
        self.sawg1.amplitude1.set(0.0)
        self.sawg1.frequency1.set(0.0)
        self.sawg1.phase1.set(0.0)
        self.sawg1.amplitude2.set(0.0)
        self.sawg1.frequency2.set(0.0)
        self.sawg1.phase2.set(0.0)
        self.sawg1.config.set_clr(1, 1, 1)

    @kernel
    def run(self):
        self.core.reset()
        self.ttl_sma.output()
        self.core.break_realtime()
        delay(1 * ms)
        self.sawg_reset()

        f0 = 10*MHz
        a0 = 0.5

        self.sawg0.frequency1.set(f0)
        self.sawg0.amplitude1.set(a0)
        self.sawg1.frequency1.set(f0)
        self.sawg1.amplitude1.set(a0)
```

test_reset.py:

```python
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    purpose: test sawg.reset()
    expectation:
      waveform stops after program termination

    """
    def build(self):
        self.setattr_device("core")
        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.break_realtime()
        delay(1*ms)
        self.sawg0.reset()
        self.sawg1.reset()
```

Steps to reproduce: 
1. reset KC705
2. artiq_run test_simple_sin1.py -- observe signal on scope (but it's ugly #752)
3. artiq_run test_reset.py -- observe signal on scope disappears
4. artiq_run test_simple_sin1.py -- no signal on scope

jordens, 2017-06-22:
07f5e991 and previous

closed 2017-06-22 by jordens

jbqubit, 2017-06-23:
Confirmed that sawgN.reset() now works using test code above. 

```
$ artiq_corelog 
[           7us]  INFO(runtime): ARTIQ runtime starting...
[        3855us]  INFO(runtime): software version 3.0.dev+1173.g07f5e991
[       10203us]  INFO(runtime): gateware version 3.0.dev+1173.g07f5e991
```