m-labs/artiq #758: run JESD204 PRBS and STAPL tests on firmware startup

sbourdeauducq, 2017-06-21:
See f3c7a7de3be9a89a14b2d5e5540affd0393d4591 for the old code.

sbourdeauducq, 2017-12-11:
@enjoy-digital how to do that?

enjoy-digital, 2017-12-11:
@sbourdeauducq: here is what i started for STATUS and PRBS:
https://github.com/enjoy-digital/artiq/commit/1c41f53cce3a3d659b91857dd1749a89f645c2c2
https://github.com/enjoy-digital/artiq/commit/14a2d60cf50aa37d3c369a8da540c425a7148d35
PRBS was not working but it should be easy to get it working.
I haven't implemented STAPL but procedure is still the same.

sbourdeauducq, 2017-12-11:
I think we shouldn't need STAPL - the purpose of this is to test the board's integrity, and PRBS should be enough, correct?

enjoy-digital, 2017-12-11:
@sbourdeauducq: yes for board's integrity this enough. (STAPL was useful to validate the core).
I'll take care of getting PRBS working in the firmware while doing others JESD related things.

sbourdeauducq, 2018-01-18:
https://github.com/m-labs/artiq/commit/07f4e2e35a4f7e007383a230ea4356989f1c9c96

closed 2018-01-18 by sbourdeauducq