m-labs/artiq #760: Experiment subwindows opening off-screen in multi-monitor systems

cjbe, 2017-06-27:
Sometimes experiment subwindows open off-screen, so that one has to scroll the main window to find them.

This only occurs on multi-monitor systems. If the monitors are horizontally arranged, with the main window on the left monitor, experiments open off-screen to the left of the main window view. This reverses if the main window is on the right monitor. Similarly behaviour with vertical symmetry occurs for vertically arranged monitors.

I see this frequently on both Linux and Windows systems, but I have not found a reliable reproduction yet.

jordens, 2017-06-28:
We don't handle the placement of new `QMdiSubWindow` s. Presumably this is a Qt misfeature/bug. Help appreciated.

cjbe, 2017-06-28:
This seems to only occur when the main window is not on the primary monitor of the system.

Oh dear - it looks like this is QTBUG-1645 : https://bugreports.qt.io/browse/QTBUG-1645
The bug has been open since 2008 ...

sbourdeauducq, 2017-08-01:
We could dodge this problem by using a tabbed view instead of multiple sub-windows for the experiments. But then you would only be able to display one experiment at a time in the main window (and switch between the open ones using the tab bar). As far as I know, it is desirable to support displaying multiple experiments side-by-side. Is this still true?

jbqubit, 2017-08-01:
>But as far as I know, it is desirable to support displaying multiple experiments 
>side-by-side. Is this still true?

Yes. It is desirable to display multiple experiments side by side.