m-labs/artiq #761: sawg.reset() insufficient timeline margin

jbqubit, 2017-06-28:
```
$ artiq_corelog
[           7us]  INFO(runtime): ARTIQ runtime starting...
[        3855us]  INFO(runtime): software version 3.0.dev+1173.g07f5e991
[       10203us]  INFO(runtime): gateware version 3.0.dev+1173.g07f5e991
```

```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    purpose: test sawg.reset()
    expectation:
      waveform stops after program termination
      timeline margin should be sufficient, no additional delay()
    observation: additional 200*us delay required
    """
    def build(self):
        self.setattr_device("core")
        self.setattr_device("sawg0")
        self.setattr_device("sawg1")
        self.setattr_device("ttl_sma")

    @kernel
    def run(self):

        # test for sufficient timeline margin in .reset()
        self.core.reset()
        self.ttl_sma.output()
        self.ttl_sma.pulse(1*us)

        # test for sufficient timeline margin in .reset()
        self.core.reset()
        delay(200 * us)
        self.sawg0.reset()
        self.sawg1.reset()
        self.ttl_sma.output()
        self.ttl_sma.pulse(1*us)
```

jordens, 2017-06-28:
Slack is something you have to manage yourself. We have never handled that and I can see no way to do that.

jbqubit, 2017-06-28:
Slack is handled by both the core

https://github.com/m-labs/artiq/blob/master/artiq/coredevice/core.py#L164
https://github.com/m-labs/artiq/blob/master/artiq/coredevice/core.py#L173

and sawg

https://github.com/m-labs/artiq/blob/master/artiq/coredevice/sawg.py#L317

I advocate for large-enough slack in these cases.

jordens, 2017-06-28:
Please refer to my comment above. I explained the first two cases. The line from SAWG hast nothing to do with slack.

closed 2017-06-28 by jbqubit