m-labs/artiq #762: sawg: summing junction prior to DUC

jbqubit, 2017-06-28:
```
$ artiq_corelog
[           7us]  INFO(runtime): ARTIQ runtime starting...
[        3855us]  INFO(runtime): software version 3.0.dev+1173.g07f5e991
[       10203us]  INFO(runtime): gateware version 3.0.dev+1173.g07f5e991
```

```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    purpose: test saturation of summing junction prior to DUC
    test: sawg0 is reference tone
    setup: scope run control single, manual trigger; 500ns/div, 100mV/div
    expectation:
        a0=1 is 160 mV amplitude
        if max=0.5 then sawg1 should not exceed 80 mV
        if min=-0.5 then sawg1 should not exceed -80 mV
    observation:
        both minimum and maximum are violated
    """
    def build(self):
        print(self.__doc__)
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")

        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        delay(300*us)
        self.sawg0.reset()
        self.sawg1.reset()
        delay(10 * ms)
        self.test()

    @kernel
    def test(self):
        f0 = 2*MHz
        t0 = 1*us
        a0 = 0.5

        # starting condition
        self.sawg0.amplitude1.set(2*a0)
        self.sawg0.frequency1.set(f0)
        self.sawg1.amplitude1.set(a0)
        self.sawg1.frequency1.set(f0)
        self.sawg1.amplitude2.set(a0)
        self.sawg1.frequency2.set(f0)

        # set clipping amplitude
        self.sawg1.config.set_duc_max(0.5)
        self.sawg1.config.set_duc_min(-0.5)
```
[image]

jordens, 2017-06-28:
The "decay" at the upper limit is the baluns highpassing your signal. When filing these bugs, please do have a look at the actual hardware you have in front of you, including cables, splitters, baluns, traces. A bit of diligence would make this debugging more efficient.

And you are submitting two events to the config channel at the same time. One gets dropped and then you are only setting one limit. Add some delay before for the second limit. I'll force that in the next commits.

closed 2017-06-28 by jordens

jbqubit, 2017-06-28:
> The "decay" at the upper limit is the baluns highpassing your signal. 
Checking the AD9154-FMC-EBZ [specification](https://wiki.analog.com/resources/eval/dpg/eval-ad9154) sheet. Looks like the balun ADT2-1T-1P+ is only specified for 8-600 MHz. Thanks for the reminder. Ok. 

[image]

But I can't see how balun bandwidth is responsible for the observed waveform. The expected behavior is sharp "clipping" which has a rapid onset and is dominated by frequency components well above 2 MHz. The balun should pass that just fine.

jordens, 2017-06-28:
Imagine you have a simple (e.g. RC) lowpass. Now convolve the step response into every clipped edge...

jbqubit, 2017-06-28:
Mathematica model of expected behavior of the high-pass. 
```
w0 = 2 \[Pi] 1;
whp = 2 \[Pi] 8;
dt = 0.001;
y = Table[Sin[w t], {t, 0.001, 2, dt}];
clip[y_, level_] := If[Abs[y] > level, Sign[y]*level, y]
yc = Map[clip[#, 0.5] &, y];
ychp = HighpassFilter[yc, whp, SampleRate -> IntegerPart[1/dt]];
ListLinePlot[{y, yc, ychp}, 
 PlotLegends -> {"original", "clipped", "clipped+hp"}]
```
[image]

jordens, 2017-06-28:
"The" high-pass?
Come on, Joe. This is simple RF electronics.

jbqubit, 2017-06-29:
>Imagine you have a simple (e.g. RC) lowpass. Now convolve the step response into every clipped edge...
There is no regime in which the model I pasted above produces the behavior I see in sawg. Your balun acting as a high-pass explanation doesn't seem to fit. 

Regardless, why does sawg ever permit the amplitude to be >>0.5?

jordens, 2017-06-29:
It doesn't. And you are thoroughly wrong.
Unfortunately we don't have the bandwidth to correct your understanding of filters, electronics, or signal processing here. Reading a book would probably be best here.

```python
dt = 2e-3
t = np.arange(1000)*dt
y = np.sin(2*np.pi*1*t)
y1 = np.clip(y, -1, .5)
y2 = signal.lfilter(x=y1, *signal.butter(1, 3*dt/2, "high"))
plt.plot(t, np.c_[y, y1, y2])
```
[image]

jbqubit, 2017-07-04:
I didn't appreciate the magnitude of the effect of the balun on the time-domain output at 2 MHz. Thanks for the scipy.signal example.  

# measure response to step input
```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    purpose: measure response of AD9154-FMC-EBZ to step stimulus
    test: step sawg1.offset
    setup: scope trigger on ttl_sma; 200ns/div, 100mV/div
    expectation:
        balun high-pass dominates
        317 mV peak decays to 1/e in 140 ns
    """
    def build(self):
        print(self.__doc__)
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")
        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        delay(300*us)
        self.sawg0.reset()
        self.sawg1.reset()
        self.ttl_sma.output()
        delay(10 * ms)
        self.test()

    @kernel
    def test(self):
        while True:
            self.ttl_sma.pulse(1 * us)
            self.sawg1.offset.set(0.99)
            delay(1*us)
            self.sawg1.offset.set(0.0)
            delay(1 * ms)

```
[image]

# check summing junction with 20 MHz tone
Shifting the test frequency to 20 MHz makes the output easier to interpret. Notwithstanding #765, the summing junction works as expected. 

```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    purpose: test saturation of summing junction prior to DAC output
    test: sawg0 is reference tone; clamp amplitude of sawg1
    setup: scope trigger on ttl_sma; 20ns/div, 100mV/div
    expectation:
        a0=1 is 300 mV amplitude
        symmetric clipping of sawg1 at +a0/2 and -a0/2
        see discussion for impact of output balun at low frequencies
        https://github.com/m-labs/artiq/issues/762
    """
    def build(self):
        print(self.__doc__)
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")
        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        delay(300*us)
        self.sawg0.reset()
        self.sawg1.reset()
        self.ttl_sma.output()
        delay(10 * ms)
        self.test()

    @kernel
    def test(self):
        f0 = 20*MHz
        t0 = 1*us
        a0 = 1.0
        asat = 0.5

        while True:
            self.sawg0.amplitude1.set(a0)
            self.sawg0.frequency1.set(f0)
            self.sawg1.amplitude1.set(a0)
            self.sawg1.frequency1.set(f0)
            self.sawg1.config.set_out_max(asat)
            delay(1*us)
            self.sawg1.config.set_out_min(-1*asat)

            self.ttl_sma.pulse(1*us)
            delay(1 * ms)
```
[image]

commit 3cbbcdfe (2017-06-28): sawg: don't enable_replace for Config

closes #762