m-labs/artiq #764: sawg: saturation of summing junction involving offset

jbqubit, 2017-06-28:
```$ artiq_corelog
[           7us]  INFO(runtime): ARTIQ runtime starting...
[        3855us]  INFO(runtime): software version 3.0.dev+1173.g07f5e991
[       10203us]  INFO(runtime): gateware version 3.0.dev+1173.g07f5e991
```

```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    purpose: test saturation of summing junction involving offset (just before DAC output)
    test: sawg0 is reference tone; clamp amplitude of sawg1
    setup: scope run control single, manual trigger; 500ns/div, 100mV/div
    expectation:
        a0=1 is 120 mV amplitude
        if max=0.5 then sawg1 should not exceed 60 mV
        if min=-0.5 then sawg1 should not exceed -60 mV
    observation:
        maximum is violated
        minimum is not sharply clamped at -0.5
    """
    def build(self):
        print(self.__doc__)
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")

        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        delay(300*us)
        self.sawg0.reset()
        self.sawg1.reset()
        delay(10 * ms)
        self.test()

    @kernel
    def test(self):
        f0 = 2*MHz
        t0 = 1*us
        a0 = 0.5

        # starting condition
        self.sawg0.amplitude1.set(2*a0)
        self.sawg0.frequency1.set(f0)
        self.sawg1.amplitude1.set(a0)
        self.sawg1.frequency1.set(f0)
        self.sawg1.offset.set(a0)

        # set clipping amplitude
        self.sawg1.config.set_out_max(0.5)
        self.sawg1.config.set_out_min(-0.5)
```
[image]

jordens, 2017-06-28:
Please consider closing this if understanding that this is AC coupled explains your observation.

jordens, 2017-07-03:
ping

jbqubit, 2017-07-04:
Notwithstanding #765, resolved as discussed in #762.

closed 2017-07-04 by jbqubit