m-labs/artiq #765: sawg: automatically serialize events to config channel

jbqubit, 2017-06-28:
In #762 @jordens points to a bug.

> And you are submitting two events to the config channel at the same time. One gets dropped and then you are only setting one limit. Add some delay before for the second limit. I'll force that in the next commits.

jordens, 2017-06-28:
Please stop the noise.

closed 2017-06-28 by jordens

jbqubit, 2017-07-03:
This bug persists in 3.0.dev+1207.g5b26e5de. Second call to config channel is silently dropped without delay.

```
            # set clipping amplitude
            self.sawg1.config.set_out_max(0.5)
            #delay(1*us)
            self.sawg1.config.set_out_min(-0.5)
```

jordens, 2017-07-04:
You do get a collision error in the core log.

whitequark, 2017-07-04:
Related to #691?

jordens, 2017-07-04:
Yes.

jbqubit, 2017-07-06:
> You do get a collision error in the core log.

Configuration that doesn't advance the timeline should not cause a collision error. 

```
        # no collision errors for these sawg commands
        self.sawg0.amplitude1.set(2*a0)
        self.sawg0.frequency1.set(f0)
        self.sawg1.amplitude1.set(a0)
        self.sawg1.frequency1.set(f0)
        self.sawg1.amplitude2.set(a0)
        self.sawg1.frequency2.set(f0)

        # collision error
        self.sawg1.config.set_duc_max(0.5)
        self.sawg1.config.set_duc_min(-0.5)
```

jordens, 2017-07-06:
No. Having two events at the same time for the config channel is almost always a mistake that should become visible. There should not be silent replacement. For the other channels it is reasonable to replace in that case.

jbqubit, 2017-07-06:
> Add some delay before for the second limit.  I'll force that in the next commits.

You've already agreed to add a delay that avoids the collision error. Please extend to all the config functions. set_div, set_duc_max, etc.

jordens, 2017-07-06:
Where did I agree to that?

jordens, 2017-07-07:
You are intentionally misinterpreting me. I said that I would force serialization by signalling `RTIOCollisions`.

jbqubit, 2017-07-07:
In #762 @jordens said

> And you are submitting two events to the config channel at the same time. One gets dropped and then you are only setting one limit. Add some delay before for the second limit. I'll force that in the next commits.

My interpretation of "I'll force that in the next commits" was different. I thought you were indicating agreement to add a delay to sawg.config.* as in reset(). My bad. As it stands I confirm that when the user responds to said "force" and adds a delay, no RTIO collision errors are generated and the sawg output is as expected. 

```
        # collision error
        self.sawg1.config.set_duc_max(0.5)
        delay(1*us)
        self.sawg1.config.set_duc_min(-0.5)
```
For those following at home... When ```delay(1*us)``` is commented out here's what a collision error looks like as reported by artiq_corelog.
```
[   779536143us] ERROR(runtime::rtio_mgt): RTIO collision
```
And here's where to read about RTIO Collisions
in the manual: ```artiq/doc/manual/_build/html/rtio.html?highlight=rtio%20collision#collisions```