m-labs/artiq #767: Help with KC705 board

trxw, 2017-06-30:
We are trying connect a KC705 (FMC) in Ubuntu. The first question is, are we supposed to connect via HDMI port or to the network interface? Well, we are connected via both, and this is what we get in the attached picture. We are also connected via a micro-usb port to an evaluation board, if that would be useful. 
[image]

We are trying to flash the board. What do you suggest? 

Thanks,
Omid

sbourdeauducq, 2017-07-01:
You need to connect the computer that runs artiq_flash to the micro-USB port of the kc705. Note that the kc705 has several USB ports and you must use the one with JTAG. Is the JTAG device recognized in dmesg and lsusb? HDMI is not used anywhere.

closed 2017-07-04 by sbourdeauducq

trxw, 2017-07-05:
We get the same error when connected to JTAG; so JTAG is not recognized.

sbourdeauducq, 2017-07-05:
Please answer the question.

whitequark, 2017-07-05:
@sbourdeauducq based on the screenshot it's a permission problem, no?

sbourdeauducq, 2017-07-05:
It could be, but the output of lsusb/dmesg would provide a clearer indication.

trxw, 2017-07-05:
The following outputs were given by lsusb/dmesg:

[image]

sbourdeauducq, 2017-07-05:
You have a USB problem; the JTAG adapter is not found at the USB level (as if the cable was disconnected). If you are using a virtual machine check that the USB device is attached to the VM. Also check the quality of your USB cables and that the USB connector is properly soldered on the KC705 JTAG circuit (I have seen a board that was shipped with a wrongly soldered connector there already).

trxw, 2017-07-05:
We are using a native Linux machine, so that shouldn't be an issue. This is the JTAG connection on the board:
[image]
Is it soldered correctly? If not, is there any other way to connect to the board?

sbourdeauducq, 2017-07-05:
Looks fine. The KC705 model that had this problem used glue to attach the USB connector, it seems you have a different model.
If you plug it into another computer is it recognized? e.g. if you plug it into a Windows machine does it shows the "new hardware detected" pop-up? Did you power up the KC705 board?

trxw, 2017-07-05:
Removing the hdmi cable made it work. Thank you :)

sbourdeauducq, 2017-07-05:
Uh, what? Where was the HDMI cable connected?

trxw, 2017-07-06:
The HDMI cable was accidentally connected which prevented the board from fully turning on. We were able to flash the board, but while trying to run led.py we get the following error:  
[image]

jordens, 2017-07-06:
* What's the output from the serial console of the kc705?
* Can you ping it?

sbourdeauducq, 2017-07-07:
Did you set the IP/MAC and is Ethernet connected?

sbourdeauducq, 2017-07-07:
Also I'm curious how you can connect the HDMI cable in a way that prevents the board from turning on. Was the other end connected to the HDMI output of the PC you use for flashing?

trxw, 2017-07-07:
We can't ping it. We set the IP/MAC as per the first set of instructions and the Ethernet is connected.

The HDMI was connected to the PC used for flashing.

sbourdeauducq, 2017-07-07:
Are the Ethernet LEDs on on both ends? Have you tried with a different network card, switch, etc.?

trxw, 2017-07-07:
The LEDs are on both ends. We are using the same network card as we use for the internet, so there shouldn't be an issue there.

sbourdeauducq, 2017-07-07:
Yes, but there is, so please try. Also review firewall settings, routers, etc. Are you connected to the PC directly without any switch/router?

trxw, 2017-07-07:
We are not using any firewall or router. We are connected to the PC directly.

sbourdeauducq, 2017-07-07:
Then check your IP settings on the PC side as well, and if you don't find and fix a problem there, try a different network card. What is the output of ``ip addr show`` for that network card?

trxw, 2017-07-07:
[image]

sbourdeauducq, 2017-07-07:
Well yes, you need to set an IP address for the PC.

trxw, 2017-07-07:
We ran the following command:

sudo ifconfig enp2s0 198.192.0.1 netmask 255.255.255.0

We got same error.

sbourdeauducq, 2017-07-07:
> 198.192.0.1

is that a LAN address?

sbourdeauducq, 2017-07-07:
If you have not done this before, try to connect to PCs with an Ethernet cable directly and make them communicate.

trxw, 2017-07-07:
Yes, that is a LAN address we set. We are also connected via a crossover Ethernet cable. The card works ok with routers so it should be fine.

jordens, 2017-07-07:
And you have a router on that network that is aware of routes to 198.192.0.1   and 128.61.118.48. And 128.61.118.48 is aware of that router? (Or 128.61.118.48 is aware of a direct route to 198.192.0.1).

sbourdeauducq, 2017-07-08:
@trxw Please read up on TCP/IP networking, do this exercise of connecting two computers directly via an Ethernet cable, and once you master that you should be able to get the KC705 to work.

Flashing does not use Ethernet so it is not strange at all.

llopez32, 2017-07-25:
We are able to connect a Ubuntu desktop (IP: 192.168.0.3) and a Windows laptop (IP: 192.168.0.2), so it should be aware of the route. Using the same router and Ethernet cables, the desktop can ping the board but a socket error is still thrown while trying to run the script.

jordens, 2017-07-25:
Please double check your numbers. The IPs you mention all.over this issue report are inconsistent. I suspect you are just hitting your own typos. If not, please provide your actual configuration for confirmation.

llopez32, 2017-07-25:
[image]

jordens, 2017-07-25:
Hmm. This is all very different from before. Please also paste your `device_db.py`.

Also, in general, it's easier if you just paste the text as such (using markdown markup to format it) instead of pasting screenshots.

llopez32, 2017-07-31:
device_db.py was incorrectly configured; we can run scripts now. Thank you!

However, we are now having trouble mapping the channels to the pins. We also have a AD9154 FMC board.

jbqubit, 2017-08-01:
The current practice for the ARTIQ community is to use github Issues for bug reporting and discussing new futures. Please use the ARTIQ mailing list for general Q&A. 

https://ssl.serverraum.org/lists/listinfo/artiq