m-labs/artiq #769: ad9154 bad CODEGRPSYNCFLG

jbqubit, 2017-07-01:
```
$ flterm /dev/ttyUSB0

MiSoC BIOS
(c) Copyright 2007-2017 M-Labs Limited
Built Jun 26 2017 03:16:28

BIOS CRC passed (ec5efbb1)
Initializing SDRAM...
Write leveling: 15* 16* 13  13   8   7   3   5  completed
Read bitslip: 7 6 5 4 3 
Read delays: 7:01-11  6:01-12  5:03-12  4:03-14  3:10-21  2:00-05  1:00-09  0:00-09  completed
Memtest OK
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from flash...
Loading 470188 bytes from flash...
Executing booted program.
[         106us]  INFO(runtime): ARTIQ runtime starting...
[        3936us]  INFO(runtime): software version 3.0.dev+1183.gf4624e08
[       10286us]  INFO(runtime): gateware version 3.0.dev+1183.gf4624e08
[       16619us]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     1016071us]  INFO(runtime): continuing boot
[     1076492us]  WARN(board::ad9154): config attempt #0 failed (bad CODEGRPSYNCFLG), retrying
[     1153757us]  WARN(board::ad9154): config attempt #1 failed (bad CODEGRPSYNCFLG), retrying
[     1202579us]  WARN(board::ad9154): config attempt #2 failed (bad CODEGRPSYNCFLG), retrying
[     1311536us]  WARN(board::ad9154): config attempt #3 failed (bad CODEGRPSYNCFLG), retrying
[     1367696us]  WARN(board::ad9154): config attempt #4 failed (bad CODEGRPSYNCFLG), retrying
[     1412778us]  WARN(board::ad9154): config attempt #5 failed (bad CODEGRPSYNCFLG), retrying
[     1534105us]  WARN(board::ad9154): config attempt #6 failed (bad CODEGRPSYNCFLG), retrying
[     1578923us]  WARN(board::ad9154): config attempt #7 failed (bad CODEGRPSYNCFLG), retrying
[     1651973us]  WARN(board::ad9154): config attempt #8 failed (bad CODEGRPSYNCFLG), retrying
[     1703961us]  WARN(board::ad9154): config attempt #9 failed (bad CODEGRPSYNCFLG), retrying
[     1777263us]  WARN(board::ad9154): config attempt #10 failed (bad CODEGRPSYNCFLG), retrying
[     1821643us]  WARN(board::ad9154): config attempt #11 failed (bad CODEGRPSYNCFLG), retrying
[     1873934us]  WARN(board::ad9154): config attempt #12 failed (bad CODEGRPSYNCFLG), retrying
[     1940350us]  WARN(board::ad9154): config attempt #13 failed (bad CODEGRPSYNCFLG), retrying
[     2028499us]  WARN(board::ad9154): config attempt #14 failed (bad CODEGRPSYNCFLG), retrying
[     2097405us]  WARN(board::ad9154): config attempt #15 failed (bad CODEGRPSYNCFLG), retrying
[     2177155us]  WARN(board::ad9154): config attempt #16 failed (bad CODEGRPSYNCFLG), retrying

```

sbourdeauducq, 2017-07-01:
Details? Obviously it doesn't do that every time.

jbqubit, 2017-07-01:
I've flashed that version three times and saw erratic LED blinking each time. That's why I reverted to g07f5e991. What additional information would be helpful?

jordens, 2017-07-01:
Does this happen with current master? Does it happen every time?

jbqubit, 2017-07-01:
bb reports timing failed for build of artiq-kc705-phaser for current master. Please let me know when there's a binary available and I'll test again. 

http://buildbot.m-labs.hk/builders/artiq-board/builds/688/steps/conda_build

sbourdeauducq, 2017-07-02:
Don't assume that if it fails timing on the buildbot it will also do so on your machine - Xilinx software is not that consistent. This bug also smells like the usual intermittent Xilinx breakage, we did not touch that part.
What is it that you want to test that would not work with the previous build?

sbourdeauducq, 2017-07-02:
Anyway, there is a recent package now.

jordens, 2017-07-03:
ping

jbqubit, 2017-07-03:
I no longer get this error with build 3.0.dev+1207.g5b26e5de.

closed 2017-07-03 by jbqubit