m-labs/artiq #770: phaser: 55 seconds to load demo_2tone.py

jbqubit, 2017-07-01:
Rolling back to last version known to be good. Continuing phaser testing using M-Labs code. 

Observation: It takes > 50 seconds to load demo_2tone.py. Not sure when this regression occurred. 

Expectation: Loading in about one second. 

```
$ artiq_run repository/demo_2tone.py
```
After long delay the output on scope looks as expected. 

```
$ artiq_corelog
[           7us]  INFO(runtime): ARTIQ runtime starting...
[        3855us]  INFO(runtime): software version 3.0.dev+1173.g07f5e991
[       10204us]  INFO(runtime): gateware version 3.0.dev+1173.g07f5e991
[       16537us]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     1016006us]  INFO(runtime): continuing boot
[     1056090us]  INFO(runtime): using MAC address 00-0a-35-03-1e-53
[     1060915us]  INFO(runtime): using IP address 192.168.1.71
[     1066429us] ERROR(runtime::rtio_mgt): unrecognized startup_clock configuration entry
[     1074177us]  INFO(runtime::rtio_mgt): startup RTIO clock: Internal
[     1080727us]  INFO(runtime): UART log level set to INFO by default
[     1087257us]  INFO(runtime::mgmt): management interface active
[     1100951us]  INFO(runtime::session): accepting network sessions
[     1114787us]  INFO(runtime::session): running startup kernel
[     1119250us]  INFO(runtime::session): no startup kernel found
[     1124939us]  INFO(runtime::session): no connection, starting idle kernel
[     1131809us]  INFO(runtime::session): no idle kernel found
[    48291217us]  INFO(runtime::session): new connection from 192.168.1.68:48989
[    48302631us]  WARN(runtime): network error: dropped by socket
[    48307513us]  WARN(runtime): network error: dropped by socket
[    48551546us]  WARN(runtime): network error: dropped by socket
[    48556179us]  WARN(runtime): network error: dropped by socket
[    49080210us]  INFO(runtime::kern_hwreq): resetting RTIO
[    52078420us]  WARN(runtime): network error: dropped by socket
[    55082442us]  WARN(runtime): network error: dropped by socket
[    58086446us]  WARN(runtime): network error: dropped by socket
[    61090497us]  WARN(runtime): network error: dropped by socket
[    61371395us]  INFO(runtime::session): no connection, starting idle kernel
[    61377162us]  INFO(runtime::session): no idle kernel found
[    66691655us]  INFO(runtime::session): new connection from 192.168.1.68:48990
[    66705794us]  WARN(runtime): network error: dropped by socket
[    66711278us]  WARN(runtime): network error: dropped by socket
[    66908589us]  WARN(runtime): network error: dropped by socket
[    66913914us]  WARN(runtime): network error: dropped by socket
```

sbourdeauducq, 2017-07-01:
Likely not a problem with the latest version.

sbourdeauducq, 2017-07-01:
That one you are reporting about is before a lot of TCP fixes.

closed 2017-07-01 by sbourdeauducq

jbqubit, 2017-07-01:
Sure. I can believe this got better with TCP fixes. In light of #769 what version do you recommend for continuing phaser testing?

jbqubit, 2017-07-03:
With recent updates repository/demo_2tone.py loads in < 2 seconds. OK.

```
$ artiq_corelog
[           8us]  INFO(runtime): ARTIQ runtime starting...
[        3853us]  INFO(runtime): software version 3.0.dev+1207.g5b26e5de
[       10201us]  INFO(runtime): gateware version 3.0.dev+1207.g5b26e5de
```

sbourdeauducq, 2017-07-04:
Note that if you use ``artiq_run``, a good part of those 2 seconds are from setuptools (https://github.com/pypa/setuptools/issues/510). Master-run experiments are not affected.