m-labs/artiq #771: sawg: sawg.offset(1.0) produces -1.0 amplitude

jbqubit, 2017-07-04:
```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    purpose: check sign of sawg1.offset
    expectation: see inline comments below
    """
    def build(self):
        print(self.__doc__)
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")
        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        delay(300*us)
        self.sawg0.reset()
        self.sawg1.reset()
        self.ttl_sma.output()
        delay(10 * ms)
        self.test()

    @kernel
    def test(self):
        while True:
            self.ttl_sma.pulse(1 * us)
            # expect: positive polarity pulse
            # observe: positive polarity pulse
            self.sawg1.offset.set(0.99)
            delay(100*ns)
            self.sawg1.offset.set(0.0)
            delay(1*us)
            # expect: positive polarity pulse
            # observe: negative polarity pulse
            self.sawg1.offset.set(1.0)
            delay(100*ns)
            self.sawg1.offset.set(0.0)
            delay(1*us)
            # expect: negative polarity pulse
            # observe: negative polarity pulse
            self.sawg1.offset.set(-1.0)
            delay(100*ns)
            self.sawg1.offset.set(0.0)
            delay(1*us)

            delay(1 * ms)
```
[image]

```
$ artiq_corelog
[           8us]  INFO(runtime): ARTIQ runtime starting...
[        3853us]  INFO(runtime): software version 3.0.dev+1207.g5b26e5de
[       10201us]  INFO(runtime): gateware version 3.0.dev+1207.g5b26e5de
```

jordens, 2017-07-04:
The positive maximum of a signed integer is one less (`(1 << N - 1) - 1`) than the negative maximum (`(1 << N - 1)`). This is intrinsic and you will also find it with such fine hardware as National Instruments DACs. Currently, scaling involves multiplication with a power of two. Changing the behavior would make that calculation more expensive.

sbourdeauducq, 2017-07-04:
Could the gateware have one more bit and then clip?

jordens, 2017-07-04:
This is not just gateware. If there was one more bit, it would make each of the four coefficients on that spline one **byte** wider, from the kernel support, all the way down into the gateware. I'd much rather change the scaling factor.

hartytp, 2017-07-04:
Better just to leave it as is, and add an extra explanatory note in the documentation?

The current situation is very standard for any DDS type hardware (cf AD9910 ASF).

jbqubit, 2017-07-04:
@hartytp This would result in an inconsistent API. sawg.offset has the 2**N issue but sawg.amplitude1 and sawg.amplitude2 do not. 

@hartytp Agreed that the 2**N wrapping bit is common for DACs. But sawg is a sufficiently high level interface to the DAC -- it's even got CORDICs and DUCs! -- that it ought to handle this too.

jbqubit, 2017-07-04:
I'd be fine with a non-linear solution like the following
2**N -> 2**N -1 (fix)
2**N -1  -> 2**N - 1 (usual)

jordens, 2017-07-04:
@jbqubit no. amplitude1/2 have the same issue. just there the scaling factor is already dense and therefore I moved the problem from "1.0" to "1.0 + epsilon" and thus "out of your zone of curiosity".
By that logic every "sufficiently high level interface to" a device should be expected to have solved every problem "common" to those devices by now. That is obviously not the case.

jordens, 2017-07-04:
That "nonlinear" solution would be even more expensive. I'll just change the scaling factor.

jbqubit, 2017-07-04:
I'd be fine with applying a scaling factor as for amplitude1 and amplitude2.

sbourdeauducq, 2017-07-04:
We can also test that the floats passed are in the range ``[-1.0;1.0[``? (or ``]-1.0;1.0[`` if it makes it easier to test with any applicable bit twiddling hack)

jordens, 2017-07-04:
I don't think testing will help. It does give the user some error message. But it is more expensive than changing the scaling factor and it fails as soon as someone becomes curious and expects those tests to be applied to the spline outputs as well.

closed 2017-07-04 by jordens

jbqubit, 2017-07-07:
Confirmed.

commit 91ca9fbc (2017-07-04): sawg: also give offset some headroom

closes #771