m-labs/artiq #772: sawg: sawg.frequency0 to sawg.frequency1 phase offset

jbqubit, 2017-07-04:
This was fixed in #748 but has regressed. Electrical paths are matched. 

```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    purpose: test absolute alignment of sawg0.frequency0 and sawg1.frequency1
    test: make sawg0 and sawg1 phase aligned
    expectation: no phase offset between sawg0 and sawg1
    observation: delay between rising edges -- FAIL
    setup: scope single run, manually trigger acquisition; 10ns/div, 100mV/div
    """
    def build(self):
        print(self.__doc__)
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")

        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        delay(300 * us)
        self.sawg0.reset()
        self.sawg1.reset()
        self.ttl_sma.output()
        delay(1 * ms)

        f0 = 50*MHz
        a0 = 0.5

        self.sawg0.amplitude1.set(a0)
        self.sawg0.frequency0.set(f0)
        self.sawg0.phase0.set(0.)
        self.sawg1.amplitude1.set(a0)
        self.sawg1.frequency1.set(f0)
        self.sawg1.phase1.set(0.)

        delay(1*ms)
```
[image]

```
$ artiq_corelog
[           8us]  INFO(runtime): ARTIQ runtime starting...
[        3853us]  INFO(runtime): software version 3.0.dev+1207.g5b26e5de
[       10201us]  INFO(runtime): gateware version 3.0.dev+1207.g5b26e5de
```

closed 2017-07-04 by jordens

jordens, 2017-07-04:
Na. That's not a solution. I'll revert it.

The underlying issue is still the same. You will always find something that does not match when trying to compare the DDS with the parallelized DUC. They are just different. I could correct it for phase but then it will fail for amplitude. Or you'll compare the offset channel to phase1 or amplitude1. Let's state that equal things are well synchronized but unequal things may have a deterministic latency difference of strictly less than one coarse RTIO cycle.

jbqubit, 2017-07-05:
> Let's state that equal things are well synchronized but unequal things may have a deterministic latency difference of strictly less than one coarse RTIO cycle.

That's a fine compromise. Please update the docs as to which oscillators are expected to be synchronizable by phaseN.set().

jordens, 2017-07-07:
3222f5036f72b35a6f3a1374c3210b5acba51bc8

jbqubit, 2017-07-07:
@jordens Thanks for documenting this.

commit 8e0a1cbd (2017-07-04): sawg: advance dds 1/2 by one sample group

closes #772