m-labs/artiq #773: rtio: RTIOcollision for simultaneous events

jbqubit, 2017-07-04:
```exception artiq.coredevice.exceptions.RTIOCollision```

>Raised when an event is submitted on a given channel with the same coarse timestamp as the previous one but with a different fine timestamp. ([link](https://m-labs.hk/artiq/manual-release-2/core_drivers_reference.html#artiq.coredevice.exceptions.RTIOCollision))

Logically, RTIOCollision should also be raised if the event has the same coarse timestamp and fine timestamp as the previous. This would catch for example

```
        with parallel:
            self.ttl_sma.output()
            self.ttl_sma.input()
```

jbqubit, 2017-07-04:
Other examples.

```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    purpose: test trapping of RTIO errors
    test:
    """
    def build(self):
        print(self.__doc__)
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")

        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        delay(300 * us)
        self.sawg0.reset()
        self.sawg1.reset()
        self.ttl_sma.output()
        delay(1 * ms)

        f0 = 50*MHz
        a0 = 0.5

        # following causes no error
        # expectation: RTIO collision error
        with parallel:
            self.sawg0.amplitude1.set(0.5)
            self.sawg0.amplitude1.set(0.1)

        # following causes no error
        # expectation: RTIO collision error
        with parallel:
            self.sawg0.frequency1.set(10*MHz)
            self.sawg0.frequency1.set(20*MHz)

        # following causes RTIO collision error
        # expectation: no RTIO collision error
        with parallel:
            self.sawg1.config.set_duc_max(0.5)
            self.sawg1.config.set_duc_min(-0.5)

        # following causes no error
        # expectation: RTIO collision error
        with parallel:
            self.ttl_sma.output()
            self.ttl_sma.input()
```

sbourdeauducq, 2017-07-04:
TTL devices do RTIO replace, so that e.g. ``.on()`` followed by ``.off()`` is equivalent to just ``.off()``. Following a mailing list discussion, this feature ought to be kept (https://ssl.serverraum.org/lists-archive/artiq/2016-November/001052.html). But it is enabled at the RTIO device level for all RTIO operations to that device, and cannot currently be applied to certain operations only.

sbourdeauducq, 2017-07-04:
Having some filtering (e.g. per address) on what RTIO operations have ``replace`` enabled is doable, but it will take some work and increase FPGA resource usage a bit.

jbqubit, 2017-07-06:
For SAWG here's the desired behavior
- collision error for contemporaneous actions that modify the same register 
- no error for contemporaneous actions that modify different registers

> But it is enabled at the RTIO channel level for all RTIO operations to that channel, and cannot currently be applied to certain operations only.

Does this mean it can be disabled for SAWG?

jordens, 2017-07-06:
Coincident modification of multiple registers is not supported.

jbqubit, 2017-07-06:
> Coincident modification of multiple registers is not supported.

By "contemporaneous" I mean happening at the same point on the experiment time line.

-----
Based on your response I believe my confusion has to do with the auto-magical creation of sawg knot points. I'd like to have a discussion about this element of the API and propose a tweak. Please don't deflect by pointing out that I can write a wrapper for sawg to get whatever interface I want. 

 It looks like there are two classes of parameters for sawg.
- __sawg_spline_p__: frequency0, phase0, amplitude1, frequency1, phase1, amplitude2, frequency2, phase2, offset, i_enable, q_enable
- __sawg_config_p__: duc_max, duc_min, out_max, out_min, div_n, div_div, clr0, clr1, clr2, iq_en

Features of each class of parameters:

__sawg_knot_p__
- minimum resolution in time for sawg knot points is t_sawg_spline = t_rtio_coarse*(div.div+1)
- each sawg_knot_p has its own setter function
- setters can be in ```with parallel``` block 
- if time line cursor is not modified, ```with parallel``` is implicit
- if the time line cursor is not moved and .set() is called twice in succession for the same sawg_knot_p parameter, the second call overwrites the first
- if any of sawg_knot_p are set() and mod( now(), t_sawg_spline ) != 0, something undocumented happens
- if any of sawg_knot_p are set() a new sawg know point is "auto magically" created at now() 
- auto magical knot creation does not advance the time line

__sawg_config_p__
- are mutually independent registers
- they modulate future waveform evolution
- they can't be modified concurrently
- they have setter functions
- some setter functions modify time line, some do not modify time line

# proposed API modification
Define a user-side data structure to hold sawg knot data. 
Define an explicit routine for adding knot points to the sawg timeline as an alternative to auto magic knot addition. 
```
class sawg_knot_p:
    """sawg knot point"""
    def __init__(self):
        self.offset = 0.
        self.i_enable = True
        self.q_enable = False
        self.amplitude0 = 0.
        self.frequency0 = 0.
        self.phase0 = 0.
        self.amplitude1 = 0.
        self.frequency1 = 0.
        self.phase1 = 0.
        self.amplitude2 = 0.
        self.frequency2 = 0.
        self.phase2 = 0.

def artiq.coredevice.sawgN.now(self, sawg_knot_p):
    """schedules sawg knot sawg_knot_p at now()
        multiple calls to add_knot_now at single point in time line causes RTIO collision error """
```

Example
```
    p = sawg_knot_p()
    tpi = 1*us; tarm = 2*us

    p.amplitude1 = 1.; p.frequency0 = 10*MHz; sawg0.now(p)
    delay(tpi/2); 
    p.amplitude1 = 0.; sawg0.now(p)
    
    delay(tarm)
    
    p.amplitude1 = 1.; p.phase0=0.5; sawg0.now(p)
    delay(tpi)
    p.amplitude1 = 0.; sawg0.now(p)
    
    delay(tarm)
    
    p.amplitude1 = 1.; p.phase0=0.0; sawg0.now(p)
    delay(tpi/2)
    p.amplitude1 = 0.; sawg0.now(p)
```
Since ```p``` is a simple data structure there's no confusion about overwriting, concurrency or when an RTIO event is issued. All the intuition users have about how a simple data structure behaves pertain. 

Example
```
    pa = sawg_knot_p(); pb = sawg_knot_p(); poff = sawg_knot_p()
    pa.amplitude1 = 1.; pa.frequency0 = 10*MHz; pa.phase0 = 0.
    pb.amplitude1 = 1.; pb.frequency0 = 10*MHz; pb.phase0 = 0.5
    
    tpi = 1*us; tarm = 2*us

    sawg0.now(pa); delay(tpi/2); sawg0.now(poff)
    delay(tarm)
    sawg0.now(pb); delay(tpi); sawg0.now(poff)
    delay(tarm)
    sawg0.now(pa); delay(tpi/2); sawg0.now(poff)
```

jbqubit, 2017-07-06:
I updated my comment to be less pseudo-code and not abuse Spline() so badly.

jordens, 2017-07-06:
What do you mean by "auto-magical creation of sawg knot points"? And I don't understand what problem you are trying to solve.

jbqubit, 2017-07-06:
The docs are good at saying which things advance the timeline. 
- spline.smooth()
- sawg.reset()
- delay()

The docs are less clear about which calls add sawg/spline knot points or modify existing knot points. A concise statement and example could make this clearer. 
- spline.smooth()
- spline.set_mu()
- spline.set_coeff_mu() 
- spline.set()
- and, e.g. sawg0.amplitude1.XX() for XX() above

sbourdeauducq, 2017-07-07:
> But it is enabled at the RTIO channel level for all RTIO operations to that channel, and cannot currently be applied to certain operations only.
> Does this mean it can be disabled for SAWG?

The SAWG is a multi-channel device, and configuration requests have one dedicated channel (that has replace disabled).

jordens, 2017-07-07:
The only thing you can do on a spline channel is setting the current spline coefficients.

closed 2017-07-07 by jordens

jbqubit, 2017-07-08:
The behaviour of RTIO ```rtlink.OInterface()``` with ```enable_replace=False``` is not documented. 

@jordens A brief explanation on what happens behind the scenes in the language of RTIO would help help users develop a mental model for how SAWG works.  Here's a proposal.

- Each SAWG device (e.g. ```sawg0```) has nine ```artiq.coredevice.spline.Spline``` interpolators (e.g. ```sawg0.frequency0```) each with its own RTIO channel. 
    - A call to ```sawg0.frequency0.set()``` instructs the ARTIQ runtime to schedule an RTIO event at ```now()```.  
    - ```RTIO replace``` is enabled for SAWG spline channels; if multiple RTIO events are issued on the same spline channel with the same time stamp, only the last takes effect. 
- Each SAWG device has a single RTIO channel for all configuration commands (e.g. ```sawg0.config.set_duc_max()```).
    - ```RTIO replace``` is [disabled](https://github.com/m-labs/artiq/blob/3cbbcdfe969ef8258ae2770db316f95842a22432/artiq/gateware/rtio/phy/sawg.py) for the SAWG config channels; if two RTIO events are issued on the same config channel with the same time stamp, only first is retained and an RTIO collision error is generated.

jordens, 2017-07-08:
Valid request. That would need to be done for all RTIO channels, not just SAWG. Please file a separate issue. This one has drifted.

jbqubit, 2017-07-08:
How many new Issues do you see here? I count two. Maybe three. 
- The behaviour of RTIO ```rtlink.OInterface()``` with ```enable_replace=False``` is not documented.
- The ```enable_replace``` status of RTIO channels is not comprehensively documented for each channel type.

I proposed an update to SAWG documentation which was not addressed in your comment. The language connects the user-facing SAWG API and the underlying mechanisms of RTIO. ARTIQ documentation is increasingly succinct and abstract but leaves me wanting connections back to how the concepts are deployed. That is linking "what xxx is in abstraction" to "example instances of xxx in the the code base".

jordens, 2017-07-09:
* One issue: #777 This is generic.
* `rtlink` is an internal API. We don't duplicate the code functionality in the description of internal APIs. If you want to know how the code works you have to read the code. It is ultimately pointless to attempt to explain the implementation via the documentation only.

jbqubit, 2017-07-10:
@jordens ping regarding 
>I proposed an update to SAWG documentation which was not addressed in your comment.

jordens, 2017-07-10:
Above you propose an API modification (which I don't understand), describe the current documentation, and raise something about "add sawg/spline knot points or modify existing knot points". I have responded to the latter. I am unsure what you are referring to.

jbqubit, 2017-07-10:
My ping today relates to my proposal to [better document](https://github.com/m-labs/artiq/issues/773#issuecomment-313873058) of the current API in lieu of an [API modification](https://github.com/m-labs/artiq/issues/773#issuecomment-313492340).

jordens, 2017-07-10:
I don't get it. The link you supply is not your proposal, it is my implementation, and it is already committed. What do you want me to comment on?

jordens, 2017-07-10:
Github shows my commit above your comment. I already commented on that request of yours:

> rtlink is an internal API. We don't duplicate the code functionality in the description of internal APIs. If you want to know how the code works you have to read the code. It is ultimately pointless to attempt to explain the implementation via the documentation only.

jbqubit, 2017-07-10:
Not sure why the links (there are two) are causing confusion. Maybe different browsers render differently -- I see a difference in vertical scrolling in Chrome that could be confusion which isn't present in Firefox. Anyway....

Thank you for your comment on ```rtlink```. My ping relates to the rest of that comment. Reproduced below. 

---

A brief explanation on what happens behind the scenes in the language of RTIO would help help users develop a mental model for how SAWG works.  Here's a proposal.

- Each SAWG device (e.g. ```sawg0```) has nine ```artiq.coredevice.spline.Spline``` interpolators (e.g. ```sawg0.frequency0```) each with its own RTIO channel. 
    - A call to ```sawg0.frequency0.set()``` instructs the ARTIQ runtime to schedule an RTIO event at ```now()```.  
    - ```RTIO replace``` is enabled for SAWG spline channels; if multiple RTIO events are issued on the same spline channel with the same time stamp, only the last takes effect. 
- Each SAWG device has a single RTIO channel for all configuration commands (e.g. ```sawg0.config.set_duc_max()```).
    - ```RTIO replace``` is [disabled](https://github.com/m-labs/artiq/blob/3cbbcdfe969ef8258ae2770db316f95842a22432/artiq/gateware/rtio/phy/sawg.py) for the SAWG config channels; if two RTIO events are issued on the same config channel with the same time stamp, only first is retained and an RTIO collision error is generated.

jordens, 2017-07-10:
Ack. It's a bit verbose but otherwise would address part of #777

commit f0841f54 (2017-07-07): spline: be really verbose

closes #773