m-labs/artiq #774: SAWG: doesn't flag RTIO sequence errors

jbqubit, 2017-07-04:
Expectation is that SAWG and TTL behave the same for events with duration. 
```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    """
    purpose: test trapping of RTIO errors
    test:
    """
    def build(self):
        print(self.__doc__)
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")

        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        delay(300 * us)
        self.sawg0.reset()
        self.sawg1.reset()
        self.ttl_sma.output()
        delay(1 * ms)

        f0 = 50*MHz
        a0 = 0.5

        # following causes no error
        # expectation: RTIO sequence error
        with parallel:
            self.sawg0.amplitude1.smooth(0.5, .0, 1*us, 3)
            self.sawg0.amplitude1.smooth(0.2, .0, 4*us, 3)

        # following generates RTIO sequence error
        # expectation: RTIO sequence error
        with parallel:
            self.ttl_sma.pulse(0.5*us)
            self.ttl_sma.pulse(1.0*us)
        delay(1*ms)
```

jordens, 2017-07-04:
The SAWG behavior is not a bug. Replacing events on those channels is supported.

For TTL.pulse() it is also not a bug. Those events do not collide. The rising edge event is replaced as above.

closed 2017-07-04 by jordens

jbqubit, 2017-07-04:
>For TTL.pulse() it is also not a bug.
Agreed. I'm not reporting a TTL bug. I'm contrasting the behavior of TTL and SAWG in a similar situation. I expect that same-time event replacement is flagged as an error.

sbourdeauducq, 2017-07-04:
@jordens This issue is about sequence errors (non-increasing timestamps on the same channel), not collisions/replacements.

jordens, 2017-07-04:
@jbqubit `ttl.pulse()` are two events. `sawg.smooth()` is one event. With one event you can only ever get a replacement or a collision. With two events you can also get sequence errors. you can not compare those two cases.
@sbourdeauducq ack