m-labs/artiq #775: typo in example device_db.py

jbqubit, 2017-07-07:
```
    "core_log": {
        "type": "controller",
        "host": "::1",
        "port": 1068,
--->        "command": "aqctl_corelog -p {port} --bind {bind} " + core_addr
    },
```
applies to
examples/drtio
examples/phaser
examples/master

jbqubit, 2017-07-07:
Error 1 is typo in artiq_corelog

Error 2 is arguments ```-p``` and ```--bind``` which are not supported by ```artiq_corelog```.

```
$ artiq_corelog --help
usage: artiq_corelog [-h] [-v] [-q] [--device-db DEVICE_DB]
                     {clear,set_level,set_uart_level} ...

ARTIQ core device log tool

positional arguments:
  {clear,set_level,set_uart_level}
    clear               clear log buffer
    set_level           set minimum level for messages to be logged
    set_uart_level      set minimum level for messages to be logged to UART

optional arguments:
  -h, --help            show this help message and exit
  --device-db DEVICE_DB
                        device database file (default: 'device_db.py')

verbosity:
  -v, --verbose         increase logging level
  -q, --quiet           decrease logging level
```

cjbe, 2017-07-07:
This is not a typo. `aqctl_corelog` runs [artiq.frontend.aqctl_corelog](https://github.com/m-labs/artiq/blob/master/artiq/frontend/aqctl_corelog.py) - the core device logging controller

jbqubit, 2017-07-07:
@cjbe Thanks for the heads up. I see that controllers are now renamed in v3. (artiq/doc/manual/_build/html/release_notes.html). For some reason I did have aqctl_lda but not aqctl_corelog in anaconda3/envs/artiq3test4/bin. So I thought artiq_session was still using artiq_corelog. 

I created a fresh conda environment, reinstalled artiq. aqctl_corelog is now where it should be. Resolved.

closed 2017-07-07 by jbqubit