m-labs/artiq #779: generalizing 3U peripheral interfaces

jbqubit, 2017-07-10:
I'd like to begin using the 3U BNC, SMA and RJ45  [EEM](https://github.com/m-labs/sinara/wiki/EEM) boards with ARTIQ. Initially this will be using a KC705. Here's the specific path I will use to start. 

```KC705-> FMC-VHDCI adapter -> VHDCI cable -> VHDCI carrier -> IDC cable-> 3U BNC```

The use of the KC705 is expected to be deprecated in time. This is a specific instance of a more general pattern. 

If __Kasli is brain__: 
```Kasli brain -> 3U interface [B] -> 3U device [A]```

[A] is in {BNC, SMA, RJ45, ... } are [EEM](https://github.com/m-labs/sinara/wiki/EEM) peripherals
[B] is in {IDC cable, backplane}

If __KC705, Metlino or Sayma is brain__:
```Brain [D]-> VHDCI-interface [C] -> VHDCI cable -> VHDCI carrier -> 3U interface [B] -> 3U EEM device [A]```

[C] is in {direct, FMC to VHDCI adapter}
[D] is in {KC705 LPC FMC, KC705 HPC FMC, Metlino VHDCI Port 1, Metlino VHDCI Port 2, Sayma FMC}

Parts of this path that will be reused in different combinations are [A], [B], [C].  Is there an agreed upon way for abstracting interfaces so they can be re-used? Where will they fit into the ARTIQ/Misoc repositories?

closed 2017-07-20 by sbourdeauducq

jbqubit, 2017-07-24:
@sbourdeauducq It there a commit that illustrates the approach you have in mind for this?

sbourdeauducq, 2017-07-25:
This is part of a bigger problem, which will be discussed and dealt with in due time.

jbqubit, 2017-07-25:
Is there an Issue for the bigger problem? If not please reopen this Issue.

sbourdeauducq, 2017-07-25:
There isn't.

sbourdeauducq, 2017-07-25:
But this description you wrote isn't good, and dealing with this is not a priority. Please use the existing APIs to build your own KC705+EEM designs, as I explained in the other issue you opened.

jbqubit, 2017-07-25:
Agreed that the current tools are sufficient to kludge something together that will work in my lab.

I understand that my Issue is a specific case of a broader problem. I also gather that your statement of the problem would be different, likely more general. All the same this is a fine placeholder for the broader problem. I've applied state:needs-discussion.