m-labs/artiq #782: name space for platform.add_extension()

jbqubit, 2017-07-14:
Platform has IO mappings added to it at many levels of the M-Labs gateware build hierarchy including MIGEN, MiSoC and ARTIQ. Namespaces would help keep things sorted. 

An example where this is helpful is when adding peripherals to headers on prototype boards. A user shouldn't have to worry about accidentally colliding with items added to the namespace by lower level layers. An example is "i2c" for KC705 which is added in ```migen/build/platforms/kc705.py``` but may not be the only i2c in a system.

jordens, 2017-07-14:
What do you mean by "namespace", how would the help and what can you do with them that you can't do right now? "i2c" is the only i2c in that kc705 platform. In your platform with more i2c buses, you can call them e.g. "i2c_eem0" and "i2c".

jbqubit, 2017-07-14:
I2C wasn't the best example. Consider instead artiq/gateware/nist_clock.py which creates an "spi". This very nearly collides with migen/build/platforms/kc705.py "spiflash" and "mmc_spi". 

Anyway, since when is it good design to pile things into a single namespace sourced from three repositories?

jordens, 2017-07-14:
Simply because the acronym "i2C" (or "SPI") occurs at all levels and in all repositories! Just because they refer to the same acronym doesn't mean "namespaces" are in order. Anyway, what do you mean by "namespace". And still the same three questions as above.

jbqubit, 2017-07-14:
Thank you for reply. I don't have more to add.

closed 2017-07-14 by jbqubit

npisenti, 2017-07-14:
Perhaps the appropriate way forward here is name mangling -- top-level user io mappings could be conventionally prefaced with a meaningful designator which is orthogonal to the underlying artiq/migen/misoc platform extensions. 

Probably not worth error checking for name collisions, but for the casual physics programmer it's worth noting in any (forthcoming?) documentation of building custom gateware targets that this could be an issue. I think the confusion was compounded by the multiple files/repositories which could potentially contain the conflicting name.

sbourdeauducq, 2017-07-15:
> since when is it good design to pile things into a single namespace sourced from three repositories?

There is the basic Migen platform file that contains the standard set of peripherals that come with the board, and then the ARTIQ platform file adds extensions that correspond to the boards developed by physicists (NIST QC2 etc.). Where is the problem with that? MiSoC itself is not adding extensions.