m-labs/artiq #783: can't build firmware

jbqubit, 2017-07-14:
https://gist.github.com/jbqubit/753141b87624ec79d572229b09b2d2c5

jordens, 2017-07-14:
```
artiq                     3.0.dev         py_1094+git91ad2bc6    m-labs/label/dev
artiq-dev                 3.0.dev         py_1075+gitfdb4e3f8cfad    m-labs/label/dev
```
those are old.

whitequark, 2017-07-14:
But the bug exists separately. It's related to ARTIQ firmware wishing to know the hash of the git checkout.

jordens, 2017-07-14:
```conda create -c m-labs/label/dev -n artiq-dev-test artiq-dev``` works fine here.

jbqubit, 2017-07-14:
Conda was stuck on an old version artiq-dev package. That is, conda update didn't work. Nor did conda install=version-no. So, erased ~/anaconda3 and reinstalled anaconda from scratch. And addressed #785. Now I can build. Closing.

closed 2017-07-14 by jbqubit