m-labs/artiq #784: conda asyncserial missing

jbqubit, 2017-07-14:
New install of anaconda. New environment.
```
$ conda install artiq-dev 
```

```
$ python setup.py develop
/home/britton/anaconda3/envs/a3t8/lib/python3.5/site-packages/setuptools/dist.py:333: UserWarning: Normalizing '3.0.dev+1228.g6aa8e144' to '3.0.dev0+1228.g6aa8e144'
  normalized_version,
running develop
running egg_info
writing namespace_packages to artiq.egg-info/namespace_packages.txt
writing dependency_links to artiq.egg-info/dependency_links.txt
writing top-level names to artiq.egg-info/top_level.txt
writing entry points to artiq.egg-info/entry_points.txt
writing artiq.egg-info/PKG-INFO
writing requirements to artiq.egg-info/requires.txt
reading manifest file 'artiq.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'artiq.egg-info/SOURCES.txt'
running build_ext
Creating /home/britton/anaconda3/envs/a3t8/lib/python3.5/site-packages/artiq.egg-link (link to .)
Adding artiq 3.0.dev0+1228.g6aa8e144 to easy-install.pth file
Installing artiq_coreanalyzer script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_flash script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_ctlmgr script to /home/britton/anaconda3/envs/a3t8/bin
Installing aqctl_novatech409b script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_master script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_coreboot script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_rpctool script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_coreconfig script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_influxdb script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_corelog script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_mkfs script to /home/britton/anaconda3/envs/a3t8/bin
Installing aqctl_korad_ka3005p script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_devtool script to /home/britton/anaconda3/envs/a3t8/bin
Installing aqctl_pdq script to /home/britton/anaconda3/envs/a3t8/bin
Installing aqctl_thorlabs_tcube script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_run script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_compile script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_client script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_pcap script to /home/britton/anaconda3/envs/a3t8/bin
Installing aqctl_corelog script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_session script to /home/britton/anaconda3/envs/a3t8/bin
Installing aqctl_lda script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_browser script to /home/britton/anaconda3/envs/a3t8/bin
Installing artiq_dashboard script to /home/britton/anaconda3/envs/a3t8/bin

Installed /home/britton/github/jbqubit/artiq
Processing dependencies for artiq==3.0.dev0+1228.g6aa8e144
Searching for pythonparser
Reading https://pypi.python.org/simple/pythonparser/
Downloading https://pypi.python.org/packages/db/dc/9af2ec3dcf2df883cc3fba8b17ec1225909afd4db17ca4c7249f4c6c0a03/pythonparser-1.1.tar.gz#md5=0c0075607f0290cf7b19128066a68d7c
Best match: pythonparser 1.1
Processing pythonparser-1.1.tar.gz
Writing /tmp/easy_install-gtx0f60k/pythonparser-1.1/setup.cfg
Running pythonparser-1.1/setup.py -q bdist_egg --dist-dir /tmp/easy_install-gtx0f60k/pythonparser-1.1/egg-dist-tmp-y2kef8d1
zip_safe flag not set; analyzing archive contents...
pythonparser.coverage.__pycache__.__init__.cpython-35: module references __file__
creating /home/britton/anaconda3/envs/a3t8/lib/python3.5/site-packages/pythonparser-1.1-py3.5.egg
Extracting pythonparser-1.1-py3.5.egg to /home/britton/anaconda3/envs/a3t8/lib/python3.5/site-packages
Adding pythonparser 1.1 to easy-install.pth file

Installed /home/britton/anaconda3/envs/a3t8/lib/python3.5/site-packages/pythonparser-1.1-py3.5.egg
Searching for asyncserial
Reading https://pypi.python.org/simple/asyncserial/
Couldn't find index page for 'asyncserial' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.python.org/simple/
No local packages or working download links found for asyncserial
error: Could not find suitable distribution for Requirement.parse('asyncserial')

```

jbqubit, 2017-07-14:
Why not include asyncserial in conda package artiq-dev?

sbourdeauducq, 2017-07-15:
conda install asyncserial

closed 2017-07-15 by sbourdeauducq

sbourdeauducq, 2017-07-15:
asyncserial (and pythonparser, which setup.py is also installing itself, albeit successfully) are already dependencies of conda-dev. Please look at the source before proposing a solution. Likely you are using conda wrong, or there is a conda bug.

jbqubit, 2017-07-17:
I reported the Issue because I had seen the message 'Couldn't find index page for 'asyncserial' (maybe misspelled?)' for several weeks. I upgraded conda to 4.3.22 and now the error message is gone. So you're right, it was a conda bug.