m-labs/artiq #785: conda version dependency not in meta.yaml

jbqubit, 2017-07-14:
Found an error in installing misoc that was resolved by updating conda. This error was that conda listed the package as being installed but there was no corresponding directory or egg at ```$ ls ~/anaconda3/envs/a3t9/lib/python3.5/site-packages/```

```
conda:       4.1.6-py35_0      --> 4.3.22-py35_0 
conda-env:   2.5.1-py35_0      --> 2.6.0-0       
anaconda:    4.1.1-np111py35_0 --> custom-py35_0 
```
artiq-dev meta.yaml does not specify a conda version dependency. Shouldn't there be an explicit version dependency for conda?

whitequark, 2017-07-14:
We are not necessarily aware of every minor bug in conda, of which there are many. It is always a good idea to keep conda updated.

jbqubit, 2017-07-14:
Adding the conda version constraint will save everybody time as it will flag an outdated version of conda. What's the problem with doing that in meta.yaml?

sbourdeauducq, 2017-07-15:
> it will flag an outdated version of conda

This is a strong assumption considering the average density of bugs in conda. I see this as something that provides only a very minor benefit - as whitequark points out it is a good idea to just keep conda updated - and as something that may either do nothing or tickle conda bugs that break other things.
And what should the version constraint be, exactly? When were the bugs fixed? Doing this requires more work than you think.

closed 2017-07-15 by sbourdeauducq

jbqubit, 2017-07-17:
The meta.yaml version should match that used by the M-Labs bb. Besides nagging users to update conda, in the event that the conda team introduces a bug with a conda upgrade the meta.yaml can prevent users from upgrading to a buggy version.

sbourdeauducq, 2017-07-17:
Have you tested it? When the user has already installed another version of conda, and then installs ARTIQ, does conda upgrade/downgrade itself *and then* installs ARTIQ? If not, then it won't work as you propose.

jbqubit, 2017-07-17:
>Adding the conda version constraint will save everybody time as it will flag an outdated version of conda. 

You're right. I'm making assumption about how conda would respond if my conda version was 4.3.21 and I edited artiq-dev/meta.yaml as follows

```
requirements:
  run:
    - conda >=4.3.22
```
I don't know how to tell conda to use a local version of meta.yaml so I've not checked this. I do know the root environment has to be selected for conda to upgrade itself. So automatic upgrade of conda seems unlikely but it might flag the failed requirement which would be a useful flag to the user.

jbqubit, 2017-07-24:
Alternately, consider updating installing.rst to include the following. 

> On a system with a pre-existing conda installation, it is advised to update to the latest versions of ```conda``` and ```setuptools``` prior to installing ARTIQ.