m-labs/artiq #790: DRTIO switch support for complex Kasli trees

sbourdeauducq, 2017-07-20:


sbourdeauducq, 2017-10-01:
Funded by Oxford.

sbourdeauducq, 2018-09-20:
done

closed 2018-09-20 by sbourdeauducq

hartytp, 2018-11-07:
Testing switching out on a Kasli->Kasli->Kasli system. (`RoutingTable { 0: 0; 1: 1 0; 2: 1 1│
 0; }`). Using the switching125 branch with misoc patched to revert the latest CPU speed reduction and using a -3 speed grade Kasli for the master.

hartytp, 2018-11-07:
Posting detailed testing notes here so I have them to refer back to in the future. Mute if you're not interested!

Checking that the link comes up reliably on each power cycle, and that link loss is correctly reported, and link reestablished when any Kasli is power cycled.

Power cycling the leaf node seems fine (measured over ~10 power cycles). Master reports
```
[  1336.642778s]  INFO(runtime::rtio_mgt::drtio): [DEST#2] destination is down
[  1345.262540s]  INFO(runtime::rtio_mgt::drtio): [DEST#2] destination is up
[  1345.267965s]  INFO(runtime::rtio_mgt::drtio): [DEST#2] buffer space is 128
```
Repeater reports
```
[   988.940721s] ERROR(satman::repeater): [REP#0] received packet of an unknown type
[   988.946753s]  WARN(satman::repeater): [REP#0] aux packet error
[   988.952578s]  INFO(satman::repeater): [REP#0] link is down
[   995.707509s]  INFO(satman::repeater): [REP#0] link RX became up, pinging
[   998.038740s]  INFO(satman::repeater): [REP#0] remote replied after 24 packets
```

On power cycling the repeater, master reports
```
[  1493.022498s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] destination is down
[  1493.029365s]  INFO(runtime::rtio_mgt::drtio): [DEST#2] destination is down
[  1500.647700s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link RX became up, pinging
[  1502.667330s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] remote replied after 11 packets
[  1502.811600s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link initialization completed
[  1502.818409s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] destination is up
[  1502.824772s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] buffer space is 128
[  1502.831973s]  INFO(runtime::rtio_mgt::drtio): [DEST#2] destination is up
[  1502.838339s]  INFO(runtime::rtio_mgt::drtio): [DEST#2] buffer space is 128
```
Repeater reports
```
[     0.000004s]  INFO(satman): ARTIQ satellite manager starting...
[     0.005617s]  INFO(satman): software ident 4.0.dev+1501.g9f96b6bc;lab3satellite
[     0.012936s]  INFO(satman): gateware ident 4.0.dev+1501.g9f96b6bc;lab3satellite
[     0.296349s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[     2.263755s]  INFO(board_artiq::si5324):   ...locked
[     2.339505s]  INFO(satman::repeater): [REP#0] link RX became up, pinging
[     2.748188s]  INFO(satman): uplink is up, switching to recovered clock
[     2.781939s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[     4.508187s]  INFO(board_artiq::si5324):   ...locked
[     4.619269s]  INFO(board_artiq::si5324::siphaser): nshifts to 0->1 siphaser transition: 157 (126deg)
[     4.643189s]  INFO(satman::repeater): [REP#0] remote replied after 5 packets
[     4.780116s]  INFO(satman): TSC loaded from uplink
[     4.907731s]  INFO(satman): rank: 1
[     4.909835s]  INFO(satman): routing table: RoutingTable { 0: 0; 1: 1 0; 2: 1 1 0; }
```

sbourdeauducq, 2018-11-07:
When you shut down the repeater, it probably sends glitches/bogus data. The error reporting function is only run while the link is up, which is why you are seeing the message with a delay.

hartytp, 2018-11-07:
ack. Anyway, so far I haven't managed to break anything which is always a good sign...

Cycling the master gives this on the master
```
[     4.357628s]  INFO(runtime::session): accepting network sessions
[     4.372381s]  INFO(runtime::session): running startup kernel
[     4.376900s]  INFO(runtime::session): no startup kernel found
[     4.382640s]  INFO(runtime::session): no connection, starting idle kernel
[     4.389535s]  INFO(runtime::session): no idle kernel found
[     4.394882s]  INFO(runtime::rtio_mgt::drtio): [DEST#0] destination is up
[     4.601420s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link RX became up, pinging
[     6.419607s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] remote replied after 10 packets
[     6.536546s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link initialization completed
[     6.543247s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] destination is up
[     6.549711s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] buffer space is 128
[     6.556893s]  INFO(runtime::rtio_mgt::drtio): [DEST#2] destination is up
[     6.563269s]  INFO(runtime::rtio_mgt::drtio): [DEST#2] buffer space is 128
```
and this on the repeater
```
[   775.631235s]  INFO(satman): uplink is down, switching to local crystal clock
[   775.665511s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[   777.460642s]  INFO(board_artiq::si5324):   ...locked
[   787.614416s]  INFO(satman): uplink is up, switching to recovered clock
[   787.648168s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[   789.376684s]  INFO(board_artiq::si5324):   ...locked
[   789.653494s]  INFO(board_artiq::si5324::siphaser): nshifts to 0->1 siphaser transition: 485 (389deg)
[   789.841861s]  INFO(satman): TSC loaded from uplink
[   789.942232s]  INFO(satman): rank: 1
[   789.944345s]  INFO(satman): routing table: RoutingTable { 0: 0; 1: 1 0; 2: 1 1 0; }
```

sbourdeauducq, 2018-11-07:
Looks normal.

hartytp, 2018-11-07:
> Looks normal.

Yes, I'm just reporting what I see so that I have a point of reference to come back to if there is any issue later on.

hartytp, 2018-11-07:
- [x] timing is deterministic between SERDES and simple TTLs both on the master
- [ ] timing is deterministic between SERDES TTL on the master and SERDES/simple TTLs on the repeater/leaf

jordens, 2018-11-07:
As another data point I had also tested the switching branch with 125 MHz RTIO (just master-satellite with DIO, without another repeater through) a while back and ran it through the usual stress testing and determinism tests and found no additional issues.
Might want to check the more complicated phys though.

hartytp, 2018-11-08:
Timing of SERDES/simple TTLs is fully deterministic across multiple power cycles. Latency is in line with expectations:
- Master SEDES to master simple 8ns
- Slave SERDES to master SERDES 292ns
- repeater SERDES to master SERDES 567ns

Good!

hartytp, 2018-11-08:
NB those numbers include some non-negligible fibre delays.

hartytp, 2018-11-09:
Dumb throughput test

```python
import numpy as np

from artiq.experiment import *

class TTLTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl0") # master SERDES
        self.setattr_device("ttl4") # master simple
        self.setattr_device("rttl0") # repeater SERDES
        self.setattr_device("rttl4") # repeater simple
        self.setattr_device("lttl0") # leaf SERDES
        self.setattr_device("lttl4") # leaf simple

    @kernel
    def run(self):
        self.core.reset()
        self.lttl0.output()

        num_pts = 100
        taus = np.full(num_pts, 0.)

        for i in range(num_pts):
            tau = 10*us
            while True:
                tau_mu = self.core.seconds_to_mu(tau)
                print(i, ": trying tau", tau/us, "us")

                self.core.break_realtime()
                try:
                    for _ in range(100000):
                        self.lttl0.pulse_mu(tau_mu)
                        delay_mu(tau_mu)
                except Exception:
                    print("underflow")
                    taus[i] = tau
                    break
                tau = 0.9*tau
        self.print_stats(taus)

    def print_stats(self, taus):
        print(taus)
        print("max", np.max(taus))
        print("min", np.min(taus))
        print("mean", np.mean(taus))
```

Mean value of `tau` (minimum delay for toggling a TTL without causing underflows) is approx 580ns. Max value I measured was about 720ns.

sbourdeauducq, 2018-11-09:
Try it on new, it should be faster.