m-labs/artiq #792: IOSERDES TTLs for Kintex Ultrascale

sbourdeauducq, 2017-07-20:


hartytp, 2018-12-26:
@sbourdeauducq is this funded? I'm currently planning to use this for Sayma synchronisation, so it would be good to have it finished before v2.0 hw arrives.

hartytp, 2018-12-26:
> I'm currently planning to use this for Sayma synchronisation

Actually, I take that back. So long as we put the SYSREF generator on the RTM FPGA (which seems to be the logical place for it) I won't need KU SERDES TTL.

sbourdeauducq, 2018-12-27:
Yes, this is fully funded.
The code for the IOSERDES is there but often causes timing failures due to the Ultrascale TPWS flaw. The divided and high-speed clocks need to be generated from a pair of BUFGCE_DIV or a pair of BUFG_GT, the complicated part is that the divided clock has to be aligned to the existing RTIO clock.

There are two possible workarounds:
1. Generate clocks from the DRTIO transceiver and put the dividing BUFG_GT into the GTH clock alignment path. According to Xilinx tech support, the GTH reportedly supports such an external cock division. This is very resource-efficient and needs only a few extra elements, but needs a bit more careful testing to avoid introducing transceiver regressions. Also, the transceiver clock output is limited to 511.719MHz, so the TTL resolution is limited to 977ps (DDR) - it won't work in 8X mode at 150MHz RTIO frequency (But 125MHz 8X and 150MHz 4X are fine).
2. Use an extra MMCM and BUFGCE_DIV to implement RTIOClockMultiplier (same as on the Kasli target). We then have to measure the phase between the output of the dividing BUGCE_DIV and the input of the MMCM, and phase-slip the MMCM or the BUGCE_DIV until they are aligned. Unlike with the transceiver, it is not possible to put the BUFGCE_DIV into the MMCM feedback path. I don't know if this is a silicon limitation, but Vivado craps out, and Xilinx tech support seem to have no idea what I am talking about. So we need to measure the phase manually, and slip either the MMCM or the BUFGCE_DIV. Slipping the MMCM is tricky to do without introducing more timing uncertainty (since the slips are so small); slipping the BUFGCE_DIV makes more sense since this is where the phase non-determinism comes from and also the granularity is higher, so there won't be any additional timing uncertainty. But the BUFGCE_DIV control interface is another example of trashy Ultrascale design, and I'm not quite sure how to use it without async bugs, nor if it is possible at all. Xilinx seem to have no understanding of synchronous systems here (for example, they insisted - over multiple emails - that the MMCM LOCKED signal is fully asynchronous, but by some magic can reliably sync multiple BUFGCE_DIV when connected as per their docs. Go figure.)

sbourdeauducq, 2019-01-04:
done, needs testing

hartytp, 2019-01-04:
:+1: thanks!

sbourdeauducq, 2019-01-06:
```python
    while True:
            for i in range(4):
                self.ttl.pulse_mu(8+2*i)
                delay_mu(8)
            delay(1*ms)
```
[image]

closed 2019-01-06 by sbourdeauducq

jbqubit, 2019-01-08:
That's great!