m-labs/artiq #793: 5Gbps/40x DRTIO for Sayma/Metlino

sbourdeauducq, 2017-07-20:
We will be implementing scheme B in https://github.com/m-labs/sinara/wiki/SinaraClocking.

sbourdeauducq, 2017-10-08:
@jordens what is the final clocking plan and DRTIO transceiver speed decision?

jordens, 2017-10-09:
With `f_data = 600 MHz` and 4x DAC interpolation to 2.4 GHz, from the `0.2 f_data` transition skirt width of the AD9154 interpolation we have:

* for 250 MHz max output, we have 0.05 dB ripple, 47 dB suppression (of the Nyquist image at 350 MHz)
* for 260 MHz output we have 0.1 dB ripple and 37 dB Nyquist suppression of 340 MHz.

With the `f_0` carrier at 200 MHz, 150 MHz `f_RTIO_coarse`, and the `0.2 f_RTIO_coarse` transition skirt width of the `f1/f2` interpolator  of the SAWG, we can cover 140 to 260 MHz (including the design 150-260 MHz) without incurring any penalty there and without any Nyquist images in the passband.

This is both the most economical and the most performant solution. 6 Gbit/s JESD204B lanes and `f_DRTIO`, 150 MHz `f_RTIO_coarse`.

hartytp, 2017-10-09:
@jordens @jbqubit Last lime we spoke about this, I though we agreed to go for:
- f_DAC = 2GHz
- f_DATA = 1GHz (2x interpolation)
- f_RTIO_coarse = 125MHz

I'd be keen to stick to that plan, as it gives us quite a bit more base-band bandwidth.

sbourdeauducq, 2017-10-09:
@jordens Again, the backplane for DRTIO is limited to 5G. So that would be 3beG, not 6.

jordens, 2017-10-09:
@sbourdeauducq ack.

@hartytp For the 150-260 MHz use case 600 MHz f_data has the best performance (200 MHz +- 60 MHz from f1/f2 without the AA filter kicking in) and is easier to implement since it is resource saving. You'd need to discuss with @k-beck, @TheCakeIsAPi, and @jbqubit .

hartytp, 2017-10-09:
> or the 150-260 MHz use case 600 MHz f_data has the best performance (200 MHz +- 60 MHz from f1/f2 without the AA filter kicking in)

What performance issues in particular are you concerned with? Do you mean spurs/SFDR? Can you quantify?

I would have thought that the 150-260MHz use case would work excellently with a 1GHz f_DATA. The higher data rate is an advantage since it gives us more bandwidth for other uses cases -- this is certainly something that we would really want.

> and is easier to implement since it is resource saving. 

Is it really that much easier to implement? Isn't it just a matter of reconfiguring the cores? Or, are you worried that it won't fit on the FPGA/meet timing?

jordens, 2017-10-09:
@hartytp In the old minutes there was a statement about 1 GHz. I still thing 600 MHz is better for the reasons mentioned above.
I'll let the users decide (@k-beck, @TheCakeIsAPi, and @jbqubit).
If there is no need for tones that widely spaced (i.e. if having them closer, max +-50 MHz, together is fine) and if the ~58 dB Nyquist suppression from the AA filter is fine as well, then the 1GHz will work for them.

jordens, 2017-10-09:
They were worried about spur free dynamic range within 150-260 MHz. Since the AA filter only suppresses to 58 dB, that will be limiting for f_RTIO=125 MHz.
It is a factor of 1.5-2 in resource usage. Yes. That's much easier. And PnR goes much worse than linear. And timing issues increase.

hartytp, 2017-10-09:
> They were worried about spur free dynamic range within 150-260 MHz. Since the AA filter only suppresses to 58 dB, that will be limiting for f_RTIO=125 MHz.

> If there is no need for tones that widely spaced (i.e. if having them closer, max +-50 MHz, together is fine) and if the ~58 dB Nyquist suppression from the AA filter is fine as well, then the 1GHz will work for them.

I need to go back and think about how the phaser/gateware works before commenting on this.

> Yes. That's much easier. And PnR goes much worse than linear. And timing issues increase.

ACK, this is harder. But IMO, f_data=1GHz is an important use case to support. So, if it doesn't fit/meet timing then I'd consider that a problem. If it fits and meets timing then I'd argue that the extra resource usage isn't an argument against this if it's better from the users' perspective.

jordens, 2017-10-09:
I don't see a roadblock. It will just take longer and more fiddling/trial and error to deliver a 1 GHz f_data bitstream than 600 MHz. And once there are problems with e.g. DRTIO there will be a need to exclude electrical issues and hence the need for a slower design anyway.

sbourdeauducq, 2017-10-09:
Can we just do the 600 MHz for now? There have been way enough feature explosions and delays already.

jordens, 2017-10-09:
That's what I would like to do as well.

jbqubit, 2017-10-12:
Sounds good to me.

enjoy-digital, 2017-12-01:
DRTIO seems to be working fine with 150Mhz refclk / 3Gbps linerate.

sbourdeauducq, 2017-12-02:
@enjoy-digital Did you implement the multi-lane initialization procedure? Section "Using TX Buffer Bypass in Multi-Lane Auto Mode" (or "Manual Mode", I let you choose) of the GTH manual.

sbourdeauducq, 2017-12-31:
Got a DRTIO link between Sayma1 (SFP1) and Sayma3 (SFP0) using copper. Seems reliable, though it has only received limited testing since the development cycles on Sayma are so tedious (1V8 bug killing the boards often and at random times, long bitstream loading times, no Ethernet).
The link between SFP0 on Sayma1 and SFP0 on Sayma2 only works in one direction, possibly due to a hardware problem.

enjoy-digital, 2017-12-31:
@sbourdeauducq: ok good first step. As discussed, i'll finish multi-link initialization and do the tests soon.

gkasprow, 2018-01-02:
@sbourdeauducq I can give you a project that I used to test the SFP performance.

sbourdeauducq, 2018-02-26:
> The link between SFP0 on Sayma1 and SFP0 on Sayma2 only works in one direction, possibly due to a hardware problem.

...or problem with the transceiver electrical settings (may need copying more values from the wizard like we did to fix similar issues before).

gkasprow, 2018-02-26:
I run successfully  IBERT test with 2 SFPs linked by copper SFP cable. It works at 10Gbit without errors.

gkasprow, 2018-02-26:
I can share bit file which does such test.

gkasprow, 2018-02-26:
@sbourdeauducq did you try with copper SFP-SFP cable

sbourdeauducq, 2018-02-26:
> using copper.

Yes, see above.

sbourdeauducq, 2018-02-27:
Broken down into #933 and #934.

closed 2018-02-27 by sbourdeauducq