m-labs/artiq #794: Clocking, DAC support and JESD synchronization on one Sayma card

sbourdeauducq, 2017-07-20:


sbourdeauducq, 2018-04-30:
On my test:
```
[     6.513475s]  INFO(board_artiq::ad9154):   phase min: 58, phase max: 121, phase opt: 89
[     9.581414s]  INFO(board_artiq::ad9154):   phase min: 58, phase max: 121, phase opt: 89
```

sbourdeauducq, 2018-04-30:
```
[     7.734548s]  INFO(board_artiq::ad9154):   phase: 0, sync error: 0
...
[     8.603360s]  INFO(board_artiq::ad9154):   phase: 58, sync error: 481
...
[     9.576190s]  INFO(board_artiq::ad9154):   phase: 122, sync error: 482
```

enjoy-digital, 2018-04-30:
Everything should be implemented, before closing we need to:
- [ ] Get HMC830 working.
- [ ] Test JESD SC1 on various boards, note HMC7043 sysref phase results.
- [ ] Change the firmware to use a fixed HMC7043 sysref phase.

hartytp, 2018-04-30:
@enjoy-digital I'll try to have a go at some of that tomorrow.

Is the phase scan implemented in the code by default atm?

hartytp, 2018-04-30:
Thanks for all the work you've done on this recently. Things are starting to shape up nicely.

enjoy-digital, 2018-04-30:
@hartytp: yes the phase scan is implemented in the code. If you do a test, can you post your results here? (it will allow us to know if we case use same values for all boards).

hartytp, 2018-04-30:
Will do.

What are you doing for the SC1 test? Just looking for DAC realignment events by reading out the register values, or looking at RF outputs on a scope?

enjoy-digital, 2018-04-30:
We modify the phase until having at least 2 realignments. (since we are not sure to have the full scan for the first one) Then we put the phase in the middle of 2 realignements and trigger another sync on the DAC.  I don't have the equipment to test SC1 on the RF output. If you have it and can do the test, we'll be happy to have the results :)

hartytp, 2018-04-30:
Okay, I'll start with your phase scan and see how I get on. If things go well, I'll have a look on my scope.

sbourdeauducq, 2018-05-01:
> I don't have the equipment to test SC1 on the RF output.

You do; even a cheap DSO can do it with some signal processing. Here are some hacky scripts I was using:
```
import asyncio
import ds1xxxz
import os
import time

def main():
    for i in range(30):
        os.system("echo > /dev/ttyACM0")
        time.sleep(6)
        os.system("sh load_rtm")
        time.sleep(60)
        print("getting waveforms", i)
        ds1xxxz.save_waveforms("192.168.1.132", 5555, "waveforms_{:02d}.npz".format(i))
    asyncio.get_event_loop().close()

if __name__ == "__main__":
    main()
```

```
import sys
import numpy as np
import matplotlib
import matplotlib.pyplot as plt
from scipy import signal, constants

nrows = 10
ncols = 3

matplotlib.rcParams.update({'font.size': 6})

for i in range(30):
    filename = "waveforms_{:02d}.npz".format(i)
    globals().update(np.load(filename))
    t = np.arange(y1.shape[0])*x[0] + x[1]
    y = np.c_[y1, y2].T
    z = signal.decimate(y*np.exp(1j*2*np.pi*50e6*t), q=100, ftype="fir", zero_phase=True)[:, 100:]
    z = signal.decimate(z, q=100, ftype="fir", zero_phase=True)[:, 100:]
    angle = np.angle(np.mean(z[0]*z[1].conj()))

    plt.subplot(nrows, ncols, i+1).set_title("{} {:.4f}".format(filename, angle))
    plt.plot(t[:120], y[:, :120].T)
    plt.savefig("sayma.pdf")
plt.show()
```

enjoy-digital, 2018-05-01:
OK thanks, maybe i have the equipment, but you are asking me to do system level tests while i'm not relevant for this (without your previous answer i would have no idea how to test it). So let's keep every one its job:
- AD9154 is reporting synchronizing in JESD SC1 and  we seems are able to set the sysref correctly.
- Someone that is relevant to set up a system level test has to create a test and reports problems if any. @hartytp suggested to do that and is a lot more relevant than me to says if it's working at system level, so let's wait the first results. And if things are reported not working, of course i'll try to understand and fix.

sbourdeauducq, 2018-05-01:
if you never try things you have no idea about, you never learn :P

hartytp, 2018-05-01:
I'm happy to have a go at this once my Sayma arrives back, which I hope will be today (@gkasprow do you have a tracking number for it btw?)

To confirm, the two DACs are should currently have deterministic latency with respect to the input 1.2GHz clock, but with respect to the RTIO clock, right? So, if I set channels on different DACs to the same frequency and phase then the phase offset between them should be constant, right? Similarly, if I set the DAC output to a sub-multiple of the 1.2GHz clock frequency, I should see a constant phase delay between the RF output and the DAC clock, right?

enjoy-digital, 2018-05-01:
@sbourdeauducq: sure, that's also what i think and how i learn. But that's not something you can always do. I'll be happy to learn on that, but maybe later, for now i'm just trying being practical since i have limited time and would rather focus on things i'm really relevant for.

sbourdeauducq, 2018-05-01:
@hartytp just compare 2 DAC channels from each DAC chip. The rest is not necessarily det-lat yet.

hartytp, 2018-05-01:
ack

sbourdeauducq, 2018-05-01:
No, I meant comparing *between* the two DAC chips. Take one channel from AD9154-0 and one channel from AD9154-1.

hartytp, 2018-05-01:
Okay, good. I'll do that once my AMC+RTM turn up (tomorrow?).

sbourdeauducq, 2018-05-02:
@enjoy-digital The phase shift value definitely needs to be fixed; I'm getting intermittently 56 or 88 for DAC-1 of one board.

sbourdeauducq, 2018-05-02:
Also the logs are excessively verbose, it would be sufficient to print only when the error count is first measured and then changes.

enjoy-digital, 2018-05-09:
I reduced the logs.

Now for the intermittently results, let's try to analyze that:
- coarse delay resolution is 416ps (with HMC7043 input  at 1.2GHz)
- analog delay resolution is 25ps. (so 16 taps = 400ps)

Combined together, we can see this as 26ps (416ps/16) resolution configurable delay . (This no really what is happening since we have a gap between between 400 and 416 ps, but we use that for the analysis).

So when testing on board:
- found interval is always 63 or 64 taps = 1664ps (~= DAC_CLK period)
- found ideal values are always 88 or 56. 88-56=32 taps = 832ps (~= 1/2 DAC_CLK period)

Not sure what to conclude from that now, at least i see that theses intermittently results are not coming from corner cases in the sysref scan, but are the behaviour of the system.

sbourdeauducq, 2018-05-25:
DAC synch is not working at all; the DACs just have random phases wrt each other after each power-up.
This can be measured with the script below that uses the Red Pitaya currently connected in the lab, and sines.py modified to output 9MHz (currently flashed as startup kernel).
From what I observed, the phase varies within a ~0 to ~0.4 radian window of the 9MHz waveform, i.e. ~70ns.
```python
import socket
import numpy as np
import matplotlib.pyplot as plot
from scipy import signal, constants

class RPSCPI:
    def connect(self, host):
        self.sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
        self.sock.connect((host, 5000))
        self.sock_f = self.sock.makefile()

    def close(self):
        self.sock_f.close()
        self.sock.close()

    def sendmsg(self, msg):
        self.sock.send(msg.encode() + b"\r\n")

    def recvmsg(self):
        return self.sock_f.readline().strip()

    def trigger(self):
        rp.sendmsg("ACQ:START")
        rp.sendmsg("ACQ:TRIG NOW")
        while True:
            rp.sendmsg("ACQ:TRIG:STAT?")
            if rp.recvmsg() == "TD":
                break

    def get_data(self, channel):
        rp.sendmsg("ACQ:SOUR{}:DATA?".format(channel))
        buff_string = rp.recvmsg()[1:-1].split(',')
        return np.array(list(map(float, buff_string)))

rp = RPSCPI()
rp.connect("192.168.1.200")
try:
    rp.trigger()
    y1 = rp.get_data(1)
    y2 = rp.get_data(2)

    t = np.arange(y1.shape[0])/125e6
    y = np.c_[y1, y2].T
    z = signal.decimate(y*np.exp(1j*2*np.pi*9e6*t), q=10, ftype="fir", zero_phase=True)[:, 10:]
    z = signal.decimate(z, q=10, ftype="fir", zero_phase=True)[:, 10:]
    angle = np.angle(np.mean(z[0]*z[1].conj()))

    print(angle)

    # optional, check visually that we are not measuring noise
    plot.plot(y1)
    plot.plot(y2)
    plot.show()

finally:
    rp.close()
```

hartytp, 2018-05-25:
I think that's expected with the current state of the code @sbourdeauducq since we don't issue a reset command after configuring the dividers.

From a quick skim over the data sheet, I think that to enable synchronisation you need to edit these lines to enable synchronisation for all channels by setting bit 6 high:

https://github.com/m-labs/artiq/blob/db4d1878d3528d65b9cace5a151030fa125648a2/artiq/firmware/libboard_artiq/hmc830_7043.rs#L261-L266

should be
```
if enabled {
                // Only clock channels need to be high-performance
                if (channel % 2) == 0 { write(channel_base, 0xD1); }
                else { write(channel_base, 0x51); }
            }
            else { write(channel_base, 0x10); }
```

cf page 39 of the data sheet.

Then, send a sync request via spi as the last part of the init.

hartytp, 2018-05-25:
Maybe I'm missing something here, but I thought that @enjoy-digital's phase adjustment algorithm gave reproducible results across power-cycles. Yet, from looking at the code, I think the output channels should random (n/1.2GHz) phase offsets between power cycles. Not sure what's up there, but should be obvious when looked at with a fast scope.

hartytp, 2018-05-26:
@sbourdeauducq So, do you think that the issue is now with the phase finding code?

In any case, I don't think there is much point me looking at this further until I have access to the hw.

e.g. if I were debugging this, my next step would be to physically check that after reseeding the HMC7043 outputs all have deterministic relative phases. If the issue is still outstanding when I get my hands on working HW, I can perform some quick measurements to help you track this issue down.

hartytp, 2018-05-26:
(Or, @enjoy-digital is welcome to come to our labs and test this using our T & M equipment next time he's in the UK).

sbourdeauducq, 2018-05-26:
The phase scan code is not configuring any chip based on the results. I was just trying to correlate the two, as I am surprised by how non-deterministic the phase scanning code is, too.

hartytp, 2018-06-04:
To look at this, I changed the FPGA_CLOCK divider to 12 (100MHz output) and looked at J61 on a fast scope triggered from my 100MHz reference. I can confirm that the HMC7043 configuration currently used in ARTIQ master does not provide deterministic latency. I'll apply the patch I proposed above and recheck.

hartytp, 2018-06-04:
Nope, even with that patch, the 7043 outputs don't have deterministic latency w.r.t. the reference.

hartytp, 2018-06-04:
Will dig into this further tomorrow.

sbourdeauducq, 2018-06-04:
Great, thanks for all your help!

jbqubit, 2018-06-04:
Analog Devices talks about deterministic latency of HMC7043 this in a report on a huge clock tree.

http://www.analog.com/en/technical-articles/synchronizing-sample-clocks-of-a-data-converter-array.html

hartytp, 2018-06-04:
Joe I'm being daft and measuring the wr on thing. That was never going to work as I was measuring ref to hmc output phase which we can't control. Should have measured phase between hmc7043 outputs!

hartytp, 2018-06-04:
Well can't = can't via SPI alone

hartytp, 2018-06-05:
One more time with brain attached. Now looking at the two 150MHz outputs from the HMC7043. I can confirm that the relative phases of these outputs is not stable across FPGA loads with the current ARTIQ master.

I'll apply the patch and see if that fixes it.

hartytp, 2018-06-05:
Patch doesn't fix it because we also need to configure the internal SYSREF timer. Doing that now.

hartytp, 2018-06-05:
The relative phases of the HMC7043 outputs are fixed as of https://github.com/m-labs/artiq/pull/1049

sbourdeauducq, 2018-06-05:
SC1 now seems to work intermittently. Maybe the desynch is simply due to the JESD204 elastic buffer bug. Here are the phase differences in a 9MHz waveform that I measured between the two DACs, reloading the bitstream every time:
-3.0122085695401912
0.003484989809197922
0.0035013579182848783
0.003437719904394562
0.0034271564472742777
-3.012130036638558
0.0034841246460592993
0.0035376824528900223
0.0035230510713831747
0.003476287717748864
3.0188539470060625
0.0034901997337270447
-3.012092422739266
3.0190223329751356
-3.0119953449379806

There seems to be a resurgence of serwb bugs, which makes this testing a bit annoying.

sbourdeauducq, 2018-06-05:
After JESD update:

0.379348417299257
**0.3321831450064115**
0.3793770410950291
0.37937666640012396
0.37935438836814794
0.3793441872220443
0.3793281246479074
0.3794722930935313
0.37942851699262903
**0.0036921377386391853**
0.3793404124933203

Observation: the difference between the second result and the usual value is 831ps, which is close to the 833ps period of the 1.2GHz clock.

jbqubit, 2018-06-08:
That’s great!

hartytp, 2018-06-10:
Looking at this on my Sayma.

Running the following startup kernel:

```python
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.sawgs = [self.get_device("sawg{}".format(i)) for i in range(8)]

    @kernel
    def run(self):
        self.core.reset()
        for sawg in self.sawgs:
            sawg.reset()
            delay(300 * us)

        for sawg in self.sawgs:
            sawg.frequency0.set(10*MHz)
            sawg.amplitude1.set(1.)
            delay(10*us)
```

Looking at the phase between 1 channel on each DAC using a fast scope. NB 1 deg at 10MHz = 277ps, 1 deg at 600MHz (DAC clock) = 1.66ns

Over 15 loads of the FPGAs, including 3 power cylces, I've seen the same phase difference between the two channels to 1deg (measurement accuracy). So, this seems to work just fine for me.

Good!

hartytp, 2018-06-10:
Cool, so we have working SAWG and SC1. If we can just fix the remaining crashes/serwb issues then we're sorted.

jbqubit, 2018-06-10:
Running 4.0.dev+1133.g0b086225. Compare phase between a pair of channels spanning both DACs on single Sayma. Set frequency0 to 40 MHz. Measured phase difference using Tek FCA 3003 Timer using 10k samples of 10 ms intervals. Saw stdv < 1 deg with peak deviation < 9 deg. Cycling off power and reloading 5 times I see variation in mean relative phase of 
-96.11, -95.46, -95.66, -96.13, -95.50 deg.

jbqubit, 2018-06-10:
I got the fancy scope back so can make a better measurement. Scope and 100 MHz clock to Sayma are phase locked. Still using 4.0.dev+1133.g0b086225 . 

First, compare phase between a pair of channels on single DAC. Here, frequency0 is 210 MHz. No skew, as expected. 

[image]

Now compare DAC channels spanning the pair of chips. frequency0 is 10 MHz . Cycle power and reload .bit's in between. I'm generally getting two results for mean skew with a separation of f_RTIO_coarse. 1/(150 MHz) = 6.7 ns
6.72 ns
6.85 ns
-0.18 ns
6.82 ns
0.13 ns

Expected outcome is skew < 1/f_DAC between board resets. Please reopen Issue. 

[image]
[image]
[image]
[image]
[image]

hartytp, 2018-06-11:
@jbqubit thanks for checking that more carefully. I don't think I can reproduce this on my board...

@enjoy-digital am I right in thinking:
- This can't be an issue with the HMC7043 outputs, since that would have phase errors quantised by the 1.2GHz clock (833ps)
- This can't be an issue with the DAC SYSREF, since that would be quantised by the 600MHz DAC clock (1.66ns)
- So, this has to be an issue with the JESD204B core or similar, right? I wonder if the SYSREF signal is failing to meet S/H on the FPGA?

hartytp, 2018-06-11:
@enjoy-digital since we don't control the phase of `jref` do we need a multireg here?https://github.com/m-labs/jesd204b/blob/25fd79dbf9d5b9536482f11a4d716747479191ec/jesd204b/link.py#L279

sbourdeauducq, 2018-06-12:
HK board with pristine master at a9d97101fc. Board rebooted by artiq_flash start while keeping the RTM bitstream loaded. Here are the phases at 9MHz:
```
0.3769162502517541
0.37677490410203457
0.37691833633914035
0.37692563924427985
0.37678212015178447
0.37694126578251747
0.37685344552359556
0.3769461834287442
0.3770134515919706
0.377082895853537
0.37686592220072035
0.0010597068617359622
0.3768294509602343
0.37692239822937823
0.37687923152541697
0.3768243589643927
0.0011501941022557412
0.37684858846258423
0.3768968922114969
0.37688956304182797
0.3769251419853008
0.0011011526573554288
```
Note that no crash was observed after those 22 reboots (according to https://github.com/m-labs/artiq/pull/1068#issuecomment-396551546 it crashes 1/10 of the time with this firmware).

sbourdeauducq, 2018-06-18:
> @enjoy-digital since we don't control the phase of jref do we need a multireg here?

This jref code is sketchy indeed.

Isn't it instead: we must control the phase of jref (sysref driving the FPGA) so that it meets s/h wrt the JESD/RTIO clock?

sbourdeauducq, 2018-06-18:
@jbqubit Can you test again? With the latest code I am reliably getting ~30ps of absolute skew between the two DACs (at that level it can be the cables or the Red Pitaya).

hartytp, 2018-07-02:
@sbourdeauducq What's the plan for testing this at the full 2.4GHz DAC clock? It seems worth checking that works reliably before we move onto Sayma v2.0 in case there are any changes we decide to make to the hw.

jbqubit, 2018-07-10:
@gkasprow Please test on the Sayma hardware you have.

hartytp, 2018-07-10:
@jbqubit let's try and prioritize things.

IMO https://github.com/sinara-hw/sinara/issues/567 should be the top Sayma priority because (a) it's making it hard for @sbourdeauducq to work on Sayma and (b) it seems like it's probably a relatively straightforwards hardware issue that can be solved quickly, unlike the other issues which involve complex parts of the software stack.

Once that's done we should probably look at https://github.com/m-labs/artiq/issues/1055 next. There is no point in @gkasprow measuring phases between output channels when we know that the HMC7043 isn't behaving itself. This may need a bit more discussion with ADI (is @sbourdeauducq handling that?).

After that, we should probably look at https://github.com/m-labs/artiq/issues/1083 since, again, there is no point looking at output phases when the SYSREF signal isn't behaving itself properly. With luck, this may be a side-effect of the same issue as https://github.com/m-labs/artiq/issues/1055.

After that, we need to look at https://github.com/m-labs/artiq/issues/1080 to figure out why the DACs can't be clocked from that input. Probably needs @gkasprow to check for potential SI issues and then, if we don't find anything there, a discussion with Xilinx/another read through of the relevant Xilinx docs.

Once those are all working we can start thinking about checking output phase synchronization.

sbourdeauducq, 2018-07-11:
> IMO sinara-hw/sinara#567 should be the top Sayma priority because (a) it's making it hard for @sbourdeauducq to work on Sayma and (b) it seems like it's probably a relatively straightforwards hardware issue that can be solved quickly, unlike the other issues which involve complex parts of the software stack.

He has not received the buggy board yet, so it makes sense to do other things in the meantime. And I agree that the other bugs should be fixed before looking at synchronization at the DAC outputs.

> Once that's done we should probably look at #1055 next. There is no point in @gkasprow measuring phases between output channels when we know that the HMC7043 isn't behaving itself. This may need a bit more discussion with ADI (is @sbourdeauducq handling that?).

I'll let @gkasprow handle the RFSYNCIN biasing issues, and David seems to be making good progress with ADI forum archaeology and tests on the HK board. 

> After that, we need to look at #1080 to figure out why the DACs can't be clocked from that input. Probably needs @gkasprow to check for potential SI issues and then, if we don't find anything there, a discussion with Xilinx/another read through of the relevant Xilinx docs.

That one should be high priority as it is blocking most things for you and other people.

hartytp, 2018-07-11:
> David seems to be making good progress with ADI forum archaeology and tests on the HK board.

Aah, I didn't realize he was doing more work on that. That's great then, I'll leave it to him.

sbourdeauducq, 2018-07-12:
Looking good with the latest code.

jbqubit, 2018-07-12:
As I understand it... 

- sinara-hw/sinara#567 might have a simple patch for HK board
- #1055 is closed! 
- #1080 is the last one hanging. This is in the hands of M-Labs, right?

hartytp, 2018-07-12:
> https://github.com/m-labs/artiq/issues/1083

Isn't closed yet. Hopefully it's fixed by the HMC7043 changes, but I would like to test this on my board before we close.

sbourdeauducq, 2018-07-12:
> #1080 is the last one hanging. This is in the hands of M-Labs, right?

We're out of debugging budget, we cannot reproduce it, and it looks like a hardware issue or some obscure xilinx bug.