m-labs/artiq #795: DAC synchronization across Sayma cards

sbourdeauducq, 2017-07-20:


jordens, 2017-11-02:
depends on https://github.com/m-labs/jesd204b/issues/5

jbqubit, 2018-03-29:
@sbourdeauducq Has this been tested on the M-Labs setup?

sbourdeauducq, 2018-07-09:
The code is there and somewhat works intermittently, but I cannot do anything until https://github.com/sinara-hw/sinara/issues/567 is resolved.

sbourdeauducq, 2018-07-16:
@gkasprow Can you expedite the rework, testing, and shipment of the replacement Sayma?

gkasprow, 2018-07-16:
I'm solvinge #475 . It was caused by at least 3 independent factors. 2 were found and fixed, the last one I'm trying to identify but I think I'm very close.

sbourdeauducq, 2018-07-16:
The microtca mess is annoying but it's not blocking other people's developments and experiments, unlike this.

gkasprow, 2018-07-16:
I solved the issue. I will ship one Sayma AMC ASAP.

gkasprow, 2018-07-16:
sure.

jbqubit, 2020-03-27:
In email today with @marmeladapk and @hartytp, @sbourdeauducq said

> Sayma v2 DAC synchronization doesn't quite work and I don't understand why.

By this do you mean synchronization between DAC chip on a single Sayma AMC v2?  What debugging steps did you do?

sbourdeauducq, 2020-03-27:
DAC to FPGA. I cannot test between DACs since only one DAC is working on the board I have.

jbqubit, 2020-03-27:
Don't you have three Sayma and one Metlino at this point?

sbourdeauducq, 2020-03-27:
No, I have only 1 Sayma AMC and 1 Sayma RTM, without panels.

sbourdeauducq, 2020-03-27:
All Metlino tests (and tests involving several Sayma) were done when I was in Warsaw.

marmeladapk, 2020-03-27:
@sbourdeauducq We sent you 2 additional pairs a while ago. O.o

You confirmed reception ( https://github.com/sinara-hw/Sayma_AMC/issues/121#issuecomment-591333534 )

sbourdeauducq, 2020-03-27:
I received 2 Sayma pairs in total indeed, one without front panels and with only 1 DAC working (which is the one I am talking about here), and another one which isn't for M-Labs.

jbqubit, 2020-03-27:
Glad you received the boards. Please look for DAC-DAC synchronization on the other two Sayma. When can you do this? It would help coordinate with Tom.

hartytp, 2020-03-27:
I'm not so worried about synchronisation between separate Sayma cards; we know DRTIO works well. I'd focus on DAC to FPGA synchronisation on a single Sayma for now.

hartytp, 2020-03-27:
@sbourdeauducq what's the plan/time-frame for looking at DAC->FPGA sync on a single board?

jbqubit, 2020-03-27:
@marmeladapk and @sbourdeauducq Already showed basic SAWG + TTL sync.

hartytp, 2020-03-27:
Based on [this](https://github.com/m-labs/artiq/issues/795#issuecomment-605057929) comment I had the impression that DAC -> FPGA doesn't work reliably yet.

@sbourdeauducq could you give a general summary of where we are with Sayma (or point me to one if it already exists). Does board bring up now work reliably for the DRTIO slave? Unsynchronised RF? BaseMod features (switch, attenuator)? Is that all decently well tested from kernels now? DAC -> FPGA synch?

sbourdeauducq, 2020-03-28:
> Please look for DAC-DAC synchronization on the other two Sayma. 

Again, this pair is not for M-Labs, and I don't have it anymore. If you will recall, we had independently ordered an additional set of Sayma boards. I've kept reminding Creotech to ship more hardware.

> Does board bring up now work reliably for the DRTIO slave? Unsynchronised RF? BaseMod features (switch, attenuator)? Is that all decently well tested from kernels now?

All of this generally works. Sometimes (and rarely AFAICT) the DRTIO link appear to become corrupted with some bitstream builds (visible symptom is a storm of broken aux packets); making a minor/unrelated change to the gateware code appears to "fix" it. I have never seen this on Kasli so this is probably GTH/Ultrascale specific.

> DAC -> FPGA synch?

There are unexplained/unpredictable/obscure bugs and I don't have a timeframe.

hartytp, 2020-03-28:
Thanks for the summary. Is there a write up of any of the symptoms you see? Anyway, it sounds like synchronisation between a single DAC and TTL is not reliable yet.

jbqubit, 2020-03-30:
> I've kept reminding Creotech to ship more hardware.

If you're not getting what you need out of Creotech or me or Tom or Xilinx to progress on your work make more noise. Let's get the hardware sorted so the synchronization testing can progress. 

>>DAC -> FPGA synch?

>There are unexplained/unpredictable/obscure bugs and I don't have a timeframe.

Do please get on with trying to reproduce these bugs and create Issues. Understood that you may not have a timeline for fixing bugs which you've not seen reproduce. What's your timeline for running tests on the hardware to try to reproduce the bugs?

hartytp, 2020-03-30:
> Do please get on with trying to reproduce these bugs and create Issues. Understood that you may not have a timeline for fixing bugs which you've not seen reproduce. What's your timeline for running tests on the hardware to try to reproduce the bugs?

Do you mean me or @sbourdeauducq? Right now I don't have a good description of the bugs from @sbourdeauducq. Also, AFIACT @sbourdeauducq isn't having any trouble reproducing these issues, so I don't see that me reproducing them as well would help. I can do it but it will take a non-negligible amount of my time without contributing much clear value to the project.

AFAICT no one has done any real work/testing on Sayma for months now. It would be useful to understand what the issues are in more detail, who is going to work on them and when.

sbourdeauducq, 2020-03-30:
>  Right now I don't have a good description of the bugs from @sbourdeauducq.

Install the beta firmware (with synchronization) and reboot the board a few times while looking at the log. You'll see the sync errors, unless this is a problem with my board in particular.

hartytp, 2020-03-30:
Can you post a log?

Anyway, currently I do not know how the synchronisation process works in any details. It's been heavily rewritten/modified since I last used Sayma. As I don't think there are any docs, I don't expect that I would understand the log messages without investing a significant amount of time reverse engineering the process from the source code. I can do that, but it would be time consuming and I'm not clear that it would add any value to the project.

I'm not clear on the responsibilities or expected time commitments for Sayma. I naively assume that the plan here is for @sbourdeauducq to investigate the synchronisation issues on hw he already has. But please correct me if that's not the plan/not possible.

jbqubit, 2020-03-30:
>> There are unexplained/unpredictable/obscure bugs and I don't have a timeframe.
> Do please get on with trying to reproduce these bugs and create Issues.

This was in reply to @sbourdeauducq. I'd like to know more about the "unexplained/unpredictable/obscure bugs."

sbourdeauducq, 2020-04-02:
> we know DRTIO works well.

We know that it works well:
 1. on Kasli.
 2. with the old siphaser alignment algorithm, which has since been replaced (cc58318500ecfa537abf24127f2c22e8fe66e0f8) but not tested as thoroughly as the first version.

@hartytp If you want to help, could you validate, with the latest code, that you get reproducible RTIO clock phases (and to what tolerance?) between AMC and Kasli, and between AMC and RTM? Checking the outputs of the three Si5324s would be sufficient at this point. If that doesn't work, this should be fixed before further synchronization attempts.

hartytp, 2020-04-02:
OK. Will give it a go next time I have access to my lab.

sbourdeauducq, 2020-04-02:
Okay so it seems that the synchronization problems are three-fold:
* The most frequent issue is caused by the calibration routine setting ``sysref_ddmtd_phase_fpga`` to 0, and then subsequent sync attempts fail. Likely this is a straightforward bug in the sync algorithms that was exposed by different timings on the v2 hardware. It can be worked around by setting another appropriate value into the flash manually and not running calibration again. Then the synchronization logs often look clean (which suggest siphaser also works, though it would be still good to specifically test it to make sure). I'll look at the DAC output synchronization against Urukul shortly, for those boots where the sync log was clean.
* Marginal DDMTD jitter on SAWG builds. This was seen before on Sayma v1 and the solution will be to move the DDMTD core back to the RTM side. Most of the time an error does not happen, and it never happens on non-SAWG builds. This is reported in the log like this:
```
[     8.213517s]  INFO(satman::jdcg::jesd204sync): testing DDMTD stability (raw=true, tolerance=4)...    
[     8.508891s]  INFO(satman::jdcg::jesd204sync):   ...passed, peak-peak jitter: 2                      
[     8.514808s]  INFO(satman::jdcg::jesd204sync): testing DDMTD stability (raw=false, tolerance=1)...   
[     9.916696s] ERROR(satman::jdcg::jesd204sync):   ...excessive peak-peak jitter: 2 (min=74 max=76 center_offset=64)
[     9.925652s] ERROR(satman::jdcg::jesd204sync): failed to align SYSREF at FPGA: excessive DDMTD peak-peak jitter
```
* General DRTIO corruption as mentioned above, when that happens the sychronization code also reports all sorts of errors.

hartytp, 2020-04-02:
> General DRTIO corruption as mentioned above, when that happens the sychronization code also reports all sorts of errors.

Where is the DRTIO corruption described?

sbourdeauducq, 2020-04-02:
https://github.com/m-labs/artiq/issues/795#issuecomment-605371138

hartytp, 2020-04-02:
Thanks! Uuurgh, nasty! Any plan for investigating?

dnadlinger, 2020-04-02:
> visible symptom is a storm of broken aux packets

FWIW, we very rarely (once every few weeks between 5 DRTIO links) see bursts of invalid aux packet errors on Kasli <-> Kasli links as well. Might not be related.

jbqubit, 2020-04-02:
> I'm not clear on the responsibilities or expected time commitments for Sayma. 

Contracts spell out who is doing what. 
- [M-Labs C.3.2.1.1](https://github.com/sinara-hw/Sayma_AMC/wiki/Sayma_v2_contract_m-labs#c321-core-deliverables) is DRTIO.
- [M-Labs C.3.2.1.5](https://github.com/sinara-hw/Sayma_AMC/wiki/Sayma_v2_contract_m-labs#c321-core-deliverables) is [DAC-FPGA synchronization](https://github.com/sinara-hw/Sayma_AMC/wiki/Sayma_v2_contract_m-labs#c301-dac-fpga-synchronization). 
- [Oxford](https://github.com/sinara-hw/Sayma_AMC/wiki/Sayma_v2_integration_support#c32-tasks) main focus is analog characterization, ADF4356 and DDMTD.  
- [Oxford](https://github.com/sinara-hw/Sayma_AMC/wiki/Sayma_v2_integration_support#c32-tasks) is supporting role as time permits for DAC-FPGA synchronization.

General DRTIO corruption and ```sysref_ddmtd_phase_fpga ``` are definitely in the M-Labs wheelhouse.  

"Marginal AMC DDMTD jitter on SAWG builds." If this never happens on non-SAWG builds it suggests the hardware is working correctly. Given how subtle these configuration-dependent "features" are in Vivado/Ultrascale/GTH I'm not sure that @hartytp can substantially contribute. 

It makes most sense for @hartytp to work on testing aspects that are part of his contractual focus and/or tap his unique skills.

sbourdeauducq, 2020-04-07:
There is still this bug where outputs are randomly 180 degrees out of phase (aka "55ns bug" since I've been using a 9MHz waveform to test, but it really is an inversion - which becomes obvious when changing the frequency). This can't be caused by JESD204 sync problems. @jordens Any idea? Can that be inside SAWG e.g. something that isn't reset?

jordens, 2020-04-07:
What are you doing exactly, what do you expect and what are you observing?
There is a lot of channel and sample shuffling in JESD and the DAC. I can imagine a lot of situations where this happens, independent of SAWG.
If you think this is really an inversion in SAWG, you should verify that using the envelope and not the phase. I.e. time a change (in amplitude or phase) on two channels. Or use the DC spline.

sbourdeauducq, 2020-04-07:
Start this and keep it running:
https://github.com/m-labs/artiq/blob/master/artiq/examples/kasli_sawgmaster/repository/sines_urukul_sayma.py

Reboot Sayma and measure phase difference between Urukul and Sayma after each Sayma reboot. It randomly jumps by 180 degrees.

> If you think this is really an inversion in SAWG, you should verify that using the envelope and not the phase. I.e. time a change (in amplitude or phase) on two channels. Or use the DC spline.

Can you try doing that? Or try other experiments to find out where the problem is?

jordens, 2020-04-07:
> Start this and keep it running:

Can that be reduced to an MRE?

> measure phase difference between Urukul and Sayma

How would I do that?

> Can you try doing that? Or try other experiments to find out where the problem is?

Not in the next days. Probably much more efficient if you continue debugging it on-site.

sbourdeauducq, 2020-04-07:
> How would I do that?

The second Red Pitaya that I connected has one channel on Sayma and the other on Urukul.

sbourdeauducq, 2020-04-08:
After replacing SAWG with [a simple DDS core](https://github.com/m-labs/artiq/blob/ec7b2bea12e505a6ffe9c713422c96d55266ca78/artiq/gateware/targets/sayma_amc.py#L245) there are no more inversions and synchronization at 2.4GHz often works. So the issue is either in SAWG or the ``sines_urukul_sayma.py`` experiment. @jordens you are the most familiar with SAWG so please look into it.

> Probably much more efficient if you continue debugging it on-site.

I'm just also using the Red Pitaya and the remote development infra, and using a physical oscilloscope etc. would not make it more efficient

sbourdeauducq, 2020-04-08:
I posted this before but here is the script I am using to measure the Urukul/Sayma phase difference.
```python
import socket
import numpy as np
import matplotlib.pyplot as plot
from scipy import signal, constants

class RPSCPI:
    def connect(self, host):
        self.sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
        self.sock.connect((host, 5000))
        self.sock_f = self.sock.makefile()

    def close(self):
        self.sock_f.close()
        self.sock.close()

    def sendmsg(self, msg):
        self.sock.send(msg.encode() + b"\r\n")

    def recvmsg(self):
        return self.sock_f.readline().strip()

    def trigger(self):
        self.sendmsg("ACQ:START")
        self.sendmsg("ACQ:TRIG NOW")
        while True:
            self.sendmsg("ACQ:TRIG:STAT?")
            if self.recvmsg() == "TD":
                break

    def get_data(self, channel):
        self.sendmsg("ACQ:SOUR{}:DATA?".format(channel))
        buff_string = self.recvmsg()[1:-1].split(',')
        return np.array(list(map(float, buff_string)))

rp = RPSCPI()
rp.connect("127.0.0.1")
try:
    rp.trigger()
    y1 = rp.get_data(1)
    y2 = rp.get_data(2)

    t = np.arange(y1.shape[0])/125e6
    y = np.c_[y1, y2].T
    z = signal.decimate(y*np.exp(1j*2*np.pi*9e6*t), q=10, ftype="fir", zero_phase=True)[:, 10:]
    z = signal.decimate(z, q=10, ftype="fir", zero_phase=True)[:, 10:]
    angle = np.angle(np.mean(z[0]*z[1].conj()))

    print(angle)

    plot.plot(y1)
    plot.plot(y2)
    plot.show()

finally:
    rp.close()

```

jordens, 2020-04-08:
> So the issue is either in SAWG or the sines_urukul_sayma.py experiment.

We've had plenty of examples in the past where this simple logic was plain wrong.

> I'm just also using the Red Pitaya and the remote development infra, and using a physical oscilloscope etc. would not make it more efficient.

That's intentionally ignoring the most important differences.
As I have explained before, some of the unresolved major problems are:

* long network round trips making interactive work impossible and everything else slow
* no documentation of the setup, no maintenance of that documentation
* long round trip delays across time zones for questions and discussion

sbourdeauducq, 2020-04-08:
> We've had plenty of examples in the past where this simple logic was plain wrong.

What is your better idea then?

sbourdeauducq, 2020-04-08:
> That's intentionally ignoring the most important differences.

The network delays are small compared to the compilation/flashing/boot time of Sayma, and are not really important for grabbing waveforms with the script above. What is the interactive work you want to do?

Most of the information about the setup is in your inbox, if you have questions please ask.

jordens, 2020-12-15:
<details><summary>Code</summary>

```python
from artiq.experiment import *

class SinesUrukulSayma(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.sawgs = [self.get_device("sawg"+str(i)) for i in range(4)]
        self.basemod = self.get_device("basemod_att0")
        self.rfsws = [self.get_device("sawg_sw"+str(i)) for i in range(4)]
        self.ttl = [self.get_device("ttl_mcx"+str(i)) for i in range(4)]

    # DRTIO destinations:
    # 0: local
    # 1: Sayma AMC
    # 2: Sayma RTM
    @kernel
    def drtio_is_up(self):
        for i in range(3):
            if not self.core.get_rtio_destination_status(i):
                return False
        return True

    @kernel
    def run(self):
        f = 50*MHz
        sawg_ftw = self.sawgs[0].frequency0.to_mu(f)

        self.core.reset()

        print("waiting for DRTIO ready...")
        while not self.drtio_is_up():
            pass
        print("OK")

        delay(10*ms)
        self.core.reset()
        delay(10*ms)

        rfsw = self.rfsws[3]
        sawg = self.sawgs[0]
        ttl = self.ttl[0]

        ttl.output()

        delay(10*ms)
        self.basemod.reset()
        delay(10*ms)
        self.basemod.set(3., 3., 3., 3.)
        delay(10*ms)

        rfsw.on()
        delay(1*ms)

        sawg.reset()
        delay(1*us)
        sawg.config.set_clr(1, 1, 1)
        delay(100*us)
        # align to coarse RTIO period
        at_mu(now_mu() & ~0x7)
        sawg.frequency0.set_mu(sawg_ftw)
        delay_mu(1<<8)
        sawg.phase0.set_mu(0)  # sawg_ftw*now_mu() >> 17)
        delay_mu(1<<8)
        sawg.amplitude1.set(.4)
        delay_mu(1<<9)
        sawg.amplitude1.set(.0)
        delay_mu(1<<8)
        ttl.on()
        delay_mu(1<<9)
        ttl.off()
```

</details>

[image]

* 3: Sayma MCX TTL0 out
* 4: Sayma SAWG0 out 50 MHz
* Sayma v2 AMC and RTM (the one with broken DAC1 note)
* 150 MHz RTIO Satellite behind Kasli v1.1 Master
* ARTIQ 1ce505c5 (`sayma_amc -V satellite --sfp`)

There are two non-determinisms and one problem.

1. Sometimes (~4e-2) it outputs 300 MHz (600 MS/s) large scale data.
2. Sometimes (~1e-1) e-there is a 32 sample delay.
3. Most of the time the delay is either 0 or 2 samples.

Note that:

* there is no "180 phase shift"
* no other category of behavior that I've seen in a couple hundred reboots
* one RTIO period is 4 samples

I'd propose: 

0. Ensure delay determinism across two TTL MCX out on different Saymas (or Kasli-Sayma). This shouldn't factor in the above case though but would cover only DRTIO and exclude JESD and SAWG.
1. Verify JESD, DAC setup and CDCs by using the simple satellite to output a gated (using a single-bit RTIO channel) fixed 150 MHz square wave (needs to be done from the RTIO CD!). Ensure that delay is deterministic.