m-labs/artiq #800: jtag for sayma/ultrascale

jordens, 2017-07-20:
(#563)

* [ ] openocd ultrascale support (clean up, consolidate, push to openocd)
* [ ] dual flash or 32 bit address support (clean up, push to openocd)

sbourdeauducq, 2017-07-20:
Are we programming the FTDI chip? I recommend we don't, or only with existing software that is already proven to "work" with their buggy silicon.

jordens, 2017-07-20:
Right. But IIRC there needs to be some programming to tell the chip how to use and expose the four ports. Maybe that's also something that the manufacturer can do.

sbourdeauducq, 2017-07-20:
Doesn't OpenOCD reconfigure them dynamically? IIRC the only benefits of EEPROM programming are 1) the JTAG port won't appear as a ttyUSB device before OpenOCD claims it 2) fancy USB identifiers.

jordens, 2017-07-20:
IIRC the serial port doesn't appear at all. That needs eeprom programming. Nothing in the current openocd code even touches that. Maybe the JTAG port doesn't need an EEPROM. But that kills one way of distinguishing Saymas.

jordens, 2017-08-03:
* XKCU040 proxy bitstream written, cleaned up and build. Works for both flashes: https://github.com/jordens/bscan_spi_bitstreams
* openocd changes https://github.com/m-labs/openocd/tree/kcu105
* pushed and being reviewd: http://openocd.zylin.com/#/q/owner:jordens%2540gmail.com+status:open
* FTDI EEPROM may work somewhat initially out of the box. But generating the eeprom binary won't be smooth as it is not really supported by ftdi_eeprom. See http://developer.intra2net.com/git/?p=libftdi;a=blob;f=doc/EEPROM-structure;h=08abe62012e4de1a3c1b2659e3b8b16a4d2c858a;hb=HEAD

sbourdeauducq, 2017-08-04:
In the likely event that there is any problem with ftdi_eeprom other than simply getting an EEPROM binary, I suggest simply using the official Windows tool. ftdi_eeprom/libftdi usually has very puzzling behavior if we're not doing something that everyone else is already doing, due to the abundant FTDI silicon bugs and quirks.

sbourdeauducq, 2017-08-04:
As a side note, I'd be happier if there were a good alternative to FTDI JTAG+UART. FTDI are not worth our time, buggy silicon and [nasty company](http://hackaday.com/2014/10/22/watch-that-windows-update-ftdi-drivers-are-killing-fake-chips/).

jordens, 2017-08-07:
* I'll leave out messing with the EEPROM for now. It comes up with no serial, standard description, defaults to quad rs232. Openocd detaches the kernel serial drivers and reconfigures the first channel as MPSSE.
* m-labs/sinara#247

jordens, 2017-08-07:
Done.
Since serial assignment and initial configuration (EXAR, MMC...) should be done by the manufacturer, they should do the FTDI EEPROM writing as well.

jordens, 2017-08-08:
Since the XCKU040 bitstream is 128 MBit and we will put the RTM-XC7A15T bitstream, and (definitely for Sayma standalone) the bios, and the runtime, and configuration after it, we need to support 4 byte SPI flash addresses in openocd, and in the misoc spi flash driver as well.

jordens, 2017-08-08:
With current https://github.com/m-labs/openocd/tree/sayma
and https://github.com/jordens/bscan_spi_bitstreams both SPI flashes and the multi-TAP chain with the RTM-FPGA works fine.

```tcl
interface ftdi
ftdi_device_desc "Quad RS232-HS"
ftdi_vid_pid 0x0403 0x6011
ftdi_channel 0
# EN_USB_JTAG on ADBUS7: out, high
# nTRST on ADBUS4: out, high, but R46 is DNP
ftdi_layout_init 0x0098 0x008b
ftdi_tdo_sample_edge falling
ftdi_layout_signal nSRST -data 0x0080
# ftdi_layout_signal nTRST -data 0x0010
# reset_config trst_and_srst srst_pulls_trst srst_gates_jtag srst_push_pull trst_push_pull
reset_config srst_only srst_pulls_trst srst_gates_jtag srst_push_pull

adapter_khz 25000

transport select jtag

jtag newtap rtm_xc7 tap -irlen 6 -ignore-version -expected-id 0x0362e093
jtag newtap amc_xcu tap -irlen 6 -ignore-version -expected-id 0x03822093

pld device virtex2 rtm_xc7.tap 1
pld device virtex2 amc_xcu.tap 1

set DR_LENGTH 4
set XILINX_USER1 0x02
set XILINX_USER2 0x03
set AMC_DR_LEN 2

target create amc_xcu.proxy testee -chain-position amc_xcu.tap
flash bank amc_xcu.spi0 jtagspi 0 0 0 0 amc_xcu.proxy $XILINX_USER1 $AMC_DR_LEN
flash bank amc_xcu.spi1 jtagspi 0 0 0 0 amc_xcu.proxy $XILINX_USER2 $AMC_DR_LEN

init
runtest 1000
puts "JTAG scan chain:"
scan_chain
puts "Flash banks:"
flash banks
#exit

pld load 1 bscan_spi_xcku040_sayma.bit
reset halt

flash probe amc_xcu.spi0
flash probe amc_xcu.spi1
irscan amc_xcu.tap $XILINX_USER1
flash read_bank 0 sayma_f0.bin 0 1024
# TODO needed
irscan amc_xcu.tap $XILINX_USER2
flash read_bank 1 sayma_f1.bin 0 1024

# xcu_program amc_xcu.tap
exit
```

sbourdeauducq, 2017-08-16:
Cannot get it to work with the RTM disconnected:
```
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: xcu.tap: IR capture error; saw 0x00 not 0x01
```

Works fine when RTM is there. Greg wants to check if this is a hardware issue.

sbourdeauducq, 2017-08-16:
Well. Strangely enough I cannot reproduce the issue after disconnecting the RTM again...

sbourdeauducq, 2017-08-20:
@jordens Can the script above be put into an OpenOCD target file for Sayma? Can it be submitted upstream?

jordens, 2017-08-20:
A lot of the openocd wrapper functions need rewriting to support multiple TAPs and multiple flashes. Then yes.

sbourdeauducq, 2017-08-21:
> pld device virtex2

This also should be cleaned up, it's not a virtex2.

jordens, 2017-08-21:
Sure. File a bug with openocd.

sbourdeauducq, 2017-08-21:
https://sourceforge.net/p/openocd/tickets/160/

jordens, 2017-09-29:
https://github.com/m-labs/openocd/tree/jtagspi-32bit-addr

closed 2017-09-29 by jordens