m-labs/artiq #801: SAWG external modulation

sbourdeauducq, 2017-07-20:
Quoting the ARL contract (see also draft_sayma_metlino_procurement_20160805.pdf):
* [ ] f,p,a,u modulation by RTIO (eDRTIO) for all interpolators
* [ ] Coherent phase mode (implemented in kernel API and runtime) for frequency/phase updates of the three eDRTIO channels

jbqubit, 2018-03-29:
@jordens What's the status of this?

hartytp, 2018-08-11:
@jbqubit could you outline what exactly this modulation consists of, and what the purpose is?

jbqubit, 2018-08-15:
This was funded 4/18/2017. The implementation detail is provided in the artiq-hardware writeup written with @jordens. [L245](https://github.com/m-labs/artiq-hardware/blob/1cec6224bfaf148cba14a5a1f69f537966d6887b/artiq_hardware.tex#L245) Here's a rendered version. 

[image]

Modulation by RTIO has several use cases for systems with N SAWG channels. At UMD a 32-qubit system with individual addressing requires N = 32 SAWG channels. For randomized benchmarking assume gate depth D gates, M pre-calculated randomized sequences, number of knots per gate K and bits per spline-point is B [0]. Any given rep of the experiment will choose among M so all must be pre-loaded in DRAM. Total number of DRTIO events is N*D*M*K and total number of bits to be pre-loaded is something like N*D*M*K*B. 

As a specific case consider benchmarking of single-qubit gates implemented using K = 3 spline points; frequency set once and amplitude set three times. Total number of bits is BK = 48+3*(16*4+16) = 288 bits (with no DRTIO overhead). Assume, N=32, D=2000, M=500. So the total number of pre-loaded DRTIO events is N*D*M*K = 96e6 and total number of bits is N*D*M*BK = 912e6 bytes. Assuming a DMA data rate of R = 0.5e9 bit/s the time to load is N*D*M*BK/R = 14 s. Benchmark experiment run time may be 30 minutes. 

During experiment run time parameters may drift such as magnetic field, magnetic field gradient, laser beam intensity, secular frequencies. Interleaved with the benchmarking experiment are calibration experiments that measure drift. Modulation by RTIO permits parametric update of waveform f, p, a without recomputing and reloading all the waveforms by DMA. 

One variety of two-qubit gate involve waveforms with K = 2(N+1) knots where each knot point modifies one frequency coefficient. There are order N(N-1) such gates for arbitrary pair-wise gates on an N-ion chain. Number of bits per knot point is then B = 16+48*2 = 112. For a randomized benchmarking experiment, N=32, D=200, M=500, K=2(32+1)=65. The total number of DRTIO events loaded into DMA is N*D*M*K = 208e6 and total number of bits is N*D*M*B*K = 3e9 bytes. Real benchmarking would involve both one and two-qubit gates. The present Modulation by RTIO is a decent way of addressing the problem without resorting to gate-sequence-specific optimizations in gateware. 

With ADCs and digital signal processing (ie PID) on Sayma a path called "Modulation by local DSP" is envisioned for real-time feedback similar to Modulation by RTIO. The many-channel case was envisioned for gates and it also could be used for stabilization of laser amplitude and frequency. Modulation by local DSP is not funded. 

[0] Based on waveform parameterization in [L159](https://github.com/m-labs/artiq-hardware/blob/1cec6224bfaf148cba14a5a1f69f537966d6887b/artiq_hardware.tex#L159) the maximum number of bits per knot point is 16+16*4*3+48*2*3+16*1*3+2 = 546. 

------------

The following paper offers a high-level discussion and sample algorithms using a multi-channel AOM for N-wise single-ion addressing. This is the basis of two-qubit gates pursued by UMD and other groups. Also see Appendix "Native single- and two-qubit rotations."

http://arxiv.org/abs/1603.04512

The following paper details the bi-chromatic gates on ion chains. It explains why N SAWG channels are needed for N ions.

http://iontrap.umd.edu/wp-content/uploads/2012/12/PhysRevLett.112.190502.pdf

The underlying theory of using a spin-dependent force for 2-qubit gates using transverse modes is here. The meat of bi-chromatic gates is in section "Arbitrary-speed TP gates through minimal control of the laser beams."

http://arxiv.org/abs/quant-ph/0601159

hartytp, 2018-08-16:
@jbqubit thanks for the clarification (particularly, for the clarification on the intended use-cases for this feature).

Since the current SAWG design suffers from being overly resource intensive, I think we should probably aim for the minimal complexity solution that meets our needs (lest over-designing one part of the SAWG causes serious problems elsewhere). So, some questions:

- I agree that an AM input to the SAWG is crucial. We'd also use that for a noise eater. Do we really need FM and PM inputs? It's not clear to me that the SAWG is the right place to do housekeeping jobs like tracking magnetic fields, as there is often an easier way of doing that -- either by controlling another AOM somewhere, which could be driven CW by Urukul, or by servoing the filed directly. If we don't have a concrete need for these features, then let's shelve them at least for now (we can always come back and add them later if such a need emerges) unless @jordens feels that the resource consumption for adding them is small.
- Not sure if it actually makes any difference to the design, but it's worth noting that the BW and latency requirements can probably be a lot laxer than your specification dictates. e.g. AM BW of greater than a few MHz is probably not needed.
- For the AM, do we need to be able to control the amplitude of both tones independently, or would a common-mode modulation be enough? I'd argue that common-mode is enough (e.g. to take our intensity drifts)
- For the RTIO interface, I can't see any need for 1 RTIO channel per modulation input. IIRC, this increases the resource requirements quite a bit. My guess is that we can do with a single RTIO channel for all AM inputs (see the SU-servo for how we designed that and the level of sharing between RTIO channels). On this note: @jordens would it help the Sayma complexity issues if we decreased the number of RTIO channels overall? AFAICT, we currently use up far more channels than are really necessary. I don't think it's important to be able to update every parameter of the SAWG simultaneously.

@jordens do you have any other questions about this feature/suggestions for simplifying it?

hartytp, 2018-08-16:
Am I right in thinking that "coherent phase mode" is just housekeeping done on the coredevice in software? i.e. it's a convenience feature that doesn't need any gateware?

hartytp, 2018-08-28:
@k-beck thanks for the feedback.

> Neither the PM nor the AM inputs have an immediate application for us. We currently noise-eat with a separate AOM.

Okay, shall we cut those out of this proposal then to simplify things? We can always add them later as extensions but, given the issues we currently have with Sayma, it probably makes most sense to begin with the bare essentials and extend as necessary.

AM is important but AFAICT, the main use case for it will be closed-loop applications such as servos, rather than a direct modulation via RTIO. So, it makes most sense to leave that to a separate Sayma servo issue.

> RTIO channel count Agreed, a single RTIO channel can serve the FM inputs (and a separate one can serve the AM inputs) for the channels on one card. (I prefer this, in fact.)

:+1:

Out of curiosity though, @jordens how bad is the penalty (FPGA resources, compile time, timing closure, etc) for adding extra RTIO channels now that we have SED?

> Coherent phase mode This is a useful housekeeping function, especially for implementing a branching logical diagram.

My understanding is that this is something that would only ever be implemented in software (see the top post) in any case, so it has no overhead apart from the time taken by the Kernel CPU to do the required maths.

hartytp, 2018-08-28:
> Out of curiosity though, @jordens how bad is the penalty (FPGA resources, compile time, timing closure, etc) for adding extra RTIO channels now that we have SED?

i.e. while we're at it, would it be worth considering a more general simplification/overhaul to the SAWG RTIO interface?

hartytp, 2018-08-28:
> External modulation The FM input is critical for UMD. We frequency modulate the sideband frequencies in our MS gate (as done in the paper by Debnath, et al., cited above) and need to be able to choose the sign of the feedback for each of the two tones.

@k-beck To be clear here: you mean that you want to have a way of compensating for things like drifting motional frequencies/light shifts/magnetic fields without recomputing your gate sequence. Is that right?

What's your expected pulse duration (single- and two-qubit)? Is the time that a pulse sequence takes to complete short compared with the time taken to recompute the pulse sequence? If not, does this feature really help you? 

In the Debnath paper, it quotes a 2-quibt gate time of 235us. That's a lot of time compared with the time taken to submit an RTIO event, so unless you're going to start doing gates faster than that, recomputing pulse sequences on the fly should work fine, shouldn't it?

Anyway, I don't object to this feature at all -- although, it would be useful to hear from @jordens about how much work/resource consumption/etc it involves. But, I wonder if it might be a better overall strategy to focus on getting the sustained RTIO event rate up rather than trying to add lots of knobs that allow one to tweak fixed pulse sequences.

Another strategy that might work well would by to check that the SAWG parameterization is really optimal for the standard use-cases in the lab. e.g. is it currently parameterized in a way that requires more RTIO events to produce a pulse than are really necessary? If we tweaked the parameterisation a bit, we might be able to make it much faster. Could you post a simple RBM experiment using Sayma that we can look at?

k-beck, 2018-08-28:
> To be clear here: you mean that you want to have a way of compensating for things like drifting motional frequencies/light shifts/magnetic fields without recomputing your gate sequence. Is that right?

@hartytp This is actually to feed-forward frequency noise from the laser we use for our 2-qubit gates. We're correcting for acoustical noise in the laser cavity (dominant noise peaks are a few hundred Hz), not slow drifts (like the ones you mention) that we intend to address by recomputing. 

(And, yes, our two-qubit gate times will be similar to those in the Debnath paper) 

I'd like to hear from other users before cutting AM/PM, but it's fine from my vantage point especially if it makes resource counts go from 'more than we have' to 'workable'.

hartytp, 2018-08-29:
> I'd like to hear from other users before cutting AM/PM, but it's fine from my vantage point especially if it makes resource counts go from 'more than we have' to 'workable'.

Agreed! Input from more potential users of Sayma would be great. Otherwise, I tend to adopt the rule of thumb that if no one can be bothered to advocate for a feature then it's probably not important.

In any case, if we are designing/specifying features such as AM/PM without guidance from the people who will use them -- or even a clear picture of who those people might be -- then we're unlikely to do a good job of designing something that fits the use-case well.

NB however: the suggestion isn't necessarily to *"cut"* AM/PM completely, but rather to drop them from this particular proposal, which relates to modulation via (D)RTIO. e.g. AM will be essential for a noise-eater, which is a high-priority for us, but it makes sense to discuss that in a separate Sayma servo proposal.

hartytp, 2018-08-29:
> @hartytp This is actually to feed-forward frequency noise from the laser we use for our 2-qubit gates. We're correcting for acoustical noise in the laser cavity (dominant noise peaks are a few hundred Hz), not slow drifts (like the ones you mention) that we intend to address by recomputing.

Okay, thanks for the clarification. That's quite a different application to the one that Joe posted [above](https://github.com/m-labs/artiq/issues/801#issuecomment-413330645).

Can you give me a more precise reference to how you do this at the moment, please as I don't have a clear picture of your planned use-case for this yet?

Is your plan something like to digitize a beat-note (DC or some AC signal?) and feed-forwards on the RF frequency produced by each Sayma channel to correct for fluctuations?

This issue is specifically about modulation via (D)RTIO. Does that mean that your plan is to do the feed-forwards maths on the CPU (currently a soft-CPU with no FPU) and then to program the SAWG FM input over DRTIO? What update rate + channel count are you expecting?

Is there a reason you either can't do this in gateware on Sayma (i.e. using a SAWG FM input but not DRTIO) or add an extra AOM to perform the frequency shift? An extra AOM could easily be controlled via a modified form of the SU-servo, which seems like a much cleaner way of doing things (separating house-keeping from coherent control).

jordens, 2018-08-29:
@hartytp 
The current work calls for 9 additional 0th order spline RTIO channels per SAWG channel that are then added/multiplied onto the `f`/`p`/`a`/`u` streams after spline interpolation. Addition saturation is not foreseen.
Yes. Coherent phase updates are just a `f*T` offset on absolute phase mode in software.
Simplifying the specification for SAWG including its RTIO interface would likely help. This is why I suggested revisiting the specifications for Sayma 2 and it's what my suggestions and proposals elsewhere are about. When there is funding for Sayma 2 I can imagine several opportunities to improve and simplify SAWG (different DUC scheme as suggested elsewhere, improving and shrinking the spline coefficient representation/serialization, transaction-based SAWG RTIO events which would cut down the number of RTIO channels, raw sample channels, revisiting phase and frequency resolution, study required bit depth/resource usage...).

jordens, 2019-07-17:
Canceled from the ARL contract.