m-labs/artiq #803: "Unknown RPC value tag" error during RPC

cjbe, 2017-07-24:
Using the master version of Artiq, I sporadically get the below error. 
This occurs when calling an async RPC with an integer argument (code below).
I do not see any errors in the core log.

```
2017-07-24 13:21:39,370 ERROR:worker(14,monitor_fluo.py):root:Terminating with exception (OSError: Unknown RPC value tag: 'Z')
Traceback (most recent call last):
  File "C:\Users\LocalAdmin\AppData\Local\conda\conda\envs\artiq3\lib\site-packages\artiq\master\worker_impl.py", line 272, in main
    exp_inst.run()
  File "c:\artiq-master\repository\monitor_fluo.py", line 39, in run
    self.krun()
  File "C:\Users\LocalAdmin\AppData\Local\conda\conda\envs\artiq3\lib\site-packages\artiq\language\core.py", line 53, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Users\LocalAdmin\AppData\Local\conda\conda\envs\artiq3\lib\site-packages\artiq\coredevice\core.py", line 133, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "C:\Users\LocalAdmin\AppData\Local\conda\conda\envs\artiq3\lib\site-packages\artiq\coredevice\comm_kernel.py", line 601, in serve
    self._serve_rpc(embedding_map)
  File "C:\Users\LocalAdmin\AppData\Local\conda\conda\envs\artiq3\lib\site-packages\artiq\coredevice\comm_kernel.py", line 506, in _serve_rpc
    args, kwargs = self._receive_rpc_args(embedding_map)
  File "C:\Users\LocalAdmin\AppData\Local\conda\conda\envs\artiq3\lib\site-packages\artiq\coredevice\comm_kernel.py", line 399, in _receive_rpc_args
    value = self._receive_rpc_value(embedding_map)
  File "C:\Users\LocalAdmin\AppData\Local\conda\conda\envs\artiq3\lib\site-packages\artiq\coredevice\comm_kernel.py", line 394, in _receive_rpc_value
    raise IOError("Unknown RPC value tag: {}".format(repr(tag)))
OSError: Unknown RPC value tag: 'Z'
```

```python
@rpc(flags={"async"})
def update(self, n): 
    self.set_dataset("count", n, broadcast=True)

@kernel
def krun(self):
    self.core.reset()
    while True:
        n = self.pmt.count()
        delay(50*us)
        self.pmt.gate_rising(self.t_bin)
        self.update(n)
```

whitequark, 2017-07-25:
The provided code is not enough to unambiguously construct a reproducing example. I've modified it as follows so I can run it with `artiq_run`:

```python
from artiq.experiment import *

class Count(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("pmt")
        self.t_bin = 10*ms

    @rpc(flags={"async"})
    def update(self, n):
        self.set_dataset("count", n, broadcast=True)

    @kernel
    def run(self):
        self.core.reset()
        while True:
            n = self.pmt.count()
            delay(50*us)
            self.pmt.gate_rising(self.t_bin)
            self.update(n)
```

Running this for a few minutes does not result in a crash. Can you provide a more specific example, and tell me about your environment?

cjbe, 2017-07-25:
@whitequark my complete reproduction is below - it is pretty much the same as yours. It took ~2 minutes to crash.

Here is a packet dump: [dump.zip](https://github.com/m-labs/artiq/files/1172923/dump.zip)

As you can see, the core device responds incorrectly in packet 3809 (the last 6 bytes of the message are missing).

Bad packet (3809):
```
0000  5a 5a 5a 5a 0a 01 00 00  00 02 4f 00 00 00 04 69   ZZZZ.... ..O....i
0010  00 00 00 00                                        ....
```
Equivalent OK packet (3806):
```
0000  5a 5a 5a 5a 0a 01 00 00  00 02 4f 00 00 00 04 69   ZZZZ.... ..O....i
0010  00 00 00 00 00 00 00 00  01 6e                     ........ .n
```

```python
from artiq.experiment import *

class CrashTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("pmt")

    @rpc(flags={"async"})
    def update(self, n):
        self.set_dataset("count", n, broadcast=True)

    @kernel
    def run(self):
        self.core.reset()
        while True:
            n = self.pmt.count()
            delay(50*us)
            self.pmt.gate_rising(50*ms)
            self.update(n)
```

sbourdeauducq, 2017-09-23:
@whitequark believes that this is due to former smoltcp bugs corrupting data.
@cjbe I've been running your code for 20min without problem. Can you still reproduce the issue with the latest master? If not let's close this issue.

cjbe, 2017-09-23:
@sbourdeauducq using the latest master I also cannot reproduce this.

closed 2017-09-23 by cjbe