m-labs/artiq #804: Slow super-linear compilation for multidimensional arrays

joefgoodwin, 2017-07-25:
Compilation of multidimensional arrays that will be accessed by kernel functions is very slow and super-linear in time.

May have some overlap with [issue  #709](https://github.com/m-labs/artiq/issues/709), but occurs when defining the list as an attribute rather than only when passing a list as an argument to the function.

`class SlowCompExp(EnvExperiment):

    def build(self):
        self.setattr_device("core")

    def run(self):
        # This takes 1.2s
        # self.my_array = [0 for i in range(5000*16)]

        # This takes 5s
        # self.my_array = [[[0 for k in range(100)] for j in range(50)] for i in range(16)] 

        # This takes 21s
        self.my_array = [[0 for j in range(5000)] for i in range(16)] 

        self.k_run()

    @kernel
    def k_run(self):
        #self.my_array[0] = 0
        #self.my_array[0][0][0] = 0
        self.my_array[0][0] = 0`

A one dimensional list with 5000*16=80000 elements takes ~1s to compile.
Two dimensional list with 5000 by 16 elements takes >20s, versus <5s for 2500 by 16.
Strangely, three dimensional list with 100 by 50 by 16 takes 5s, less than the two dimensional case with same total number of elements

whitequark, 2017-07-25:
This is basically expected because of Python's dynamic typing. @sbourdeauducq, can we say that only large numpy arrays will be compiled quickly?

joefgoodwin, 2017-07-25:
````
# This takes 4.3s
self.my_array = np.asarray([[0 for j in range(5000)] for i in range(16)])
# This takes 38s
self.my_array = np.asarray([[0 for j in range(20000)] for i in range(16)]) 
````

Casting as numpy array speeds things up by ~75% but still slow and non-linear

whitequark, 2017-07-25:
The numpy array type inference being slow is a bug, the Python array type inference being slow is more or less by design. I'll think about what I can do.

joefgoodwin, 2017-07-25:
We'd be happy if explicitly casting with np.asarray makes it work - no problem if the Python type inferencing remains slow.

pca006132, 2021-07-02:
It should be a lot faster after #1709. The current time is:
- 0.64s
- 1.34s
- 1.19s

closed 2021-07-02 by pca006132