m-labs/artiq #805: kernel invariants: seconds to mu conversion not for free on kernel

jonaskeller, 2017-07-27:
Running 2.4 on Windows 7, nist_qc2 gateware

From #529 I got the impression that converting kernel invariant SI time variables to mu in kernel methods should happen at compile time and thus come without any cost.

However, I see a significant difference as compared to doing the conversion on the host.
Also, pulse_mu(seconds_to_mu(...)) seems favorable to pulse(...).

Some clarification of these differences in the documentation would be helpful.

Here's a minimal example:

`

from artiq.experiment import *

class SlackTest(EnvExperiment):
    '''SlackTest'''
    kernel_invariants = {"pulse_time","pulse_time_mu"}
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl6")
        self.setattr_device("ttl7")#two ttls to avoid underflows due to limited OFIFO depth
        
    def prepare(self):
        self.pulse_time = 8*ns
        self.pulse_time_mu = seconds_to_mu(self.pulse_time,self.core)
        
    @kernel
    def pulse(self,output):
        output.pulse(self.pulse_time)
        
    @kernel
    def pulse_mu(self,output):
        output.pulse_mu(self.pulse_time_mu)

    @kernel
    def pulse_mu2(self,output):
        output.pulse_mu(seconds_to_mu(self.pulse_time))
        
    @kernel
    def run(self):
        self.core.break_realtime()
        for i in range(30):
            self.pulse(self.ttl6)
            self.pulse(self.ttl7)
            #self.pulse_mu(self.ttl6)
            #self.pulse_mu(self.ttl7)
            #self.pulse_mu2(self.ttl6)
            #self.pulse_mu2(self.ttl7)
            delay(10*ns)
`

I've looked at six cases:
"pulse" commands, pulse_time as kernel invariant: RTIO Underflow after 36 pulses, slack = ~-2800mu
[sidenote: in Artiq 2.0, it managed 58 pulses before the underflow]

"pulse" commands, no kernel invariants: RTIO Underflow after 1 pulse, slack ~-12000mu

"pulse_mu" commands, pulse_time_mu as kernel invariant: outputs all pulses, slack ~+39us

"pulse_mu" commands, no kernel invariants: outputs all pulses, slack ~+32us

"pulse_mu2" commands, pulse_time as kernel invariant: RTIO Underflow after 54 pulses, slack ~-3000mu

"pulse_mu2" commands, no kernel invariants: RTIO Underflow after 2 pulses, slack ~-30000mu

r-srinivas, 2017-08-28:
Could this be added to 2.5 as well, please?

whitequark, 2017-10-02:
Updated testcase:
```python
from artiq.experiment import *

class SlackTest(EnvExperiment):
    '''SlackTest'''
    kernel_invariants = {"pulse_time","pulse_time_mu"}

    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl5")
        self.setattr_device("ttl6")#two ttls to avoid underflows due to limited OFIFO depth

    def prepare(self):
        self.pulse_time = 8*ns
        self.pulse_time_mu = self.core.seconds_to_mu(self.pulse_time)

    @kernel
    def pulse(self,output):
        output.pulse(self.pulse_time)

    @kernel
    def pulse_mu(self,output):
        output.pulse_mu(self.pulse_time_mu)

    @kernel
    def pulse_mu2(self,output):
        output.pulse_mu(self.core.seconds_to_mu(self.pulse_time))

    @kernel
    def run(self):
        self.core.break_realtime()
        try:
            n = 0
            for i in range(100):
                n = i
                # self.pulse(self.ttl5)
                # self.pulse(self.ttl6)
                # self.pulse_mu(self.ttl5)
                # self.pulse_mu(self.ttl6)
                # self.pulse_mu2(self.ttl5)
                # self.pulse_mu2(self.ttl6)
                delay(10*ns)
        except RTIOUnderflow:
            print("underflow at", n)
```

whitequark, 2017-10-02:
Unable to reproduce on latest master. All three variants underflow at 32 pulses. The LLVM IR also checks out.

closed 2017-10-02 by whitequark