m-labs/artiq #813: bitstream loading for Sayma RTM FPGA

jordens, 2017-08-09:
Implement slave serial bitstream loading from AMC/XCKU flash to RTM XC7A (depends on m-labs/sinara#249 and m-labs/misoc#53)

sbourdeauducq, 2017-08-29:
It seems we can just switch the RTM FPGA to slave serial mode by changing the Mx resistors, and then the Xilinx-designed daisy chain loading should work. @gkasprow do you confirm?

jbqubit, 2017-08-31:
Ping @gkasprow

gkasprow, 2017-08-31:
@sbourdeauducq This mode would be very slow because only SPIx1 can be used. I tried this approach but didn't manage to configure any of FPGAs. That's why I resoldered resistors and RTM FPGA is loaded directly by AMC FPGA. Another reason could be due to long unterminated stub of config clock. I didn't inverstinage it further because you requested direct configuration in slave mode and I added this Xlinx config scheme as plan B.

sbourdeauducq, 2017-08-31:
> This mode would be very slow because only SPIx1 can be used

Don't we still get 50Mbit/s in this mode? It would only take 3-4 seconds to load; this is not "very slow".

> RTM FPGA is loaded directly by AMC FPGA

Well, only after someone has written and debugged software and gateware that does that...

gkasprow, 2017-09-14:
@sbourdeauducq 
I had issues with this config mode because either there were some signal integrity issues (I tried with lowest frequency) or the bitstream I generetad was wrong. The DONE signal did not go high because was blocked by Artix, so it is probably the second case. That's why I repopulated resistors to support direct configuration of Artix by Kintex using IOs and IP block.

jbqubit, 2017-09-26:
Moving email discussion to this Issue....

-------------
@jbqubit said:

Greg, did you check again if "the bitstream I generetad was wrong"? SB, what is cost and time to develop gateware to load the RTM FPGA from the AMC FPGA?

SB> It's not a dependency, the RTM FPGA can always be loaded by JTAG.

If JTAG is how Florent is currently configuring the RTM FPGA and he has no problems, this seems like a fine approach to get started. Assuming JTAG, what's your expected timeline for support of SAWG RF
output from Sayma at the level of #795?

@gkasprow replied: 

No, I didn't. The board Florent has is wired in such way that Artix is loaded by Kintex. Mine can be booted simultaneously. I will play with that. As far as I remember our discussion, this was plan B, while plan A was direct loading of Artix by Kinex using dedicated IP core. If for some reason simultaneous configuration does not work, we can consider adding second FLASH for Artix.

sbourdeauducq, 2017-09-26:
> Moving email discussion to this Issue....

Let's keep this discussion in emails. Most of your message above is off-topic.

jbqubit, 2017-12-14:
What's the status of this?

sbourdeauducq, 2017-12-31:
@jbqubit regarding status, see the multiple emails I sent you, e.g. "Re: RF output from Sayma via ARTIQ" on October 18th. The ball is in your court.

But, when I come to think of this, I think having the RTM FPGA under AMC control and loaded by the AMC gateware/firmware would be more robust, more transparent, and less frustrating.

For example, with the Xilinx system, if I understand correctly: if the RTM FPGA becomes unconfigured for some reason (e.g. reset from JTAG) it has no way of reading the flash on its own. The board would become inoperable until it is fully reset (power cycle or JTAG reset of the AMC FPGA). The AMC firmware also could not trigger reloads of the RTM FPGA, which are handy when the RTM FPGA gateware locks up, and to make sure that we start with a known state on the RTM side after a firmware reboot.

jbqubit, 2018-01-02:
Here's that email thread. -Joe 

Forwarded conversation
Subject: RF output from Sayma via ARTIQ
------------------------

From: Joe Britton <joe.britton.arl@gmail.com>
Date: Sun, Sep 24, 2017 at 8:33 PM
To: Sebastien B <sb@m-labs.hk>, Robert Jördens <rj@m-labs.hk>, Thomas Harty <thomas.peter.harty@gmail.com>, "Slichter, Daniel H. (Fed)" <daniel.slichter@nist.gov>, Greg Kasprowicz <kasprowg@gmail.com>

SB, RJ, what's your expected timeline for support of SAWG RF output
from Sayma? AFACT the following are dependencies...

https://github.com/m-labs/artiq/issues/798
https://github.com/m-labs/artiq/issues/813
https://github.com/m-labs/artiq/issues/794
https://github.com/m-labs/artiq/issues/791

Isn't #791 done? It's the infamous CLIN0002...

How about asynchronous toggling of TTLs on Allaki?

 -Joe

----------
From: Sébastien Bourdeauducq <sb@m-labs.hk>
Date: Sun, Sep 24, 2017 at 10:38 PM
To: Joe Britton <joe.britton.arl@gmail.com>, Robert Jördens <rj@m-labs.hk>, Thomas Harty <thomas.peter.harty@gmail.com>, "Slichter, Daniel H. (Fed)" <daniel.slichter@nist.gov>, Greg Kasprowicz <kasprowg@gmail.com>

On Monday, September 25, 2017 08:33 AM, Joe Britton wrote:
> https://github.com/m-labs/artiq/issues/813

What is your approach with this? Do you want to get the hardware issue fixed (if that's not just Greg having generated bitstreams incorrectly), or pay us to develop an additional gateware core to load the RTM FPGA?

It's not a dependency, the RTM FPGA can always be loaded by JTAG.

> Isn't #791 done? It's the infamous CLIN0002...

It is done, but I don't have RTM hardware to test it. Greg, did you make more boards?

> How about asynchronous toggling of TTLs on Allaki?

It is not funded. Within the current contract, there will be only two versions, one with hardcoded TTL values in the gateware for initial testing, and another with DRTIO TTL.

Sébastien

----------
From: Joe Britton <joe.britton.arl@gmail.com>
Date: Sun, Sep 24, 2017 at 10:55 PM
To: Sébastien Bourdeauducq <sb@m-labs.hk>
Cc: Robert Jördens <rj@m-labs.hk>, Thomas Harty <thomas.peter.harty@gmail.com>, "Slichter, Daniel H. (Fed)" <daniel.slichter@nist.gov>, Greg Kasprowicz <kasprowg@gmail.com>

Hi SB.

>> https://github.com/m-labs/artiq/issues/813
>
> What is your approach with this? Do you want to get the hardware issue fixed
> (if that's not just Greg having generated bitstreams incorrectly), or pay us
> to develop an additional gateware core to load the RTM FPGA?

Thanks for the reminder that this is still an Issue. Greg, did you
check again if "the bitstream I generetad was wrong" #813? SB, what is
cost and time to develop gateware to load the RTM FPGA from the AMC
FPGA?

> It's not a dependency, the RTM FPGA can always be loaded by JTAG.

If JTAG is how Florent is currently configuring the RTM FPGA and he
has no problems, this seems like a fine approach to get started.
Assuming JTAG, what's your expected timeline for support of SAWG RF
output from Sayma at the level of #795?

>> How about asynchronous toggling of TTLs on Allaki?
> It is not funded. Within the current contract, there will be only two
> versions, one with hardcoded TTL values in the gateware for initial testing,
> and another with DRTIO TTL.

I'm not asking for a new feature. Just wanting to know if Greg will
have what he needs to test Allaki this week.

-Joe

----------
From: Grzegorz Kasprowicz <kasprowg@gmail.com>
Date: Mon, Sep 25, 2017 at 6:48 PM
To: Joe Britton <joe.britton.arl@gmail.com>, Sébastien Bourdeauducq <sb@m-labs.hk>
Cc: Robert Jördens <rj@m-labs.hk>, Thomas Harty <thomas.peter.harty@gmail.com>, "Slichter, Daniel H. (Fed)" <daniel.slichter@nist.gov>

>Thanks for the reminder that this is still an Issue. Greg, did you check again if "the bitstream I generetad was wrong" #813? SB, what is cost >and time to develop gateware to load the RTM FPGA from the AMC FPGA?
No, I didn't. The board Florent has is wired in such way that Artix is loaded by Kintex. Mine can be booted simultaneously. I will play with that.
As far as I remember our discussion, this was plan B, while plan A was direct loading of Artix by Kinex using dedicated IP core.
If for some reason simultaneous configuration does not work, we can consider adding second FLASH for Artix.

----------
From: Joe Britton <joe.britton.arl@gmail.com>
Date: Tue, Sep 26, 2017 at 12:06 PM
To: Grzegorz Kasprowicz <kasprowg@gmail.com>
Cc: Sébastien Bourdeauducq <sb@m-labs.hk>, Robert Jördens <rj@m-labs.hk>, Thomas Harty <thomas.peter.harty@gmail.com>, "Slichter, Daniel H. (Fed)" <daniel.slichter@nist.gov>

Moved the discussion related to configuring the RTM FPGA to the relevant Issue.

https://github.com/m-labs/artiq/issues/813

----------
From: Robert Jördens <rj@m-labs.hk>
Date: Mon, Oct 2, 2017 at 2:04 PM
To: Joe Britton <joe.britton.arl@gmail.com>
Cc: Sébastien Bourdeauducq <sb@m-labs.hk>, Thomas Harty <thomas.peter.harty@gmail.com>, "Slichter, Daniel H. (Fed)" <daniel.slichter@nist.gov>, Greg Kasprowicz <kasprowg@gmail.com>

On Mon, Sep 25, 2017 at 4:55 AM, Joe Britton <joe.britton.arl@gmail.com> wrote:
> Hi SB.
>
> If JTAG is how Florent is currently configuring the RTM FPGA and he
> has no problems, this seems like a fine approach to get started.
> Assuming JTAG, what's your expected timeline for support of SAWG RF
> output from Sayma at the level of #795?

We'll work on that in the six weeks allocated for that after getting
one Sayma to work.

Robert.

----------
From: Sébastien Bourdeauducq <sb@m-labs.hk>
Date: Tue, Oct 17, 2017 at 1:21 PM
To: Joe Britton <joe.britton.arl@gmail.com>

On Monday, September 25, 2017 10:55 AM, Joe Britton wrote:
>
> SB, what is
> cost and time to develop gateware to load the RTM FPGA from the AMC
> FPGA?

I think around 7k, and a few weeks (depending on bugs and other things in parallel).

jbqubit, 2018-01-02:
@sbourdeauducq Don't pull passive aggressive bullshit like this (or [this](https://github.com/m-labs/artiq/issues/853)). M-Labs is being paid for "Project management for Sayma and Metlino testing and integration." If you need an additional $7k to enable something as basic as Sayma RF output, don't sit on your hands. Send me a quote, give me a call, post in this Issue, pester the Monroe team, yell at the mailing list that @jbqubit is holding up the project over a trifle. Everyone wants to support you to make Sayma work with ARTIQ. If you're not vocal about what you need things like this don't stay on my radar. 

> But, when I come to think of this, having the RTM FPGA under AMC control and loaded by the AMC gateware/firmware would be more robust, more transparent, and less frustrating.

Agreed that this sounds like a better solution. If the cost is <$5k I can pay you by credit card.

sbourdeauducq, 2018-02-07:
https://github.com/m-labs/artiq/issues/908#issuecomment-363650534

sbourdeauducq, 2018-02-07:
Note: my SAWG test (cb0016ceeeaa0e376a9b13ab5d2ea7ba7495100b) actually did include this core, so it may be possible to still make it work.

whitequark, 2018-02-08:
Great. Trash my work for no good reason.

sbourdeauducq, 2018-02-08:
That's not "trashing" it, as you can see in the commit it is simply commented out. People want to test the SAWG so master has to work, and disabling that core is fast and has currently no user impact.

sbourdeauducq, 2018-02-08:
But yes, FPGA development is often frustrating. I know this very well.

jordens, 2018-02-28:
This should work now. Needs the change in https://github.com/m-labs/sinara/issues/249#issuecomment-321300672

hartytp, 2018-05-03:
Out of curiosity, what's the status of this? Did it get funded?

sbourdeauducq, 2018-05-03:
Not yet, but it should be soon.

hartytp, 2018-05-03:
cool. thanks for the info.

sbourdeauducq, 2018-06-01:
Funded by UMD.

sbourdeauducq, 2018-06-01:
@gkasprow What are good test points to observe RTM FPGA bitstream loading signals as close as possible to the RTM FPGA?

We need to probe: CCLK, DIN, DONE, INIT_B and PROGRAM_B.

There is again some weird problem with this, and it frustratingly refuses to work.

gkasprow, 2018-06-01:
It's easiest to look at the resistors close to the RTM connector..
I'll post an image in a moment.

gkasprow, 2018-06-01:
@sbourdeauducq look here
[image]

jordens, 2018-06-12:
I think this won't work because FPGA_CFG_DIN (on RTM) is not connected to IO_L1N_T0_D01_DIN_14 (L17) but to IO_L1P_T0_D00_MOSI_14 (K16). That's close but not close enough.
@gkasprow What's the easiest detailed way to rework that? I.e. cut the DAC2_TXEN0 wire, preferably short DAC2_TXEN0 to DAC2_TXEN1, DNP R46, route a wire to the FPGA side of the cut DAC2_TXEN0 trace to the AMC side of R46.

hartytp, 2018-06-12:
@jordens good catch.

Yes, if @gkasprow posts a guide, I'll give that a go.

gkasprow, 2018-06-12:
I wouln't connect MOSI line with DIN because I'm not sure if it is high-Z during config.
The easiest way is to remove R46 and connect one pin of the resistor with DAC2_TXEN0. The 200R resistor must be kept in place because DIN is in 1.8V bank and we drive it with 3.3V bank.
Moreover, if we want to use slave-serial mode, M2:0 must be high, so one needs to desolder R112 and short it with R116

DAC2_TXEN0 is located here: 
[image]

R46 is here: 
[image]
Disconnect its left pin, one close to the I2C connector, and connect the right pad via 200R  to the DAC pin.

gkasprow, 2018-06-12:
R112 is here
[image]

[image]

gkasprow, 2018-06-12:
[image]

jordens, 2018-06-13:
Given the tightness of the layout that sounds good. Needs disabling of TXEN in the DACs, the gateware, and control by SPI.

sbourdeauducq, 2018-06-16:
Working correctly after rework.

closed 2018-06-16 by sbourdeauducq

hartytp, 2018-06-16:
:+1: cool

jbqubit, 2018-06-18:
Glad to hear it.

klickverbot, 2018-06-18:
We should probably collect all the rework descriptions on a Wiki page/… for later reference.

gkasprow, 2018-06-18:
It's here
https://github.com/sinara-hw/sinara/issues/216

hartytp, 2018-06-20:
Summarising: rework for this is
- DNP R176-R180 and add 0R to R200-204 on Sayma AMC.  These are: `RTM_FPGA_CFG_CCLK`, `RTM_FPGA_CFG_DIN`, `RTM_FPGA_CFG_DONE`, `RTM_FPGA_CFG_INIT_B`, `RTM_FPGA_CFG_PROGRAM_B`. Locations of these resistors are here https://github.com/sinara-hw/sinara/issues/209#issuecomment-369766285 https://github.com/sinara-hw/sinara/issues/209#issuecomment-369766805
- https://github.com/m-labs/artiq/issues/813#issuecomment-396749798

hartytp, 2018-06-22:
After the rework, this also works for me. Thanks!