m-labs/artiq #817: VHDCI for Sayma

jordens, 2017-08-22:
https://github.com/m-labs/artiq/compare/master...jbqubit:vhdci_play

and

Here's the FMC -> VHDCI adapter pins file we were looking at: https://github.com/npisenti/artiq/blob/jqi-dev/artiq/gateware/jqi_fmc_vhdci.py

/cc npisenti jbqubit

jbqubit, 2017-09-01:
Initialization routine for [Creotech FMC DIO 32ch lvds a v1.2](www.ohwr.org/projects/fmc-dio-32chlvdsa/wiki) is here: 

https://github.com/jbqubit/artiq/blob/vhdci_play/artiq/examples/phaser/repository/init_vhdci.py

sbourdeauducq, 2018-07-18:
Done, needs testing.

sbourdeauducq, 2018-07-19:
This works except for the FMC card line direction control.
@gkasprow Are there different versions of that card with different permutations of the shift register outputs, by any chance? Is this correct: https://github.com/m-labs/artiq/blob/fe93a454d68b6e6db697c0c0c6f550ec95bbb691/artiq/coredevice/fmcdio_vhdci_eem.py#L9

The code loads the shift register MSB first:
https://github.com/m-labs/artiq/blob/fe93a454d68b6e6db697c0c0c6f550ec95bbb691/artiq/coredevice/shiftreg.py#L18

sbourdeauducq, 2018-07-19:
Thanks. The code I linked was correct, but I was doing something daft in my test kernel. Things are working correctly now.

closed 2018-07-19 by sbourdeauducq