m-labs/artiq #821: Example Ion Trap Code

dtcallcock, 2017-08-25:
To get from a bare ARTIQ installation to a working ion trap experiment requires writing a lot of code and making a lot of low level decisions about how to handle things like data and scanning variables.  This seems to be having two effects:

- Smaller groups are put off using ARTIQ as they don't have the resources to do this.  A group leader at NACTI said they were setting up a new trap and asked me how much work it would be to get it up and running assuming they had turnkey hardware (ie Kasli and a bunch of EEMs).  Based on our experience I said it would be 3-6 months depending on how much previous programming the student had done.  This was clearly too long and they said they'd probably just replicate the kludgy pile of odds and ends they’re running their current trap off. 

- Groups that are planning to use ARTIQ are starting to request code examples from our lab to get an idea of what they need to do.

One solution to both of these problems would be to maintain a public repository with some ‘generic ion trap’ example code, which might eventually include a wide variety of examples submitted by different groups.  It would be important that the code be clean, well-commented, PEP 8 compliant, etc.  One way of doing this fairly painlessly would be for NIST to work with a group that is switching to ARTIQ and help them write good, well documented code that can be published.

sbourdeauducq, 2017-08-25:
This sounds like a good topic for the mailing list. Care if I forward it and reply there?

dtcallcock, 2017-08-25:
Sure.

closed 2017-08-25 by sbourdeauducq