m-labs/artiq #827: bug in newly documented artiq-dev build work flow

jbqubit, 2017-09-01:
Working with @arpitagrawal23 ... Following the work flow here:

http://www.m-labs.hk/artiq/manual-master/developing.html#artiq-anaconda-development-environment

```
git clone --recursive https://github.com/m-labs/artiq ~/artiq-dev/artiq
cd artiq-dev/artiq
source activate artiq-dev
pip install -e .
python3 -m artiq.gateware.targets.phaser
```

```
$ python3 -m artiq.gateware.targets.phaser
make: Entering directory `/home/iirabi/artiq-dev/artiq/misoc_phaser_kc705/software/bios'
 CC       main.o
 LD       bios.elf
 OBJCOPY  bios.bin
make: Leaving directory `/home/iirabi/artiq-dev/artiq/misoc_phaser_kc705/software/bios'
make: Entering directory `/home/iirabi/artiq-dev/artiq/misoc_phaser_kc705/software/ksupport'
 CARGO    cargo/or1k-unknown-none/debug/libksupport.a
   Compiling ksupport v0.0.0 (file:///home/iirabi/artiq-dev/artiq/artiq/firmware/ksupport)
error[E0308]: mismatched types
   --> /home/iirabi/artiq-dev/artiq/artiq/firmware/ksupport/lib.rs:387:25
    |
387 |             if status & rtio::RTIO_O_STATUS_UNDERFLOW != 0 {
    |                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected u8, found u32

error[E0277]: the trait bound `u8: core::ops::BitAnd<u32>` is not satisfied
   --> /home/iirabi/artiq-dev/artiq/artiq/firmware/ksupport/lib.rs:387:16
    |
387 |             if status & rtio::RTIO_O_STATUS_UNDERFLOW != 0 {
    |                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `core::ops::BitAnd<u32>` is not implemented for `u8`
    |
    = note: no implementation for `u8 & u32`

error[E0308]: mismatched types
   --> /home/iirabi/artiq-dev/artiq/artiq/firmware/ksupport/lib.rs:393:25
    |
393 |             if status & rtio::RTIO_O_STATUS_SEQUENCE_ERROR != 0 {
    |                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected u8, found u32

error[E0277]: the trait bound `u8: core::ops::BitAnd<u32>` is not satisfied
   --> /home/iirabi/artiq-dev/artiq/artiq/firmware/ksupport/lib.rs:393:16
    |
393 |             if status & rtio::RTIO_O_STATUS_SEQUENCE_ERROR != 0 {
    |                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `core::ops::BitAnd<u32>` is not implemented for `u8`
    |
    = note: no implementation for `u8 & u32`

error[E0308]: mismatched types
  --> /home/iirabi/artiq-dev/artiq/artiq/firmware/ksupport/rtio.rs:41:44
   |
41 |         while csr::rtio::o_status_read() & RTIO_O_STATUS_WAIT != 0 {}
   |                                            ^^^^^^^^^^^^^^^^^^ expected u8, found u32

error[E0277]: the trait bound `u8: core::ops::BitAnd<u32>` is not satisfied
  --> /home/iirabi/artiq-dev/artiq/artiq/firmware/ksupport/rtio.rs:41:15
   |
41 |         while csr::rtio::o_status_read() & RTIO_O_STATUS_WAIT != 0 {}
   |               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `core::ops::BitAnd<u32>` is not implemented for `u8`
   |
   = note: no implementation for `u8 & u32`

error[E0308]: mismatched types
  --> /home/iirabi/artiq-dev/artiq/artiq/firmware/ksupport/rtio.rs:60:36
   |
60 |         csr::rtio::o_address_write(addr as u32);
   |                                    ^^^^^^^^^^^ expected u16, found u32

error[E0308]: mismatched types
  --> /home/iirabi/artiq-dev/artiq/artiq/firmware/ksupport/rtio.rs:65:60
   |
65 |             process_exceptional_status(timestamp, channel, status);
   |                                                            ^^^^^^ expected u32, found u8
   |
   = help: here are some functions which might fulfill your needs:
           - .count_ones()
           - .count_zeros()
           - .leading_zeros()
           - .trailing_zeros()

error[E0308]: mismatched types
  --> /home/iirabi/artiq-dev/artiq/artiq/firmware/ksupport/rtio.rs:75:36
   |
75 |         csr::rtio::o_address_write(addr as u32);
   |                                    ^^^^^^^^^^^ expected u16, found u32

error[E0308]: mismatched types
  --> /home/iirabi/artiq-dev/artiq/artiq/firmware/ksupport/rtio.rs:82:60
   |
82 |             process_exceptional_status(timestamp, channel, status);
   |                                                            ^^^^^^ expected u32, found u8
   |
   = help: here are some functions which might fulfill your needs:
           - .count_ones()
           - .count_zeros()
           - .leading_zeros()
           - .trailing_zeros()

error[E0308]: mismatched types
  --> /home/iirabi/artiq-dev/artiq/artiq/firmware/ksupport/rtio.rs:95:22
   |
95 |             status = csr::rtio::i_status_read();
   |                      ^^^^^^^^^^^^^^^^^^^^^^^^^^ expected u32, found u8
   |
   = help: here are some functions which might fulfill your needs:
           - .count_ones()
           - .count_zeros()
           - .leading_zeros()
           - .trailing_zeros()

error[E0308]: mismatched types
   --> /home/iirabi/artiq-dev/artiq/artiq/firmware/ksupport/rtio.rs:119:22
    |
119 |             status = csr::rtio::i_status_read();
    |                      ^^^^^^^^^^^^^^^^^^^^^^^^^^ expected u32, found u8
    |
    = help: here are some functions which might fulfill your needs:
            - .count_ones()
            - .count_zeros()
            - .leading_zeros()
            - .trailing_zeros()

error: aborting due to 12 previous errors

error: Could not compile `ksupport`.

To learn more, run the command again with --verbose.
make: *** [cargo/or1k-unknown-none/debug/libksupport.a] Error 101
make: Leaving directory `/home/iirabi/artiq-dev/artiq/misoc_phaser_kc705/software/ksupport'
Traceback (most recent call last):
  File "/home/iirabi/anaconda3/envs/artiq-dev/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/iirabi/anaconda3/envs/artiq-dev/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/iirabi/artiq-dev/artiq/artiq/gateware/targets/phaser.py", line 269, in <module>
    main()
  File "/home/iirabi/artiq-dev/artiq/artiq/gateware/targets/phaser.py", line 265, in main
    build_artiq_soc(soc, builder_argdict(args))
  File "/home/iirabi/artiq-dev/artiq/artiq/gateware/amp/soc.py", line 52, in build_artiq_soc
    builder.build()
  File "/home/iirabi/anaconda3/envs/artiq-dev/lib/python3.5/site-packages/misoc/integration/builder.py", line 180, in build
    self._generate_software()
  File "/home/iirabi/anaconda3/envs/artiq-dev/lib/python3.5/site-packages/misoc/integration/builder.py", line 157, in _generate_software
    subprocess.check_call(cmd)
  File "/home/iirabi/anaconda3/envs/artiq-dev/lib/python3.5/subprocess.py", line 271, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', '/home/iirabi/artiq-dev/artiq/misoc_phaser_kc705/software/ksupport']' returned non-zero exit status 2

```

sbourdeauducq, 2017-09-01:
Your misoc version does not match the one prescribed in the conda-dev recipe.

closed 2017-09-01 by sbourdeauducq

arpitagrawal23, 2017-09-06:
Hi @sbourdeauducq 
When I execute this command:

`conda env create -f conda/artiq-dev.yaml`

it installed misoc version py_64+git7d9e6057, even though conda/artiq-dev/meta.yaml mentions misoc git hash as py_35+gitd6f86c03. So it seems conda env command is not honouring git hashs mentioned in the yaml file and instead install taking whatever version that is present in the system's cache. I had to execute this command manually:

`conda install "misoc=0.6.dev=py_35+gitd6f86c03"`

to make sure I have right version of misoc in my conda env. I think it'd great if we modify conda env create command in the installation procedure in the manual so that it will install exact same versions mentioned in the yaml file.

jordens, 2017-09-06:
@arpitagrawal23 can you reproduce that (please try)? Or is it just because your enviroonment got out of sync with the working tree?

arpitagrawal23, 2017-09-06:
Hi @jordens 
Yes, it's reproducible. Even after deleting folder ~/anaconda3 and doing a fresh installation of anadonda, adding channels (main as well as dev), cloning artiq git repo and following this procedure:
http://www.m-labs.hk/artiq/manual-master/developing.html#artiq-anaconda-development-environment
so that it downloads all packages from the internet again, I'm getting same error.

jordens, 2017-09-06:
I am not referring to the top post, but to your analysis. When doing what you said you did, which misoc version does it install? And which one is in `artiq-dev/meta.yaml`?

NB: I am pretty certain that the system and user python trees are thoroughly messed up on one (that?) machine in the lab. Reinstalling anaconda won't change a thing. We have previously explained several times that unless that's fixed we can't help. We have yet to see an indication that it was fixed.

arpitagrawal23, 2017-09-06:
It installed misoc version 0.6.dev py_64+git7d9e6057 but artiq-dev/meta.yaml file mentions misoc version as 0.6.dev py_35+gitd6f86c03.

jordens, 2017-09-06:
No, [it doesn't](https://github.com/m-labs/artiq/blob/7fe5d73/conda/artiq-dev/meta.yaml#L18). You said you "cloned the artiq git repo".

arpitagrawal23, 2017-09-06:
Actually I had done that at on 9/1 at 15:16 EST, right after @sbourdeauducq had commented on this thread. Had also taken a picture which shows that all modules were downloaded from the internet (and not taken from system's cache).
[image]

jordens, 2017-09-06:
Come on. Nobody can help you if you don't take the questions seriously. Can you reproduce it **now** and using **the steps you listed**? Anything else is less than helpful.

I can't reproduce it. And I tried. Now.

And please also ensure that you user and system python are not messed up with any residual installations of migen/misoc/artiq/etc or any of the dependencies. I.e. check especially `~/.local/lib/python*` and `/usr/local/lib/python*` and any `PYTHON*` environment variables. Those should be empty. Report back confirming that you have done so.

arpitagrawal23, 2017-09-06:
Apologies for that. But actually I didn't need any help on this topic. My setup started working fine after explicitly installing the right misoc version by giving conda install command (mentioned in my earlier comments). I just wanted to report that conda env create command is not honouring git hash mentioned in conda/artiq-dev/meta.yaml file and if someone think that it is a valid issue, then it should be fixed.

I've executed these commands right now:

```
rm ~/anaconda3 -Rf
rm ~/.local/lib/python3.5 -Rf
wget https://repo.continuum.io/archive/Anaconda3-4.4.0-Linux-x86_64.sh
chmod +x ./Anaconda3-4.4.0-Linux-x86_64.sh 
./Anaconda3-4.4.0-Linux-x86_64.sh 
conda
git clone --recursive https://github.com/m-labs/artiq
cd artiq/
conda env create -f conda/artiq-dev.yaml
source activate artiq-dev
conda list
conda list misoc
```

and the output is:
```
# packages in environment at /home/arpit23/anaconda3/envs/artiq-dev:
#
misoc                     0.6.dev         py_64+git7d9e6057    m-labs/label/dev
```

The version `py_64+git7d9e6057` is not matching what is present in the yaml file `py35_65+git6abc1b4c`.

However, if I give this command:
```
conda install "misoc=0.6.dev=py35_65+git6abc1b4c"
conda list misoc
# packages in environment at /home/arpit23/anaconda3/envs/artiq-dev:
#
misoc                     0.6.dev         py35_65+git6abc1b4c    m-labs/label/dev
```
then the version matches. If you think this is a valid issue, you might like to fix it for other new users of ARTIQ. I'm okay giving conda install commands.

Note: I have not done anything to /usr/local/lib/python* folders since I don't want to disturb this machine. I wanted to do things in my account only.

sbourdeauducq, 2017-09-06:
> I just wanted to report that conda env create command is not honouring git hash mentioned in conda/artiq-dev/meta.yaml file

@mntng also seemed to have this problem. If it is confirmed, perhaps it has to do with the contents of ``condarc`` and/or some conda bug.

sbourdeauducq, 2017-09-07:
> I'm okay giving conda install commands.

That's not a good solution:
1. conda should have installed the correct misoc version when installing artiq-dev. If not, this is a software bug and/or conda misconfiguration, and the root cause should be found and the problem fixed there. Note that this works correctly on the buildbot.
2. the conda install command would have to be kept in sync with the contents of the artiq-dev yaml file. This adds to the maintainance work and is prone to inconsistencies.

jordens, 2017-09-07:
@arpitagrawal23 could you paste your `~/.condarc` and conda version?
Also please do check that there are no stray fragments of any previous installation in the directories mentioned.

arpitagrawal23, 2017-09-08:
Okay. Here it is:

```
$ cat ~/.condarc 
channels:
  - http://conda.anaconda.org/m-labs/label/dev
  - http://conda.anaconda.org/m-labs/label/main
  - http://conda.anaconda.org/conda-forge/label/main
  - defaults
$ conda --version
conda 4.3.21
$ which conda
/home/arpit23/anaconda3/envs/artiq-dev/bin/conda
$ 
```

I have downloaded Anaconda using wget command specified in one of my above comments. I got that URL from Anaconda download.

jbqubit, 2017-09-08:
@jordens Thank you for tagging. Also please reopen Issue.