m-labs/artiq #828: artiq.coredevice.comm_kernel.LoadError: cannot load kernel: symbol lookup error: floor

mntng, 2017-09-02:
got this error in the last line of this file: https://github.com/m-labs/artiq/blob/master/artiq/examples/master/repository/coredevice_examples/simple/ad5360.py/

sbourdeauducq, 2017-09-09:
@whitequark Is this the proper solution:
```
--- a/artiq/firmware/ksupport/api.rs
+++ b/artiq/firmware/ksupport/api.rs
@@ -71,6 +71,7 @@ static mut API: &'static [(&'static str, *const ())] = &[
     /* libm */
     api!(sqrt),
     api!(round),
+    api!(floor),
 
     /* exceptions */
     api!(_Unwind_Resume = ::unwind::_Unwind_Resume),
```

Why was it apparently not a problem before? @mntng is the issue present with ARTIQ-2?

whitequark, 2017-09-09:
@sbourdeauducq This is the proper solution. Likely what happened is LLVM decided to not inline or not unroll something, and an expression that was previously constant-folded is not, anymore. This should be backported to ARTIQ-2.

commit 928cffb0 (2017-09-10): firmware: export floor() in ksupport. Closes #828
commit 5c5f86cd (2017-09-10): runtime: export floor() in ksupport. Closes #828
closed 2017-09-10 by sbourdeauducq