m-labs/artiq #835: Compiler crash on list addition

cjbe, 2017-09-29:
The following code causes the compiler to error out:
```python
class ListAddition(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        a = [0,1]
        a += [2,3]
```

Error message:
```
====== LLVM IR (BROKEN) DUMP ======
LLVM IR (broken) dumped as /tmp/tmp7fgywq1d.ll
Traceback (most recent call last):
  File "/home/artiq/.local/bin/artiq_run", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_run')()
  File "/home/artiq/artiq-dev/artiq/artiq/frontend/artiq_run.py", line 219, in main
    return run(with_file=True)
  File "/home/artiq/artiq-dev/artiq/artiq/frontend/artiq_run.py", line 205, in run
    raise exn
  File "/home/artiq/artiq-dev/artiq/artiq/frontend/artiq_run.py", line 198, in run
    exp_inst.run()
  File "/home/artiq/artiq-dev/artiq/artiq/language/core.py", line 53, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/artiq/artiq-dev/artiq/artiq/coredevice/core.py", line 124, in run
    self.compile(function, args, kwargs, set_result)
  File "/home/artiq/artiq-dev/artiq/artiq/coredevice/core.py", line 107, in compile
    library = target.compile_and_link([module])
  File "/home/artiq/artiq-dev/artiq/artiq/compiler/targets.py", line 180, in compile_and_link
    return self.link([self.assemble(self.compile(module)) for module in modules])
  File "/home/artiq/artiq-dev/artiq/artiq/compiler/targets.py", line 180, in <listcomp>
    return self.link([self.assemble(self.compile(module)) for module in modules])
  File "/home/artiq/artiq-dev/artiq/artiq/compiler/targets.py", line 137, in compile
    llparsedmod = llvm.parse_assembly(str(llmod))
  File "/home/artiq/anaconda3/envs/artiq-dev/lib/python3.5/site-packages/llvmlite_artiq/binding/module.py", line 22, in parse_assembly
    raise RuntimeError("LLVM IR parsing error\n{0}".format(errmsg))
RuntimeError: LLVM IR parsing error
<string>:63:19: error: invalid operand type for instruction
  %"UNN.10" = add {i32*, i32} %"val.ENV.a", %".21", !dbg !20
```

whitequark, 2017-10-02:
Now a diagnostic:
```
input.py:2:3-2:5: error: values cannot be mutated in-place
a += [3,4]
~ ^^      
input.py:2:1-2:11: note: try using `a = a + [3,4]`
a += [3,4]
^^^^^^^^^^
```

sbourdeauducq, 2017-10-02:
Python has some mechanism for turning in-place mutation into the other form automatically:
https://stackoverflow.com/questions/39402501/difference-between-add-and-iadd
Anyway, not supporting this is fine.

whitequark, 2017-10-02:
That's not the issue here. The real reason __iadd__ is not supported is that:
  * we only support them for lists in first place, and
  * our lists use fat pointers, and the following snippet cannot be made to work properly:
```
a = b = [1,2]
a += [3,4]
print(b) # should be [1,2,3,4]
```

whitequark, 2017-10-02:
Actually, now that I think about it, the check should only apply to lists. And we have another bug elsewhere.

commit e59b301a (2017-10-02): compiler: disallow op= on mutable values (fix #835).

In general, we can't reallocate a value in earlier stack frames,
or fixup references to it. This mainly impacts lists.
commit db8300c9 (2017-10-02): compiler: disallow op= on mutable values (fix #835).

In general, we can't reallocate a value in earlier stack frames,
or fixup references to it. This mainly impacts lists.
commit 8e3959d7 (2017-10-02): compiler: disallow op= on mutable values (fix #835).

In general, we can't reallocate a value in earlier stack frames,
or fixup references to it. This mainly impacts lists.
commit ca254ec5 (2017-10-02): compiler: disallow op= on mutable lists only (fix #835).

This only really applies to lists since those use fat pointers.
`x.y += z` is fine.
closed 2017-10-02 by whitequark