m-labs/artiq #837: KC705 MAC sensitivity to Ethernet switch model

cjbe, 2017-10-03:
Using Artiq 3 on a KC705 I see a very large change in transfer rate depending on what model of Ethernet switch I connect the KC705 to. In some cases the transfer rate is essentially zero.

Switch models I have tested:
Netgear FS605 v3 : 10/100M cheapo switch
DLink DES-108 : 10/100M switch
DLink DGS-108 : 10/100/1000M switch
Netgear ProSafe 16 JGS516 : 10/100/1000M switch

The two 10/100M switches work without issue, giving transfer rates of 2.3-2.4 MB/s from master to core device, and vice versa.

The DGS-108 switch at the auto-negotiated 1000M gives transfer rates of ~20 kB/s master to core, but the normal 2.4 MB/s core to master ([dump](https://github.com/m-labs/artiq/files/1352468/dgs108_1000M.pcapng.zip)).

The Netgear Prosafe 16 switch at the auto-negotiated 1000M does not respond at all. The core device only responds to ~1 in 100 pings. Modifying the runtime to print out all packets received by the MAC shows that very few packets are received by the MAC (i.e. the lack of ping response is due to the ICMP request packets only infrequently being output from the MAC).

I modified the runtime to access the PHY configuration registers (via MDIO). The PHY status registers are the identical for the two 1000M switches, showing that the difference in performance between the two switches is not at the link level.

Via MDIO I forced the PHY to 100M. With this I now see master to core transfer rates of ~200 kB/s on both 10/100/1000M switches.

Forcing the PHY to 10M gives master to core transfer rates of 500kB/s - 1MB/s for both 1000M switches.

So it seems like the system behaves worse with the 1000M capable switches, even when they are throttled down to 100M or even 10M.

jordens, 2017-10-03:
/cc @whitequark @enjoy-digital

whitequark, 2017-10-03:
I don't think there's anything I can do on network stack level if PHY doesn't hand me packets.

jordens, 2017-10-03:
AFAICT this is actually "received from the MAC" (hence the printing), with the MAC being involved as well. And IIRC there were changes to the MAC as well.

Have a look at the the dump. A big part of the problem are 2974 octet frames.

klickverbot, 2017-10-03:
> A big part of the problem are 2974 octet frames.

This might be automatic reassembly on the Linux side of things, though, unless @cjbe disabled it.

whitequark, 2017-10-03:
Is there anything reported in the core device log at the DEBUG log level?

cjbe, 2017-10-03:
@whitequark there is nothing interesting reported in the core device log

cjbe, 2017-10-03:
@jordens here is a [dump](https://github.com/m-labs/artiq/files/1353159/dgs108_1000M.pcapng.zip) with the same conditions as before, but with all segmentation offloading disabled - in this run the master-to-core rate was 88 kB/s.

jordens, 2017-10-03:
Has a good start,
[image]
then looses the full window,
[image]
takes **>3s** to recover the four lost packets with multiple retries being unanswered in the meantime
[image]
and does so a couple of times
[image]

whitequark, 2017-10-03:
@jordens I have an idea. Could this be related with the rate at which the MAC can fill the RX buffers versus the speed at which smoltcp can empty them? Maybe at 100M our four buffers *just happen* to be *just enough* for decent performance, and at 1000M they can be filled in less time than smoltcp can process even a single packet. (This didn't happen with lwip because lwip never achieved the 2.3 MB/s transfer rate in the first place.)

This can be easily verified by changing `_NIST_Ions.__init__` to pass `ethmac_nrxslots=6, ethmac_ntxslots=2` (I don't think we ever exhaust TX buffers, but you could also try `ethmac_ntxslots=4`), and then updating ethmac.rs in the obvious way. (If you wait a bit I'll make it easier to play with the buffer count on the Rust side.)

jordens, 2017-10-03:
@whitequark Maybe. But that doesn't explain the lack of any reaction to the resends for more than 3s.
From the trace, the situation is a bit different: it looses two packets even though there should be rx buffers in the MAC (there are only two in in use just being ACKed) and then does not react to anything that follows.
[image]

whitequark, 2017-10-03:
(Just pushed e7dac530 that makes ethmac experimentation easier.)

whitequark, 2017-10-03:
@cjbe Can you please:
  * uncomment `EthernetTracer` related lines in `runtime/lib.rs`,
  * switch core log to TRACE level,
  * run a transfer,
  * interrupt a transfer once it starts retransmitting,
  * retrieve the last core log messages together with a dump?

cjbe, 2017-10-04:
@whitequark I followed your instructions, setting the UART logging level to WARN so the system was not slowed down by UART traffic, and modifying the EthernetTracer to log as info, rather than just a straight print.

[corelog.txt](https://github.com/m-labs/artiq/files/1355324/corelog.txt)
[dump.pcapng.zip](https://github.com/m-labs/artiq/files/1355323/dump2.pcapng.zip)

whitequark, 2017-10-04:
@jordens No incoming packet after outgoing packet 17 makes it through to the core device at all. (This is not a case of the poll function not being called because otherwise it'd extract all four buffers in rapid succession.) Not sure what's going on.

jordens, 2017-10-04:
Well. We need to determine whether the packets are lost in the switch, in the PHY, or in the MAC, or in the smoltcp interface to the MAC. There are several options:

1. Try to find a switch where this happens that has this "port mirroring" feature and use that to trace packets. See [the advice from last year](https://github.com/m-labs/artiq/issues/456#issuecomment-238852015).
2. Add debugging circuitry to the PHY(-interface) and to the MAC. @enjoy-digital should probably jump in here.

Also note that the first SYN|ACK response from the core device got lost and needed retransmitting. That should be unlikely.

whitequark, 2017-10-04:
> Add debugging circuitry to the PHY(-interface) and to the MAC. @enjoy-digital should probably jump in here.

A frame counter in both the PHY and the MAC should be sufficient, and is very simple to implement.

klickverbot, 2017-10-23:
> A frame counter in both the PHY and the MAC should be sufficient, and is very simple to implement. I can do that, I think.

It would be great if you could follow up on this, if not done so already – we are a bit busy with putting things together at the moment, so we've just been working around the issue for now.

whitequark, 2017-10-23:
@klickverbot Sure, let me implement it.

sbourdeauducq, 2017-10-27:
Does it make sense that Oxford send us some problematic devices so that we can reproduce this locally? We also have a 1G switch here, I can try connecting the KC705 through it (it's on another 100M one at the moment for the current tests).

whitequark, 2017-10-27:
Yes.

sbourdeauducq, 2017-10-27:
Lab KC705 is now on the 1G switch. That switch then connects to the 100M switch, and the computer is on the 100M switch.

whitequark, 2017-10-27:
@sbourdeauducq Which 1G switch is it? I appear to be unable to reproduce the issue with it, at least.

sbourdeauducq, 2017-10-27:
Joe's switch that was breaking lwIP (#456) - Netgear GS608.

whitequark, 2017-10-30:
@cjbe I haven't been able to reproduce the issue on any of the network hardware we have. Please send us  both of the problematic switches. Before that, please recheck with the master branch, since I've added CRC error and dropped packet counters to MAC, and any changes in them will be logged at WARN.

cjbe, 2017-11-02:
@whitequark I have tested this again using the current gateware:

DLink DES-108 (100M) : ~4 occurrences of `ethernet mac: rx dropped: 1` per 10MB RPC transfer
DLink DGS-108 (1000M) : similar behaviour to the above
Netgear ProSafe 16 JGS516 (1000M) : no CRC errors or dropped packets, but still badly broken.

I have now tried this with a more KC705s, and there seems to be some difference between boards:

With the DGS-108, on one board (the one I used when opening the issue) I get ~20 kB/s, with another board I get ~1.0 MB/s, and with another I get ~1.7 MB/s (c.f. 2.4 MB/s with the 100M switch).

With the JGS516 I do not get any successful TCP transfers with any boards, from one I get ~98% ping loss, and with the other 2 I get ~70% ping loss.

I will send you a JGS516, as it seems like this has the best bet of reproducing the issue reliably. I assume I should post this to the HK address on the M-Labs website?

whitequark, 2017-11-02:
@cjbe Could you also send your DGS-108, just for completeness sake, and on the off chance there is a different bug with it?

cjbe, 2017-11-02:
@whitequark will do.

sbourdeauducq, 2017-11-03:
> I assume I should post this to the HK address on the M-Labs website?

Yes.

sbourdeauducq, 2017-11-10:
Switch received and installed between kc705 and main router.

sbourdeauducq, 2017-11-18:
Problem is very apparent even just with ping:
```
(artiq-main) sb@lab:~$ ping kc705
PING kc705.lab.m-labs.hk (192.168.1.50) 56(84) bytes of data.
64 bytes from kc705.lab.m-labs.hk (192.168.1.50): icmp_seq=1 ttl=64 time=0.290 ms
64 bytes from kc705.lab.m-labs.hk (192.168.1.50): icmp_seq=2 ttl=64 time=0.280 ms
64 bytes from kc705.lab.m-labs.hk (192.168.1.50): icmp_seq=3 ttl=64 time=0.274 ms
64 bytes from kc705.lab.m-labs.hk (192.168.1.50): icmp_seq=4 ttl=64 time=0.309 ms
64 bytes from kc705.lab.m-labs.hk (192.168.1.50): icmp_seq=5 ttl=64 time=0.297 ms
64 bytes from kc705.lab.m-labs.hk (192.168.1.50): icmp_seq=6 ttl=64 time=0.271 ms
64 bytes from kc705.lab.m-labs.hk (192.168.1.50): icmp_seq=7 ttl=64 time=0.305 ms
64 bytes from kc705.lab.m-labs.hk (192.168.1.50): icmp_seq=20 ttl=64 time=0.276 ms
64 bytes from kc705.lab.m-labs.hk (192.168.1.50): icmp_seq=27 ttl=64 time=0.315 ms
64 bytes from kc705.lab.m-labs.hk (192.168.1.50): icmp_seq=28 ttl=64 time=0.283 ms
64 bytes from kc705.lab.m-labs.hk (192.168.1.50): icmp_seq=30 ttl=64 time=0.273 ms
^C
--- kc705.lab.m-labs.hk ping statistics ---
45 packets transmitted, 11 received, 75% packet loss, time 45049ms
rtt min/avg/max/mdev = 0.271/0.288/0.315/0.022 ms
```

whitequark, 2017-11-27:
The packets are dropped because of (at least) preamble errors. AFAICT we don't actually need to check for preamble errors, just look for 55 55 D5 pattern:

> < azonenberg> as far as TX goes, the 4b5b has a control character for idle and one (well technically two in succession) for "start of stream"
> < azonenberg> So theoretically you could have the preamble be omitted and just have the dest MAC right after the SSD (start of stream descriptor)
> < azonenberg> Most compliant MACs would complain
> < azonenberg> It's basically there so you can easily bridge a 100baseTX frame to 10baseT without injecting and removing a preamble
> < azonenberg> The spec still requires it for this reason

sbourdeauducq, 2017-11-27:
The funny thing is that some packets still get through...

whitequark, 2017-11-27:
@sbourdeauducq If there is jitter of 1 octet at the beginning of the preamble that will also do the job. And it looks from my experiments like there's a probability of 4/5 that the preamble is not recognized.

jordens, 2017-12-06:
Triggering frame just on SFD instead of the preamble+SFD should fix this.
Waiting on https://github.com/m-labs/misoc/issues/64

jordens, 2017-12-06:
@cjbe: please test with current master.

jordens, 2017-12-06:
Works in our tests with the problematic switch.

closed 2017-12-06 by jordens

cjbe, 2017-12-06:
@jordens this seems to be a substantial improvement, but there still seems to be something odd going on.

I can now ping through all the problematic switches without loss (no loss in 1e4 packets).
Focusing on the JGS516, I now typically get transfer rates of 1.8 - 2.0 MB/s. However for a while (~20 runs of the transfer benchmark, ~10 minutes wall time) this dropped down to 50 kB/s. When this occurred I saw the following in the log:
```
[  4425.733542s]  INFO(runtime::mgmt): new connection from 10.255.6.32:54076
[  4433.832440s]  INFO(runtime::session): new connection from 10.255.6.32:50724
[  4434.239606s]  WARN(runtime): ethernet mac: rx dropped: 1
[  4434.385361s]  WARN(runtime): network error: truncated packet
[  4434.389762s]  WARN(runtime): ethernet mac: rx dropped: 4
[  4434.687018s]  WARN(runtime): ethernet mac: rx dropped: 5
[  4435.285560s]  WARN(runtime): ethernet mac: rx dropped: 6
[  4435.697234s]  WARN(runtime): ethernet mac: rx dropped: 7
[  4435.844028s]  WARN(runtime): ethernet mac: rx dropped: 8
[  4435.992076s]  WARN(runtime): ethernet mac: rx dropped: 9
[  4436.139521s]  WARN(runtime): ethernet mac: rx dropped: 10
[  4436.287049s]  WARN(runtime): ethernet mac: rx dropped: 12
[  4436.434579s]  WARN(runtime): ethernet mac: rx dropped: 13
[  4436.582112s]  WARN(runtime): ethernet mac: rx dropped: 15
[  4436.729639s]  WARN(runtime): ethernet mac: rx dropped: 16
[  4436.877173s]  WARN(runtime): ethernet mac: rx dropped: 19
[  4437.024700s]  WARN(runtime): ethernet mac: rx dropped: 23
[  4437.172228s]  WARN(runtime): ethernet mac: rx dropped: 27
```
This continued up to the message `... mac: rx dropped: 523`. Unfortunately the transfer rate went back to normal before I got a packet dump. With the transfer rate back to normal I did not see any dropped packets.

sbourdeauducq, 2017-12-07:
Please open a new issue, this looks different enough from the initial problem.