m-labs/artiq #838: Core device reset breaks communication with GUI TTL monitor

jonaskeller, 2017-10-07:
Running 3.0 on Windows, kc705 with nist_qc2 gateware

After an FPGA reset, the GUI TTL status indicators aren't updated anymore until the dashboard is restarted (which fixes all the issues described below).

The displayed directions are what they were before the reset, rather than "IN" for all channels. Clicking one of the "OVR" buttons produces the output

	WARNING:dashboard:quamash.QEventLoop:Event callback failed: [WinError 64] The specified network name is no longer available
	ERROR:dashboard:artiq.dashboard.moninj:lost connection to core device moninj

and further clicks produce

	WARNING:dashboard:asyncio:socket.send() raised exception.

The TTLs can be accessed normally from within experiments.

Sometimes (I'm afraid I couldn't determine how to reproducibly provoke this), the exception raised by the first click on "OVR" causes the dashboard to terminate:

	WARNING:dashboard:quamash.QEventLoop:Event callback failed: [WinError 64] The specified network name is no longer available
	Traceback (most recent call last):
	  File "C:\Anaconda3\envs\artiq3.0\lib\asyncio\windows_events.py", line 437, in finish_recv
		return ov.getresult()
	OSError: [WinError 64] The specified network name is no longer available

	During handling of the above exception, another exception occurred:

	Traceback (most recent call last):
	  File "C:\Anaconda3\envs\artiq3.0\lib\site-packages\Quamash-0.5.5-py3.6.egg\quamash\_windows.py", line 41, in _process_events
	  File "C:\Anaconda3\envs\artiq3.0\lib\asyncio\windows_events.py", line 440, in finish_recv
		raise ConnectionResetError(*exc.args)
	ConnectionResetError: [WinError 64] The specified network name is no longer available

	During handling of the above exception, another exception occurred:

	Traceback (most recent call last):
	  File "C:\Anaconda3\envs\artiq3.0\lib\site-packages\Quamash-0.5.5-py3.6.egg\quamash\_windows.py", line 44, in _process_events
	  File "C:\Anaconda3\envs\artiq3.0\lib\asyncio\windows_events.py", line 79, in set_exception
		super().set_exception(exception)
	  File "C:\Anaconda3\envs\artiq3.0\lib\asyncio\futures.py", line 361, in set_exception
		raise InvalidStateError('{}: {!r}'.format(self._state, self))
	asyncio.futures.InvalidStateError: CANCELLED: <_OverlappedFuture cancelled>

sbourdeauducq, 2017-10-07:
By "reset", you mean rebooting the FPGA board, not ``self.core.reset()``?

sbourdeauducq, 2017-10-07:
Can you file another issue about the second problem, if possible with more details about how to reproduce it?

jonaskeller, 2017-10-07:
Yes, "reset" refers to rebooting the FPGA board.

I have tried to find a way to reproducibly cause the dashboard to crash, but couldn't so far. When it happens, it's right after the first click on an "OVR" button after the FPGA reset, but it doesn't happen every time. If I can find out what makes the difference, I'll file a separate issue on it.

sbourdeauducq, 2017-10-13:
> Yes, "reset" refers to rebooting the FPGA board.

Generally the dashboard does not recover by itself from servers dying, instead it logs an error and requires a restart. It does the same if the master dies. We can have it reconnect automatically (as a rule), but it's a bit of work to ensure that all data (in applets etc.) is properly synchronized.

This wasn't a problem in ARTIQ-2 because moninj was using a UDP-based connectionless protocol, which had other issues, so ARTIQ-3 switched to a new TCP-based protocol.

jonaskeller, 2018-07-16:
I just tried again, and this time it seems reproducible (but this might be "luck").

I'm running artiq 3.6 py_42+git3e7cdaa5, nist_qc2 gateware, Windows 7 64bit

The recipe is
1. reset the FPGA while the dashboard is running
2. run this experiment:

        from artiq.experiment import *

        class DashboardCrash(EnvExperiment):
            '''cause dashboard crash'''
            
            def build(self):
                self.setattr_device("core")
                self.setattr_device("ttl0")
                        
            @kernel
            def run(self):
                self.core.break_realtime()
                self.ttl0.output()

3. click _any_ `OVR` button in the TTL panel

This crashes the dashboard with the following output

	WARNING:dashboard:quamash.QEventLoop:Event callback failed: [WinError 64] The specified network name is no longer available
	Traceback (most recent call last):
	  File "C:\Anaconda3\envs\artiq3.6\lib\asyncio\windows_events.py", line 437, in finish_recv
		return ov.getresult()
	OSError: [WinError 64] The specified network name is no longer available

	During handling of the above exception, another exception occurred:

	Traceback (most recent call last):
	  File "C:\Anaconda3\envs\artiq3.6\lib\site-packages\quamash-0.5.5-py3.6.egg\quamash\_windows.py", line 41, in _process_events
	  File "C:\Anaconda3\envs\artiq3.6\lib\asyncio\windows_events.py", line 440, in finish_recv
		raise ConnectionResetError(*exc.args)
	ConnectionResetError: [WinError 64] The specified network name is no longer available

	During handling of the above exception, another exception occurred:

	Traceback (most recent call last):
	  File "C:\Anaconda3\envs\artiq3.6\lib\site-packages\quamash-0.5.5-py3.6.egg\quamash\_windows.py", line 44, in _process_events
	  File "C:\Anaconda3\envs\artiq3.6\lib\asyncio\windows_events.py", line 79, in set_exception
		super().set_exception(exception)
	  File "C:\Anaconda3\envs\artiq3.6\lib\asyncio\futures.py", line 361, in set_exception
		raise InvalidStateError('{}: {!r}'.format(self._state, self))
	asyncio.futures.InvalidStateError: CANCELLED: <_OverlappedFuture cancelled>

The `device_db` entry for `ttl0` is

    "ttl0": {
        "type": "local",
        "module": "artiq.coredevice.ttl",
        "class": "TTLInOut",
        "arguments": {"channel": 0},
        "comment": "unassigned"
    },

jordens, 2018-12-14:
A couple notes from intensive use in the field on windows and linux for a couple weeks:

* If this appears, it's confusing that you can click the buttons, the UI changes, but the hardware does nothing. The errors in the log indirectly hint at something being wrong but that could be made clearer.
* The reconnect to moninj thing sounds relatively easy and safe. The state sync is already there and this disconnect case is much more common than the master disconnect. The master is pretty much rock solid and should hardly ever be restarted.
* I haven't been able to crash the dashboard with the recipe above on 4.0.

sbourdeauducq, 2019-06-13:
There is now a reproducible way to crash the dashboard:
* open the dashboard and let moninj connect
* ``artiq_coremgmt reboot``
* click one of the injection buttons, you get the error "lost connection to core device moninj"
* click it again, it crashes with
```
Traceback (most recent call last):
  File "/nix/store/bv9d5idzg2vl82d3xfw04ixzyzaxn90i-python3-3.7.3-env/lib/python3.7/site-packages/artiq/dashboard/moninj.py", line 97, in override_toggled
    self.set_mode(self.channel, "0")
  File "/nix/store/bv9d5idzg2vl82d3xfw04ixzyzaxn90i-python3-3.7.3-env/lib/python3.7/site-packages/artiq/dashboard/moninj.py", line 327, in ttl_set_mode
    self.core_connection.inject(channel, TTLOverride.level.value, 0)
  File "/nix/store/bv9d5idzg2vl82d3xfw04ixzyzaxn90i-python3-3.7.3-env/lib/python3.7/site-packages/artiq/coredevice/comm_moninj.py", line 64, in inject
    self._writer.write(packet)
AttributeError: 'CommMonInj' object has no attribute '_writer'
```

sbourdeauducq, 2019-06-13:
As an alternative solution, we could gray out the entire interface when a connection fails, and display a message about which connection failed and telling the user to fix the situation and restart the dashboard.