m-labs/artiq #839: Division by zero error when running artiq_browser on artiq 2.1

philipkent, 2017-10-11:
When trying to run artiq_browser on windows 7 a division by zero error is thrown in the scanwidget.py file that prevents artiq_browser from running.

Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

C:\Users\pdk1\artiq>activate artiq-2.1

(artiq-2.1) C:\Users\pdk1\artiq>artiq_browser
Traceback (most recent call last):
  File "C:\Users\pdk1\AppData\Local\Continuum\Miniconda3\envs\artiq-2.1\lib\site-packages\artiq\gui\scanwidget.py", line 232, in resizeEvent
    scale *= ev.size().width()/ev.oldSize().width()
ZeroDivisionError: division by zero

jordens, 2017-10-11:
Does this happen with 2.5 or 3.0?

philipkent, 2017-10-11:
It seems fine running on version 3.0.  I just changed environments to one with artiq 3.0 and artiq_browser opened without crashing.

jordens, 2017-10-11:
Try deleting the `artiq_browser.pyon` state file from `C:\Users\YOUR_USER\AppData\Local\m-labs\artiq\2`

commit 95b3ddba (2017-10-11): scanwidget: protect against resize from zero

fix #839
commit cd0d73a1 (2017-10-11): scanwidget: protect against resize from zero

fix #839
closed 2017-10-11 by jordens