m-labs/artiq #845: Dashboard core device connection occasionaly fails

kesht123, 2017-11-02:
Running 3.0 on Linux, kc705

Occasionally (~1/2 times a day), changing the level of a TTL override fails when using the dashboard with the following error thrown by the dashboard:

`quamash.QEventLoop:Fatal write error on socket transport`
`protocol:asyncio.streams.StreamReaderProtocol object at 0x7f08a79efc50>`
`transport: <_SelectorSocketTransport fd=41 read=idle write=<idle, bufsize=0>>`
`Traceback (most recent call last):`
`  File "/home/ion/anaconda3/envs/artiq3/lib/python3.5/asyncio/selector_events.py", line 761, in write`
`    n = self._sock.send(data)`
`TimeoutError: [Errno 110] Connection timed out`

subsequent attempts to toggle the TTL level produces:

`WARNING:dashboard:asyncio:socket.send() raised exception.`

This appears to be because the dashboard has lost connection to the core device, but I can't see why.
Restarting the dashboard fixes the issue.

whitequark, 2017-11-22:
Can you do the following:
  * run `artiq_corelog set_level DEBUG` before starting work
  * run `artiq_corelog` immediately once this issue appears next time and post the output
?

whitequark, 2017-11-26:
Waiting until the logs become available. This is possibly related to https://github.com/m-labs/smoltcp/pull/65; should be clear from the logs if it indeed is.

kesht123, 2017-11-28:
Sorry for the delay,

Looking at artiq_corelog, we see no change in behaviour before and after the issue occurs:
```
[  3237.609130s] DEBUG(smoltcp::iface::ethernet): non-unicast source address
[  3237.609241s] DEBUG(smoltcp::iface::ethernet): cannot process ingress packet: malformed packet
[  3237.609364s] DEBUG(smoltcp::iface::ethernet): packet dump follows:
EthernetII src=00-11-43-27-33-d0 dst=ff-ff-ff-ff-ff-ff type=IPv4
[  3237.609708s]  WARN(runtime): network error: malformed packet
[  3269.858515s] DEBUG(smoltcp::iface::ethernet): non-unicast source address
[  3269.858626s] DEBUG(smoltcp::iface::ethernet): cannot process ingress packet: malformed packet
[  3269.858749s] DEBUG(smoltcp::iface::ethernet): packet dump follows:
EthernetII src=00-0e-c6-f3-49-40 dst=ff-ff-ff-ff-ff-ff type=IPv4
[  3269.859093s]  WARN(runtime): network error: malformed packet
[  3273.081955s]  INFO(runtime::mgmt): new connection from 10.255.6.112:46454
[  3276.794198s]  INFO(runtime::mgmt): new connection from 10.255.6.112:46456
[  3313.498972s]  INFO(runtime::mgmt): new connection from 10.255.6.112:46488
[  3320.078733s] DEBUG(smoltcp::iface::ethernet): non-unicast source address
[  3320.078867s] DEBUG(smoltcp::iface::ethernet): cannot process ingress packet: malformed packet
[  3320.078997s] DEBUG(smoltcp::iface::ethernet): packet dump follows:
EthernetII src=70-20-84-07-41-0b dst=ff-ff-ff-ff-ff-ff type=IPv4
[  3320.079352s]  WARN(runtime): network error: malformed packet
```
We generally see a malformed packet warning a few times a minute on average. The issue happened just before the connection at 3273s when I immediately ran artiq_corelog. I have included the malformed packet errors before and after for reference.

I also saw the following shortly after:
```
[  3354.481210s] DEBUG(smoltcp::iface::ethernet): cannot process ingress packet: truncated packet
[  3354.481339s] DEBUG(smoltcp::iface::ethernet): packet dump follows:
[]
[  3354.481454s]  WARN(runtime): network error: truncated packet
[  3354.485820s]  WARN(runtime): ethernet mac: rx dropped: 14
```

We also see a slightly different error from the dashboard: `ERROR:dashboard:artiq.dashboard.moninj:lost connection to core device moninj` with no further detail. 
Then subsequently when we try and toggle a TTL as usual:
`WARNING:dashboard:asyncio:socket.send() raised exception.`

The full artiq_corelog dump is here:
[corelog.txt](https://github.com/m-labs/artiq/files/1509765/corelog.txt)

whitequark, 2017-12-08:
I've updated the master branch of ARTIQ so that the core device logs are now more useful. Can you please retry?

whitequark, 2017-12-27:
Waiting until the more detailed logs become available.

sbourdeauducq, 2018-01-09:
> I've updated the master branch of ARTIQ so that the core device logs are now more useful

Does 3.2 also include those log improvements?

whitequark, 2018-01-09:
I think those improvements were in smoltcp, but I do not remember exactly any more.

whitequark, 2018-01-13:
@kesht123 ping

sbourdeauducq, 2018-01-23:
@kesht123 (or @cjbe or @hartytp): is this still an issue? Can you post more logs if so?

kesht123, 2018-01-23:
Sorry for the delay..
I got logs a few days ago but there was several seconds of delay before I did so, so I don't know how useful they are:
[corelog2018-01-17.txt](https://github.com/m-labs/artiq/files/1656766/corelog2018-01-17.txt)
I had been trying to catch another loss of connection and get more timely logs but I haven't been able to so far

whitequark, 2018-01-23:
Nope, too late. I will make log levels adjustable for different components of the core device so that this isn't an issue.

whitequark, 2018-03-13:
@kesht123 It seems highly likely that the recent liteeth fix would solve this issue. Can you please upgrade to the latest release in 3.x series (or master) and retry?

sbourdeauducq, 2018-03-22:
Please upgrade to 3.6 and reopen if the problem persists.

closed 2018-03-22 by sbourdeauducq