m-labs/artiq #846: spiflash core on Sayma not working

sbourdeauducq, 2017-11-03:
Just reads back 0xffffff...

jbqubit, 2017-11-24:
@sbourdeauducq On which SPI flash chips has the ARTIQ code been tested? There's no mention of supported chips in the source AFACT. artiq/artiq/firmware/libboard/spiflash.rs

sbourdeauducq, 2017-11-24:
That's not the right source, you should be looking at MiSoC, and it has been tested on KC705, Papilio Pro, Pipistrello and probably others.

jbqubit, 2017-11-24:
I see. It's right [here](https://github.com/m-labs/misoc/blob/649d3eddbddda14f64474ec324b489ee02156691/misoc/targets/sayma_amc.py#L92).

@gkasprow is this configured as you expect? 

@sbourdeauducq Please consider referencing the IC name in source, here Micron N25Q256A, when extending MiSoC routines with specials. This can help rapidly identify where in the code base to look for clues about how particular chips are supported. I'm thinking about MiSoC users who may not know anything about what ICs are on the Sayma PCB. As this applies to other PHYs than spiflash does this warrant an independent Issue?

sbourdeauducq, 2017-11-25:
> I'm thinking about MiSoC users who may not know anything about what ICs are on the Sayma PCB.

They should look at the Sayma schematics then, not the MiSoC source.

closed 2017-12-14 by sbourdeauducq

jordens, 2017-12-14:
https://github.com/m-labs/misoc/commit/a1a58f62e5a236ecc2747d8d02342cea951a04bd